MockbaTheBorg/RunCPM #28: Is there a Problem in the Posix port or do I have a case of PEBCAK here?

drawkula, 2016-09-08:
I'm trying to build and run `RunCPM` on Debian-8.5/amd64 (tried on Debian-7.11/armhf (on Cubietruck) some days ago too) but I always get results like below...
```
CP/M 2.2 Emulator v2.6 by Marcelo Dantas
      Build Sep  8 2016 - 17:31:15
-----------------------------------------
CCP: CCP-DR.BIN Loaded at 0xf400

RunCPM Version 2.6 (CP/M 2.2 64K)

AA>dir
A: 1STREAD  ME  : BDOS     ASM : BDOS     SUB : CLEAN    SUB
A: DISKDEF  LIB : DUMP     ASM : ED       COM : EXIT     ASM
A: EXIT     COM : EXIT     SUB : EXIT     Z80 : LOAD     COM
A: MBASIC   COM : MLOAD    ASM : MLOAD    COM : MLOAD    DOC
A: MOVCPM   COM : OPCODES  TXT : STAT     COM : SUBMIT   COM
A: SYSGEN   COM : XSUB     COM : Z80ASM   COM : ZCPR     ASM
A: ZCPR     SUB : ZCPRHDR  LIB : ZEXALL   COM : ZEXDOC   COM
A: ZSID     COM : ZTRAN4   COM
A>exit
EXIT?

A>mbasic
MBASIC?

A>_
```
"Internal" commands like `dir` work but others do not.

MockbaTheBorg, 2016-09-08:
Hi there.

I have been actively finding and solving bugs these last few days,
especially on the other ports than Windows.
So please checkout the latest code from githut (as of today, Sept/8) and
give it a try.
It should work fine, and if it doesn't, please let me know.
Make sure also you #define CCP_DR for the initial testing, as my internal
CCP is still very crude code.
Other than that, it should work fine.
If there's any issue we can figure out and fix it.

Cheers,
The Borg.

MockbaTheBorg, 2016-09-08:
Oh wait ... I have just found another issue with the posix ... wait for a
new update where I will say "fix to the Posix file selection issue"

MockbaTheBorg, 2016-09-08:
It should be ok now. Give it a try and let me know.

drawkula, 2016-09-08:
Fresh `git pull` and compile yields the same problem.

kklis, 2016-09-08:
The filesystem on Linux is case sensitive. Your case happens if you have mbasic.com on disk. Rename it to MBASIC.COM (uppercase) and it will work.

drawkula, 2016-09-08:
The CCP, the drive A directory and all files in there are uppercase only.

kklis, 2016-09-08:
Git bisect shows that the behavior described by drawkula started with commit https://github.com/MockbaTheBorg/RunCPM/commit/07a63dcff3aaea6fa673d2deb20279208be91b8e

I just tested and on case sensitive filesystems the last correctly working version is https://github.com/MockbaTheBorg/RunCPM/commit/aa4cd2bca86e9a6ec4ee2405c0d10cc9bda9bd2e so IMHO there must be something missing when CCP-DR.BIN replaced CPM22.BIN

kklis, 2016-09-08:
I can see that everything works fine when you put the A directory on case insensitive filesystem, e.g. on a FAT32 formatted pendrive, create a symlink "A" to point to that directory, and start RunCPM. So it probably has something to do with upper/lower case not being handled properly by CCP-DR.

MockbaTheBorg, 2016-09-08:
I will verify this here, will let you know asap.

MockbaTheBorg, 2016-09-08:
I have verified that indeed the CCP from DR is having the issue, while the internal CCP and the ZCPR are not.
I recommend changing on the globals.h the CCP to ZCPR for now, so you can use the system while I attempt to figure out the reason for the DR's failure.
The internal CCP is still in its early stages (though dir/era/type and loading of executables are working), but ZCPR should give the same DR's functionality plus more.

MockbaTheBorg, 2016-09-08:
I believe the problem is solved. The original source code of DR's CCP fills up the FCB for opening an executable with lower case "com" ... therefore it can't find "MBASIC.COM" on a case sensitive filesystem.
I didn't want to change the original DR source code (for historical reasons) so I have changed the filename conversion to do the uppercase translation.
This seems to have solved the issue.

closed 2016-09-08 by MockbaTheBorg