MockbaTheBorg/RunCPM #31: Q: User areas

drawkula, 2016-09-15:
When "user areas" get turned on as default feature:
Are you planning to add creating the missing numbered subdirectories on the fly?

...and... RunCPM/globals.h, line 48:

> User numbers are 0-9, then A-F for users 12-15

...10-15...

MockbaTheBorg, 2016-09-15:
Changed ... no idea why I typed 12 there.
I have just added user area support. So I am not 100% sure yet if it is working.
Creating the numbered subdirectories is a good idea, especially because I am not checking for their presence right now which will create issues if you try to make a file there.
So let's move it to enhancement for now.

MockbaTheBorg, 2016-09-20:
I have added the auto-creation of the selected user folder ... I have been using the user support mode for a while now, so I believe it is good for release. Please test when you have a chance.
Thanks a lot.

closed 2016-09-20 by MockbaTheBorg

drawkula, 2016-09-21:
I had user areas activated for some days without seeing trouble.
And the new folder auto creation looks ok too.