MockbaTheBorg/RunCPM #35: Existing nonexisting files... I'm confuzzeld...

drawkula, 2016-09-21:
### Once upon a time...

`C:` contains `Hi-Tech-C V3.09`.
...and there is a file `OPTIONS`:
```
CP/M 2.2 Emulator v2.7 by Marcelo Dantas
      Build Sep 21 2016 - 08:26:21
-----------------------------------------
CCP: CCP-DR.BIN Loaded at 0xf400

RunCPM Version 2.7 (CP/M 2.2 64K)

A>c:
C>dir options
C: OPTIONS     
C>type options
OPTIONS?

C>a:stat options

 Recs  Bytes  Ext Acc
    0     0k    1 R/W C:OPTIONS
Bytes Remaining On C: 960k

C>
```

...looking from Debian:
```
runcpm/run$ ls -l C/0/OPTIONS 
-rw-r--r-- 1 yeti yeti 768 Sep 18 08:07 C/0/OPTIONS
runcpm/run$ stat -cx%nx C/0/OPTIONS
xC/0/OPTIONSx
runcpm/run$ _
```
It exists, has a length and has no spaces after the filename.

### Experiment to reproduce it with an own file...
```
runcpm/run$ echo WORLD > C/0/HELLO
runcpm/run$ _
```
Look from CP/M...
`CP/M 2.2 Emulator v2.7 by Marcelo Dantas
      Build Sep 21 2016 - 08:26:21
-----------------------------------------
CCP: CCP-DR.BIN Loaded at 0xf400

RunCPM Version 2.7 (CP/M 2.2 64K)

A>c:
C>dir hello
C: HELLO       
C>type hello
HELLO?

C>era hello
``
...then `RunCPM` hangs.

It is ok to look confused now?

MockbaTheBorg, 2016-09-21:
Now I am the one confused ... lol ... I understand the problem but I am not sure yet how to fix it.
Interestingly it doesn't happen with the windows port, so it is related to the posix abstraction.

I will take a look and let you know.

MockbaTheBorg, 2016-09-21:
Solved ... now you should be able to type the files normally on Debian.

closed 2016-09-21 by MockbaTheBorg

drawkula, 2016-09-21:
After renaming `OPTIONS.` back to `OPTIONS`:
```
RunCPM Version 2.7 (CP/M 2.2 64K)

A>c:
C>type options
Z80 CP/M C compiler options:

-A      Generate a self-relocating .COM program
-R      Link in command line wild card expansion code
-V      Be verbose during compilation
-S      Generate assembler code in a .AS file; don't assemble or link
-C      Generate object code only; don't link.
-O      Invoke the peephole optimizer
-I      Specify an include directory, e.g. -I1:B:
-U      Undefine a predefined symbol, e.g. -UDEBUG
-D      Define a symbol, e.g. -DDEBUG=1
-L      Scan a library, e.g. -LF scans the floating point library
-F      Generate a symbol file suitable for use with debug.com, e.g.
                -Ffile.sym; default file name is L.SYM.
-W      Set warning level, e.g. -w5 or -w-2
-X      Suppress local symbols in symbol tables
-M      Generate a map file, e.g. -Mfile.map

C>_
```
Fine!
Life is easier this way because &ast;nixish `unzip` unpacks files without extension without trailing dot.
\o/