MockbaTheBorg/RunCPM #36: readme.md ... open questions.

drawkula, 2016-09-21:
### 32bit?

> RunCPM is a 32 bits application which allows you to execute old CP/M 8 bits programs on Windows, Mac OS X, Linux, FreeBSD, MS-DOS and Arduino DUE.

I don't use any compiler options on Debian to force 32bit code generation...
```
runcpm/src/RunCPM/RunCPM$ file RunCPM
RunCPM: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=eb4fa888e367bcf02ffba3dbfb2f394b8d345bf1, not stripped
```
...so what is dropping out of the compiler is a 64bit executable.

### Getting Started

...needs to mention the user areas now and where to drop the contents from `A.ZIP` and/or packing `A.ZIP` to contain the path `A/0` may be an option...

MockbaTheBorg, 2016-09-22:
Done ... please check for typos/clarity.

closed 2016-09-22 by MockbaTheBorg

drawkula, 2016-09-22:
Only some minor things:

* You write some program names in italic but not all and I can't see why in one place and why not in the other...
* Just one space after something set in italic looks a bit tight.  
_TRY_ this.  
_TRY_&nbsp; this.  
The later one uses a non breakable space char (&amp;nbsp;) after italic but mixing HTML and MarkDown formatting is ugly and/or confusing.  
Italic is mean, this spacing problem is present for decades (from T<sub>E</sub>X to HTML or even longer) and noone seems to provide an automagic fix.  
Maybe it is possible to drop using italic it totally?  
Use `backticks` for commands and filenames instead?
* "RunCPM can emulate the user areas as well (this is the default), ..."  
Because using user areas now is the default, I'd explain the directory layout with user areas first.
* Is there any special reason to write the `bin` in lowercase in `CCP-DR.bin` and `CCP-ZCPR.bin`?
* BASIC is an abbreviation, so probably MBASIC should be all caps...
* "Cygwin can be used to build the posix implementation on Windows machines."  
Is Posix a name like Windows that always should be written starting with a capital?  
When the Makefle's suffix .posix is meant, this sure does not apply.
* "Posix builds use gcc/llvm."  
https://gcc.gnu.org and http://llvm.org write GCC and LLVM in caps when the whole project or software package is meant. 
* "... (available on Disk A.ZIP)."  
"Disk" --> "disk"

...and I just learned that flavor is american english while flavour is british english... like color vs colour... \o/ ...yay... ;-)

MockbaTheBorg, 2016-09-22:
>    The later one uses a non breakable space char ( ) after italic
>    Italic is mean, this spacing problem is present for decades (from TEX

I believe I covered it all ... please double check ...

Thanks,
Marcelo.

drawkula, 2016-09-22:
"On that folder goes both the executable and the CCP binaries for the system."  
I'd say "in that folder"...  
"Create a folder containing the RunCPM executable and the CCP binaries for the system." ... perhaps is clear enough?

That's all for now... ;-)

MockbaTheBorg, 2016-09-22:
Done