MockbaTheBorg/RunCPM #37: [FAQ?] What would I need to know to compile and use a different external `CCP` for `RunCPM`?

drawkula, 2016-09-24:
### [FAQ?] What would I need to know to compile and use a different external `CCP` for `RunCPM`?

[The CPM-2.2 example disks set](http://schorn.ch/altair_4.php) of [AltairZ80](http://schorn.ch/altair.html) includes `CCPZ  Version 4.1` ([attached here as ZIP](https://github.com/MockbaTheBorg/RunCPM/files/491080/CCPZ-taken-from-AltairZ80.zip)).

What would I need to know about `RunCPM` to get `CCPZ` built and running with `RunCPM`?

MockbaTheBorg, 2016-09-24:
Basically you need to do this:

1. Find out where the BDOS and BIOS of RunCPM are located. The INFO command on the disk A.ZIP chows you that.
2. Verify on the source code of CCPZ if these are referenced there and adjust accordingly.
3. Normally the CCP is located on BDOS - XXXX ... you may need to figure out this XXXX from its source code and then adjust CCPZ's ORG accordingly.
4. Once you have the source code ready (sometimes the modifiable values go on a LIB file which is inserted by the assembler source code, not on the source code itself.
5. Verify if Z80ASM builds it with no errors. Maybe you will need a specific assembler to do so.
6. If all things are good, build it and give it a try.

You will have to look inside the binary where the $$$.SUB fcb ends up after built, so you can adjust RunCPM's BatchFCB define accordingly, or else SUBMIT wont work, but this is only after you have it running already.

To be on the safe side, I recommend getting the latest source code, as I made many changes last night, setting it to 60K, which was the usual "maximum" in those days, and going from there.
Only after you get it working on 60K you should try 64K.

Once you have everything working fine ... share it ... :)

I won't be able to take a look at this CCP until much later today, by then I believe you will have made it work already ... :)

Good luck.

MockbaTheBorg, 2016-09-24:
After a quick look at the CFGCCPZ.LIB: both "bioslen" and "bdoslen" are 0200h.
Also, it looks like you can just dump the CFG file and modify the asm code directly.
This thing is VERY similar to ZCPR2.

MockbaTheBorg, 2016-09-24:
Damn ... I couldn't resist ... lol:
1. Copy the source code to a disk.
2. Rename CCPZ.MAC to CCPZ.Z80
3. Comment out the first 10 lines of the code, except the ORG
4. Set the ORG to 0e400h for 60k or 0f400h for 64k
Build with A:Z80ASM CCPZ, it should build with no errors.
The offset for the $$$.SUB FCB is 0x7C

MockbaTheBorg, 2016-09-24:
Correction ... the offset for the $$$.SUB FCB is 0x7A

drawkula, 2016-09-24:
I just wanted to start by copying `L80` ans `M80` and asking `a:info` but you were faster.
Let's see what I can get with `Z80ASM`...

...a while later:
```
CP/M 2.2 Emulator v2.7 by Marcelo Dantas
      Build Sep 24 2016 - 15:31:48
-----------------------------------------
CCP: CCPZ.64K Loaded at 0xf400

RunCPM Version 2.7 (CP/M 2.2 64K)

A>user 1
A1>_
```

I defined `CCPZ` directly after `CCP-DR` in `globals.h` because they are close relatives:
```
diff --git a/RunCPM/globals.h b/RunCPM/globals.h
index c098100..9c7f789 100644
--- a/RunCPM/globals.h
+++ b/RunCPM/globals.h
@@ -22,7 +22,8 @@
 
 /* Definition of which CCP to use: INTERNAL, DR or ZCPR (must define only one) */
 //#define CCP_INTERNAL // If this is defined, CCP will be internal
-#define CCP_DR
+//#define CCP_DR
+#define CCPZ
 //#define CCP_ZCPR
 
 /* Definition of the CCP memory information */
@@ -40,6 +41,12 @@
 #define BatchFCB       (CCPaddr + 0x7AC)       // Position of the $$$.SUB fcb
 #endif
 //
+#ifdef CCPZ
+#define CCPname                "CCPZ." STR(SIZEK) "K"
+#define VersionCCP     0x00
+#define BatchFCB       (CCPaddr + 0x7AC)       // Position of the $$$.SUB fcb
+#endif
+//
 #ifdef CCP_ZCPR
 #define CCPname                "CCP-ZCPR." STR(SIZEK) "K"
 #define VersionCCP     0x01
```

Now I need to do some more tests with it...

drawkula, 2016-09-24:
This was the reason to think about running `CCPZ`:
```
C>user 1
C1>c
HI-TECH C COMPILER (CP/M-80) V3.09
Copyright (C) 1984-87 HI-TECH SOFTWARE
c> ^C
RunCPM Version 2.7 (CP/M 2.2 64K)

C1>dir
No File
C1>_
```
Noticing the ["Change to Internal CCP search path" commit](https://github.com/MockbaTheBorg/RunCPM/commit/71ba8e5d00af1889400377f79f16664760c649a8) I thought playing with `CCPZ`'s way to do it would make sense...

MockbaTheBorg, 2016-09-24:
Cool ... after some testing I will probably make CCPZ the default.

drawkula, 2016-09-24:
Houston, <strike>we</strike>I have a problem...

`submit` hates me:
```
A>submit ccpz

RunCPM Version 2.7 (CP/M 2.2 64K)

A$; BUILD CCPZ.Z80 INTO CCP-CCPZ.BIN
A$ERA CCPZ.COM
No File
A$Z80ASM CCPZ

Z80ASM Copyright (C) 1983-86 by SLR Systems Rel. 1.32 #RunCPM

 CCPZ
End of file Pass 1
Non-Standard COM File
 0 Error(s) Detected.
 2031 Absolute Bytes. 235 Symbols Detected.

RunCPM Version 2.7 (CP/M 2.2 64K)

A$ERA CCP-CCPZ.BIN
CCP-CCPZ.BIN
A$REN CCP-CCPZ.BIN=CCPZ.COM
A>c:
C>type hello.sub
; create hello.com
c hello.c
hello

C>submit hello

RunCPM Version 2.7 (CP/M 2.2 64K)

C>_
```
`ccpz.sub` (I only added the `era`-lines) works on `A:`.<br>`hello.sub` on `C:` does not.<br>I have no idea where to go on with searching.

drawkula, 2016-09-24:
Sorry... closing was a pure accidentally clicking the wrong place error...

MockbaTheBorg, 2016-09-24:
I suspect the issue may be on my CheckSub function. It looks for the $$$.sub from the FCB defined by BatchFCB offset. If that FCB gets its drive modified for some reason, then it may make the CCP think the $$$.SUB is not present in A:
Also, it should look for $$$.SUB on user area 0 of drive .

Making it look like this made the "submit hello" work:

`uint8 _CheckSUB(void) {
	_HostnameToFCB(tmpFCB, (uint8*)"$$$.SUB");
	_RamWrite(tmpFCB, 1); // Forces $$$ to be checked on drive A:
	return((_SearchFirst(tmpFCB, FALSE) == 0x00) ? 0xFF : 0x00);
}
`
but this is less than optimal. A good, more definitive solution, should make sure all accesses to $$$.SUB (using BatchFCB) would always refer to a file on A: user area 0.
I will close this one as it started as a different subject (which is solved), and open another one.

closed 2016-09-24 by MockbaTheBorg