MockbaTheBorg/RunCPM #38: Improve the handling of $$$.SUB file

MockbaTheBorg, 2016-09-24:
$$$.SUB should always be checked on drive A: user area 0.
Even after many years, no standard was defined for this, and some CCPs and implementations of SUBMIT handled it differently.
A: user area 0 seems so be a good base for a standard (especially nowadays, when no one will complain about it), so I will implement this shortly.

drawkula, 2016-09-24:
`CCPZ` has a configuration (`SUBA`)variable to force the creation of the submit file always to be on `A:`. There is something different going on. I sometimes even find `$$$.SUB` on `A:` or notice it being executed after a reboot... but not always.

MockbaTheBorg, 2016-09-25:
I believe I have closed this issue with my last commit. Well, kinda.
SUBMIT is kinda dumb ... it creates $$$.SUB on the drive A:, but on the current user, when I believe it should be doing it on drive A: user 0. But again, it looks like there was never a consensus on this.
In theory, using the DRI CCP, you cannot run SUBMIT from A: if you are on a different user area. To be able to do so you would need to have SUBMIT on every user area of A: that you intend to use.
Right now I am forcing the BDOS to look for $$$.SUB on drive A: user 0.
Another option would be to looks for $$$.SUB on drive A: current user. Maybe this is a better solution, as it should then work fine with CCPZ and other Z's that load commands from A:0 by default.
To be honest, not thinking about it a little more, maybe it would be better to create $$$.SUB on the current drive/user, and force the BDOS to look for it there.
So we get back the same discussion from the early 80's, when we couldn't get to a conclusion, while listening to Men at Work on the FM.

MockbaTheBorg, 2016-09-25:
I believe the issue can be closed. All the CCPs seem to be working fine with the patched version of SUBMIT.COM (which creates the $$$.SUB on drive A:/same user).
Only Z80CCP needs the unpatched version of the SUBMIT, which creates the $$$.SUB on the current drive/user.
I did a few tests, please test a little more before we close this for good.

MockbaTheBorg, 2016-09-26:
After a few test sessions, this seems to be working fine.

drawkula, 2016-09-26:
Real life bytes... sometimes...

I get test results I don't even understand... it does not even work as expected on `A0:`:
```
A>type test0.sub
; testing
dir test*.sub
; end

A>submit test0

A>_
```
...but `submit clean` works...

On `C1:` a sub-file of some days ago works:
```
C1>type myhi.sub
; create myhi.com
c myhi.c
myhi
;

C1>submit myhi

C1$; CREATE MYHI.COM
C1$C MYHI.C
HI-TECH C COMPILER (CP/M-80) V3.09
Copyright (C) 1984-87 HI-TECH SOFTWARE

C1$MYHI
Hi, RunCPM!

C1$;
C1>_
```
...but a newly written one in the same directoty does not:
```
C1>type mandelbr.sub
; create mandelbr.com
c mandelbr.c
mandelbr
;

C1>submit mandelbr

C1>_
```
I'll try to find a pattern...

MockbaTheBorg, 2016-09-26:
Are you creating these .SUB files from inside CP/M itself? Like on "copy test.sub=con:", or creating them from inside Linux ?
If you are creating from inside Linux (or any other host file system) you need to add a {SUB} (0x1A or ^Z) character as the last character of the file. 
I have learned the hard way that you need the "eof" character as the last character of a file ... :(
Please double check and let me know.

drawkula, 2016-09-26:
All of them were written with `te` or `ws`.

Runs:
```
00000000 3B 20 63 72 │ 65 61 74 65 │ 20 6D 79 68 │ 69 2E 63 6F │ 6D 0D 0A 63  ; create myhi.com..c    
00000014 20 6D 79 68 │ 69 2E 63 0D │ 0A 6D 79 68 │ 69 0D 0A 3B │ 0D 0A 1A 1A   myhi.c..myhi..;....    
00000028 1A 1A 1A 1A │ 1A 1A 1A 1A │ 1A 1A 1A 1A │ 1A 1A 1A 1A │ 1A 1A 1A 1A  ....................    
0000003C 1A 1A 1A 1A │ 1A 1A 1A 1A │ 1A 1A 1A 1A │ 1A 1A 1A 1A │ 1A 1A 1A 1A  ....................    
00000050 1A 1A 1A 1A │ 1A 1A 1A 1A │ 1A 1A 1A 1A │ 1A 1A 1A 1A │ 1A 1A 1A 1A  ....................    
00000064 1A 1A 1A 1A │ 1A 1A 1A 1A │ 1A 1A 1A 1A │ 1A 1A 1A 1A │ 1A 1A 1A 1A  ....................    
00000078 1A 1A 1A 1A │ 1A 1A 1A 1A │                                          ........                
```

Fails:
```
00000000 3B 20 63 72 │ 65 61 74 65 │ 20 6D 61 6E │ 64 65 6C 62 │ 72 2E 63 6F  ; create mandelbr.co    
00000014 6D 0D 0A 63 │ 20 6D 61 6E │ 64 65 6C 62 │ 72 2E 63 0D │ 0A 6D 61 6E  m..c mandelbr.c..man    
00000028 64 65 6C 62 │ 72 0D 0A 3B │ 0D 0A 0D 0A │ 1A 1A 1A 1A │ 1A 1A 1A 1A  delbr..;............    
0000003C 1A 1A 1A 1A │ 1A 1A 1A 1A │ 1A 1A 1A 1A │ 1A 1A 1A 1A │ 1A 1A 1A 1A  ....................    
00000050 1A 1A 1A 1A │ 1A 1A 1A 1A │ 1A 1A 1A 1A │ 1A 1A 1A 1A │ 1A 1A 1A 1A  ....................    
00000064 1A 1A 1A 1A │ 1A 1A 1A 1A │ 1A 1A 1A 1A │ 1A 1A 1A 1A │ 1A 1A 1A 1A  ....................    
00000078 1A 1A 1A 1A │ 1A 1A 1A 1A │                                          ........                
```

MockbaTheBorg, 2016-09-26:
Yup ... there seems to be an issue here ... reopening.

MockbaTheBorg, 2016-09-26:
For some reason, with this specific file, SUBMIT opens the $$$.SUB but writes nothing to it. There is no call to the "WriteSeq" BDOS function.
So I guess I will need to do some debugging on SUBMIT.COM itself.
I will keep you posted.

drawkula, 2016-09-26:
Testing `SUPERSUB` which is `DO.COM` on AltairZ80's CP/M-2.2 disk:
```
C1>do

How to use SUPERSUB:

SUPERSUB<CR>            :print this HELP message
SUPERSUB /<CR>          :go into interactive mode
SUPERSUB /<cmd lines>   :use SUMMARY mode
SUPERSUB <FILE> <PARMS> :as in standard SUBMIT.COM

IN "/" (interactive) mode, SUPERSUB will prompt you
a line at a time for the SUBMIT job input...logical
lines may be combined on the same input line by sep-
erating them with semicolons.  Example:
   A>SUPERSUB /STAT;DIR
specifies two commands on the same input line.

Submitted jobs may be nested...SUPERSUB does not erase
any existing submit job (appends to them instead).

To insert a control character into the output, pre-
fix it with a "^" (works in any mode).

C1>ssub mandelbr
SuperSUB V1.1

C1$; CREATE MANDELBR.COM
C1$C MANDELBR.C
HI-TECH C COMPILER (CP/M-80) V3.09
Copyright (C) 1984-87 HI-TECH SOFTWARE

C1$MANDELBR
                                                                               
                                                                               
                                                                               
                                                       **                      
                                                    *******                    
                                          *     * *** *** *** *                
                                            ************************           
                                         ***************************           
                              *         *******************************        
                          *********   *********************************        
                        *********************************************          
         *        ************************************************             
                        *********************************************          
                          *********   *********************************        
                               *        ******************************         
                                         ****************************          
                                            ************************           
                                                * ***********                  
                                                    ****** *                   
                                                       **                      
                                                                               
                                                                               
                                                                               

C1$;
C1>_
```
`$` parameters work as expected with `SUPERSUB`:
```
C1>type cr.sub
; $1.c -> $1.com
era $1.com
c $1.c
$1

C1>do cr echo
SuperSUB V1.1

C1$; ECHO.C -> ECHO.COM
C1$ERA ECHO.COM
ECHO    .COM
C1$C ECHO.C
HI-TECH C COMPILER (CP/M-80) V3.09
Copyright (C) 1984-87 HI-TECH SOFTWARE

C1$ECHO
echo> 123
123
C1>_
```
Maybe assimilating `SUPERSUB` and documenting `SUBMITS` misbehaviour is the efficient solution?

MockbaTheBorg, 2016-09-27:
Found the issue ... SUBMIT has a bug ... if the last line of the .SUB is empty, it bails out without writing anything.
If you leave empty lines on the script, it also fails.
Same happens on a real CP/M machine.

closed 2016-09-27 by MockbaTheBorg

drawkula, 2016-09-27:
Aaaaaah:
```
   C:CR.SUB            L00006  C01 Insert                                       
               N O N - D O C U M E N T   E D I T   M E N U
    CURSOR      SCROLL        ERASE      OTHER               MENUS
 ^E up       ^W up         ^G char    ^J help             ^K block & save 
 ^X down     ^Z down       ^T word    ^I tab              ^P print controls 
 ^S left     ^R up screen  ^Y line    ^V turn insert off  ^Q quick functions 
 ^D right    ^C down      Del char    ^O set tab width   Esc shorthand 
 ^A word left   screen     ^U unerase ^N split the line
 ^F word right             ^B top bit ^L find/replace again 
; $1.c -> $1.com                                                               <
era $1.com                                                                     <
c $1.c                                                                         <
$1                                                                             <
                                                                               <
_                                                                              ^
                                                                               ^
                                                                               ^
                                                                               ^
                                                                               ^
                                                                               ^
                                                                               ^
                                                                               ^
                                                                               ^
                                                                               ^
```
...versus...
```
te: CR.SUB                                   Lin:0005/0005/0512 Col:01/80 Len:00
!.......!.......!.......!.......!.......!.......!.......!.......!.......!.......
; $1.c -> $1.com
era $1.com
c $1.c
$1
_

Press [ESC] to show the menu.
```
`_` shall show the cursor position.

In `ws` the cursor stops after the last line, in `te` the cursor stops  in the last line.

That's why my scripts got different...