MockbaTheBorg/RunCPM #41: Bdos Error on A : Select

SmallRoomLabs, 2017-02-16:
Compiling using the Posix makefile and running this on Linux (Lubuntu) for both CCP_CCPZ with 64K and the CCP_DR with 60K only gives Bdos errors as below.  No amount of pressing Enter or ^C ever resolved the problem.

```CP/M 2.2 Emulator v2.8 by Marcelo Dantas
      Build Feb 17 2017 - 00:49:00
       with Lua scripting support
-----------------------------------------
CCP: CCP-DR.60K  CCP Address: 0xe400
RunCPM Version 2.8 (CP/M 2.2 60K)
Bdos Error on A : Select
```
Trying the CCP_INTERNAL I also get the error. Here two Enters gives me the A> -prompt, but the error comes back as soon as I try a disk access.

```
CP/M 2.2 Emulator v2.8 by Marcelo Dantas
      Build Feb 17 2017 - 00:54:55
       with Lua scripting support
-----------------------------------------
CCP: INTERNAL v1.3  CCP Address: 0xe400
RunCPM Version 2.8 (CP/M 2.2 60K)
Bdos Error on A : Select
Bdos Error on A : Select
A>DIR
Bdos Error on A : Select
```

The build process seems ok except for some minor nagging by the archiver that says:
```ar: `u' modifier ignored since `D' is the default (see `U')```

SmallRoomLabs, 2017-02-16:
Ohnoes....  This was a case of me not reading the dox good enough.  I thought that the images in the CCP folder was complete boot disks with the basic utilities on them.

After creating an A/0/ folder and populating it with some CP/M apps it works just fine :+1:

closed 2017-02-16 by SmallRoomLabs

MockbaTheBorg, 2017-02-17:
Got me scared for a moment ... :) ... enjoy!

guidol70, 2018-10-27:
Now at the actual Windows 10 version I got this BDOS-Error-problem with all of my 3 used versions (v2.8 , v3.0 and v3.2 - downloaded from you github-page).
v2.8 and v3.0 did work months before in this directory-situation 
I got a C:\RunCPM\A\0 directory with the old or new files out of A.ZIP

But now it wouldnt work anymore.
Also did try with Admin-rights - but no go :( (V3.3 on my DUE is working also like on a Pi computer the linux version).

Any i
[image]
[image]
[image]

dea?

guidol70, 2018-10-28:
Hmmm - now I did found a working RunCPM in my C:\TEMP\RunCPM_Win Directory.
Copies v3.0 and v3.2 there and also the "new" Drive A: files.
And its working again in this Path and in the former not working C: \RunCPM_Win

I dont know whats the difference - but I backuped the working version now as a .ZIP ;)