MockbaTheBorg/RunCPM #42: Terminal emulation?

AndrewInEssex, 2017-04-18:
I've tried configuring wordstar to use an ansi/vt100 terminal but it isn't working; is the terminal emulation something I need to turn on?

drawkula, 2017-04-18:
On Debian-8.7/amd64 running "RunCPM/posix" in an XTerm: Wordstar behaves as expected.

[image]

[image]

Which RunCPM version are you running?  
With which terminal?

AndrewInEssex, 2017-04-18:
Thanks for the reply; I’m running 2.8; I don’t know how to set the terminal – that’s probably the problem.

-Andrew

 

 <https://cloud.githubusercontent.com/assets/9768622/25145948/e4d1cbd2-2461-11e7-821d-efbf778110a4.png> 

 <https://cloud.githubusercontent.com/assets/9768622/25146036/0f9017a2-2462-11e7-9c97-159713788b0c.png>

AndrewInEssex, 2017-04-18:
Sorry, I forgot to mention I'm running under Windows 10/64bit

drawkula, 2017-04-18:
I cannot help in the Windows universe.
:-(

AndrewInEssex, 2017-04-18:
I found a utility called ansicon which adds ANSI terminal support to a Windows command prompt. It's working now!

closed 2017-04-18 by AndrewInEssex

drawkula, 2017-04-18:
Maybe "ansicon" should be added to the RunCPM-FAQ... ;-D

...as soon as there is a FAQ...

MockbaTheBorg, 2017-04-19:
Well ... I see it is "fixed" already, but for some reason it is running fine for me.
I am also on W10/64bits, and the latest "Creator" update (more like "Destructor" update) seemed to break a lot of things but RunCPM.
When I was on Windows 7 I had to use Ansicon to give ANSI support to the CMD, but on Windows 10 it seemed to have ANSI support natively.
Again, I may be wrong, as Windows seems to do as it pleases among different people's computers.

AndrewInEssex, 2017-04-19:
Looking around the net, ANSI support in the command screen seems to have been quietly discontinued quite recently.

 

-Andrew