MockbaTheBorg/RunCPM #45: cannot run adv.com  (Classic Adventure Game)

mbramwel, 2017-09-28:
I have teensy 3.6
Arduino 1.8.4
RunCPM Version 2.8 

I have tried 64k and 60k
I have tried various cpm, cpmz, cpmz2, cpmz3, internal etc....

When I try to run adv.com, I get "Glitch! Symbol table not found. STOP UGG!"
That string is found inside the adv.com

I have tried the same files on a real Z80 CPM 2.2 and they run aok.

Is there some debugging I can try?

MockbaTheBorg, 2017-10-08:
Hmmm ... this is strange.
If adv.com doesn't use anything "fancy" like direct access to disk sectors and other hardware related stuff, then it should work fine.
RunCPM fully emulates the filesystem, but not as a physical disk ... and if physical access is used, it will fail silently.
Another option would be to disassemble adv.com and try to figure out what it does right before presenting the message, that could give us a clue.

MockbaTheBorg, 2017-10-08:
I just tried two different (copies I could find of) ADV.COM here and I get "Missing variable: HERE". running on RunCPM Windows.
Looks like ADV.COM is a "tough cookie" ... we would need to disassemble it and take a look inside. Have you tried that?

MockbaTheBorg, 2017-10-08:
Tried disassembling it here, found the strings, but the code is a bit tricky to follow. Lots of unreferenced code and variables.
I couldn't find a reference to that string in the code, so just by a quick look at the disassembly I couldn't find a reason for the error.
So at this point I would suggest generating a log of all the called BDOS functions, this could help, but could potentially generate a huge log file.
Other option would be to put "prints" in the code (jumps to a "bdos print piece of code") showing like: "got to here 1". "got to here 2", etc.

Keep in mind also that RunCPM has a (limited) debugger on it.

mbramwel, 2017-10-09:
Thanks for taking a quick look.  I'll do some logging to see what happens.

drawkula, 2017-10-09:
I tried the ADV version I could find in [AltairZ80](http://schorn.ch/altair.html)'s downloadable disks on RunCPM and got the "Symbol table not found." error too.
```
>NEWS

 
This is the brand-spanking-new B01 version of Adventure.
The cave is essentially stable at this point except for bug
fixes being done as needed.  The cave is almost twice as big as
before, and has lots of new creatures in it - have fun!
 
Please contact Mike Goetz at (212) 671-2490 if anything weird happens
(that is, anything weird that looks like it shouldn't happen).  In
particular, if you ever see the message "Glitch!", please save
the printout for error analysis if that's possible.
``` 
So this version runs somewhere else and shows the error in RunCPM and maybe the information `B01 version of Adventure` can help to find the source code somewhere?

MockbaTheBorg, 2017-10-23:
Hi guys ... did one of you ever had a chance to try and troubleshoot this one a little bit?
I am swamped at work, but knowing that RunCPM is failing at something is bothering me.
I wonder if this adventure cuts some corners to get things done, and ends up not using some BDOS or System call as expected.
Or maybe there is indeed some issue with RunCPM that I am missing.

mbramwel, 2017-10-23:
I took a look at it.  No luck.  
I tried to find source code.  No luck. 

I cannot even tell what language it might be because this *game* was ported into just about every known environment using various languages and tools.

 

I tried running with debug.  I carefully went through the logs.  Nothing jumped out as odd.  It reads and processes lots of data then glitches and stops.

 

Related topic:  I noted some other guys were trying to get an old BASIC running under z80pack.  The author just got it going by implementing correctly the undocumented opcodes. Perhaps that is the problem.  The next release of z80pack will have the fixes for those opcodes. I am curious to see what are the changes.  Considering the age of this game, perhaps a few undocumented opcodes were used in those early compilers.

kklis, 2017-10-23:
If it brings anything valuable to the discussion, I tried to run ADV.COM downloaded from schorn.ch, on stock Amstrad CPC 6128 with original CP/M system disks. Under CP/M 2.2 it crashes with "Out of memory while storing keys from file: ADVT PTR STOP HELP!". Under CP/M Plus it runs fine and displays "Welcome to the *new* Adventure! Say "NEWS" to get up-to-date game details."

MockbaTheBorg, 2017-10-25:
Well ... I am not sure about undocumented opcodes. RunCPM passes both ZEXALL and ZEXDOC tests, but I don't know either is those tests do cover all relevant undocumented opcodes.
I would love to have RunCPM 100% compatible with all Z80 opcodes, but ZEXALL was the only test I put my hands on.
If anyone has any other test, please send over, as this can indeed be a cause of strange problems.

Other than this, due to RunCPM's obvious limitations of not implementing a direct track/sector access on the file system, we could also see strange behaviors on disk related actions.

Hoping that we will find something on the Z80 side.

MockbaTheBorg, 2017-10-25:
@kklis : That is a good point. If ADV.COM requires any non CP/M 2.2 functionality to run, then it won't probably run on RunCPM, as RunCPM is (or tries to be) strictly compatible with CP/M 2.2 only.

drawkula, 2017-10-25:
@MockbaTheBorg — The version I tested above runs on AltariZ80 with CP/M-2.2...
```
Altair 8800 (Z80) simulator V3.9-0 build 1000 (scp created Jun 20 2017 at 15:02:55 with gcc 6.3.0 20170516)

64K CP/M Version 2.2 (SIMH ALTAIR 8800, BIOS V1.27, 2 HD, 02-May-2009)

A>_
```
...but not on RunCPM/Posix.
```
CP/M 2.2 Emulator v2.8 by Marcelo Dantas
      Build Oct 25 2017 - 16:43:01
       with Lua scripting support
-----------------------------------------
CCP: CCP-CCPZ.64K  CCP Address: 0xf400

A>g:
G>dir
ADV     .COM  |  ADVDDIA .SAV  |  ADVDDOM .SAV  |  ADVI    .DAT
ADVI    .PTR  |  ADVT    .DAT  |  ADVT    .PTR
G>adv

Glitch!  Symbol table not found. STOP  UGG!  

G>_
```
It probably is the same version @kklis tested because I downloaded from the same location.

The error on @kklis's "Amstrad CPC 6128 with original CP/M" is different to the one mentioned by @mbramwel and may be caused by a smaller TPA on the CPC 6128.

Can we easily hack logging of unimplemented BIOS/BDOS calls into RunCPM?

I have rebuilt RunCPM with DEBUG and DEBUGLOG and [the log is attached here](https://github.com/MockbaTheBorg/RunCPM/files/1415449/RunCPM.log-1508950299.gz).
```
$ zgrep 'B.os call:  .. ' RunCPM.log-1508950299.gz | sort | uniq -c
      3 Bdos call:  10 (Read Buffered) IN from 0xf6c9:
      2 Bdos call:  13 (Reset Disk) IN from 0xf513:
      1 Bdos call:  14 (Select Disk) IN from 0xf522:
      1 Bdos call:  14 (Select Disk) IN from 0xfb03:
     41 Bdos call:  15 (Open File) IN from 0x864c:
      1 Bdos call:  15 (Open File) IN from 0xf619:
     41 Bdos call:  16 (Close File) IN from 0x87a4:
    190 Bdos call:  20 (Read Sequential) IN from 0x8808:
    284 Bdos call:  20 (Read Sequential) IN from 0xf5dd:
      3 Bdos call:  25 (Get Current Disk) IN from 0xf53f:
      2 Bdos call:  25 (Get Current Disk) IN from 0xf568:
    190 Bdos call:  26 (Set DMA Address) IN from 0x87fb:
     82 Bdos call:  26 (Set DMA Address) IN from 0x89dc:
      2 Bdos call:  26 (Set DMA Address) IN from 0xf565:
      1 Bdos call:  26 (Set DMA Address) IN from 0xfb72:
    284 Bdos call:  26 (Set DMA Address) IN from 0xfbc9:
     51 Bdos call:   2 (Console Output) IN from 0x8c4d:
     14 Bdos call:   2 (Console Output) IN from 0xf5b5:
      2 Bdos call:  32 (Get/Set User) IN from 0xf510:
      3 Bdos call:  32 (Get/Set User) IN from 0xf547:
      4 Bdos call:  32 (Get/Set User) IN from 0xf650:
      1 Bdos call:  32 (Get/Set User) IN from 0xfb20:
      1 Bdos call:  32 (Get/Set User) IN from 0xfb94:
      1 Bios call:   3 IN:
```
...that doesn't smell like a direct sector access problem.

Why does the last line look different? No BIOS function name in parentheses and no caller address...

MockbaTheBorg, 2017-10-26:
The last BIOS call (0x3) is a call to go back to CCP (warm boot). That's where the program "dies" and goes back to the CP/M prompt.
I don't think I added naming to the BIOS calls ... I would have to check.

I have a version of ADV.COM which works on my copy of Z80Emu ver 2.0, with regular CP/M 2.2, so I am ruling out CP/M v2.2 compatibility issues too.
When I run it on RunCPM it just says "Missing variable: HERE" and goes back to the CP/M prompt.
I found it here: http://ifarchive.jmac.org/if-archive/games/cpm/cpm-advent-b03.zip

I will take a look at your log to see if anything looks "funny".

MockbaTheBorg, 2017-10-26:
I believe I have found the issue, not sure how to fix it yet.
It seems to be related to seeking/reading files bigger than 128 (0x80) sectors / file extension translation.
I am investigating, but this is definitely a bug on RunCPM. Will report back.
If anyone wants to take a look as well please do.

MockbaTheBorg, 2017-10-26:
Hi all ... problem solved.
I was misinterpreting the File Open parameters and cleaning up some FCB counters myself, when that needs to be done solely by the calling application.
Improved also the compatibility of Sequential Read and Sequential Write with the "real" CP/M 2,2.
This one took me a while to figure out ... but it was fun.
Have fun with ADV.COM ... :)

closed 2017-10-26 by MockbaTheBorg

mbramwel, 2017-10-27:
Thanks for the update!  I can confirm it works much better on my Teensy.

I am curious which "Linux" machine you are using.  I have tried Centos 7.4 and the latest debian.  Both are giving me curses lib issues.  I have to comment out LUA in the makefile to get it to compile but end up with runtime issues with apps (only cpm commands works). A small hint and I could probably get it going.

drawkula, 2017-10-27:
@mbramwel — can you please start a new issue/topic for the compiling on "Linux" problem?

kklis, 2017-10-27:
@mbramwel You need to install libncurses-dev and libreadline-dev packages prior to building RunCPM and you should be fine.