MockbaTheBorg/RunCPM #46: Comal 80 broken in Posix port

kklis, 2017-10-28:
Since commit c80f7a09b44e196f40b41b195ec6788d526a8417 Comal-80 for CP/M does not accept input on MacOS X and Linux.

How to reproduce:
- Download Comal-80 from http://www.autometer.de/unix4fun/z80pack/ftp/comal.tgz
- Extract files with from disk image with cpmtools and put in a disk directory (e.g. C).
- Start latest RunCPM. Go to disk C: and run "comal". You can't input anything in the Comal prompt.
- Revert commit c80f7a09b44e196f40b41b195ec6788d526a8417 and rebuild RunCPM.
- Start RunCPM again. Go to disk C: and run "comal". Now everything works ok.

MockbaTheBorg, 2017-10-30:
Hi all.

What is the plan here? I vaguely remember this commit was because something else was not working on the Mac?

Does removing it breaks something else?

kklis, 2017-10-30:
I think that @csoren should be able to provide a motivation which was behind the change, as an author of the original commit. I hope that he can provide a fix, too :)

csoren, 2017-10-30:
Sure, I'll take a look. The fix is for an issue where input was reported as being available when nothing is actually available to read. I suspect the new check is too restrictive.

MockbaTheBorg, 2017-10-30:
Hi guys,
I have merged Csoren's latest commit.
Please test and let me know if it is all good.
Thanks a lot!!!

drawkula, 2017-10-30:
Short: Looks ok.

Long:
```
CP/M 2.2 Emulator v2.9 by Marcelo Dantas
      Build Oct 30 2017 - 21:54:49
       with Lua scripting support
-----------------------------------------
CCP: CCP-DR.60K  CCP Address: 0xe400

RunCPM Version 2.9 (CP/M 2.2 60K)

A>h:
H>dir
H: APPLES   SAV : ASM      COM : BEST1ST  TXT : BOOT     COM
H: BOOTGEN  COM : CALENDAR SAV : CMDLIST  DOC : COMAL    COM
H: COMALANG TXT : COMALDDJ TXT : COMALYOU TXT : COPY     COM
H: CPMBOOT  COM : CPU      COM : CREF80   COM : DDT      COM
H: DDTZ     COM : DIF      COM : DO       COM : DUMP     COM
H: ED       COM : FORMAT   COM : FRUSTRAT TXT : GO       COM
H: HALT     COM : HANOI    SAV : HDIR     COM : HELLO    SAV
H: HI       SAV : INSTALL  COM : INSTALL  DAT : INSTALLD SAV
H: KEYSC128 SAV : KEYWORDS TXT : L80      COM : LIB80    COM
H: LOAD     COM : LS       COM : LU       COM : M80      COM
H: MAGICSQR SAV : NIM      SAV : NUM2WORD SAV : PHONEBK  SAV
H: PIP      COM : POLYNY   SAV : PRIMES   SAV : PRINTERR SAV
H: PRINTUSE SAV : R        COM : RSETSIMH COM : RUNTIME  COM
H: SHOWSEC  COM : SID      COM : SORTING  SAV : SPEED    COM
H: STARTED  DOC : STAT     COM : SURVEY   COM : SYSCOPY  COM
H: TIMER    COM : TRIANGLE SAV : UNCR     COM : UNERA    COM
H: USQ      COM : W        COM : WM       COM : WM       HLP
H: XFORMAT  COM : XSUB     COM : ZAP      COM : ZSID     COM
H: ZTRAN4   COM
```
That's the whole contents of AltairZ80's `comal.dsk`.
```
H>comal

         (C) Copyright 1987
  COMAL Users Group, U.S.A. Limited

load"primes
Error number 68 : File does not exist                                           
```
Comal seems to expect files still on `A:`?
Maybe that's not a bug, but a feature instead...
```
load"h:primes                                                                   
list                                                                            
0010 si#:=3962;count#:=0
0020 DIM flags#(0:si#)
0030 FOR i#:=0 TO si# DO 
0040   IF NOT flags#(i#) THEN 
0050     prime#:=i#+i#+3
0060     k#:=i#+prime#
0070     count#:+1
0080     PRINT prime#;
0090     FOR k#:=i#+prime# TO si# STEP prime# DO flags#(k#):=TRUE
0100   ENDIF 
0110 ENDFOR i#
0120 PRINT 
0130 PRINT "count =";count#
run
3 5 7 11 13 17 19 23 29 31 37 41 43 47 53 59 61 67 71 73 79 83 89 97 101 103 107
```
...lots of lines not copied...
```
 7759 7789 7793 7817 7823 7829 7841 7853 7867 7873 7877 7879 7883 7901 7907 7919 7927 
count = 1000

End of program   

_
```

kklis, 2017-10-30:
Works OK on Mac OS X. Thanks!

closed 2017-10-30 by kklis

MockbaTheBorg, 2017-10-30:
Well... On this matter I will have to just trust you guys. If you are
happy, then I am happy too... 😀