MockbaTheBorg/RunCPM #48: Hi Tech C 3.09 hangs

tsupplis, 2017-12-23:
While testing a variety of programs, I met one issue. The compiler run under runcpm hangs. It seems to be associated to the C.COM program chaining the various compilation steps. I did not go far enough to figure out what triggers the failure.
the same compiler works properly in a simh environment and a couple of other emulators.
the programs generated with it work perfectly under it.

drawkula, 2017-12-23:
Here too:
```
C>c hello.c
HI-TECH C COMPILER (CP/M-80) V3.09
Copyright (C) 1984-87 HI-TECH SOFTWARE
HI-TECH C COMPILER (CP/M-80) V3.09
Copyright (C) 1984-87 HI-TECH SOFTWARE
█
```
I noticed this some days ago but thought I had screwed up my `runcpm` install.

tsupplis, 2017-12-23:
Super, Thank's, this gives us a first regression date....we just need to inspect all the changes from then. when the error occurs, it seems that the system is indeed in the disk calls with weird FCB addresses and content...
The only relevant changes made on the 26th are touching OpenFile, WriteSeq and ReadSeq and one of the constants MaxCR.
Will test by elimination tomorrow.

tsupplis, 2017-12-23:
https://github.com/MockbaTheBorg/RunCPM/commit/a6ca0a646599da93659dde4c3d14d908b52c15d0#diff-441fac67a0c8e278dbe6999fe1474964

MockbaTheBorg, 2017-12-24:
These are the changes I made to get ADV.com working. So I need to double check and make sure they don't become mutually exclusive.
Someone please get me a link do download the same HiTech C 3.9 and I will take a look.

Thanks!

drawkula, 2017-12-24:
https://schorn.ch/altair_6.php has Hi-Tech-C in [aztechitechc.zip](https://schorn.ch/cpm/zip/aztechitechc.zip).

Hi-Tech-C from that disk image as ZIP: [hi-tech-c-v3.09.zip](https://github.com/MockbaTheBorg/RunCPM/files/1584722/hi-tech-c-v3.09.zip)

tsupplis, 2017-12-28:
ok. so reducing even further the problem. the removal of the zeroing of ex,s1,s2,cr in the OpenFile function is what is creating the issue.

MockbaTheBorg, 2017-12-29:
But I believe that was what fixed ADV.com... need to test again ... but I am almost sure.

MockbaTheBorg, 2017-12-29:
HI there ... I have re-added just the zeroing of s1 and s2. It seems to have fixed HiTech C without breaking ADV. So I will leave it like this, but won't close this issue for now.
I would like you guys to test them and also other stuff, and let me know if there are any other issues.
Then I will close here.

Merry Christmas and Happy New Year to everyone. See you in 2018 :).

tsupplis, 2017-12-29:
Super, Thank you so much. Testing it early tomorrow. will put results in.
And best wishes for the new year to you too....

tsupplis, 2017-12-29:
Preliminary test is a win... c compiler works a treat. will carry on testing tomorrow.

drawkula, 2017-12-29:
I still have some of the problems with commit a3cae38047868b124c00d5c6e37e1de6317da303 (posix variant) like already mentioned above: 
```
CP/M 2.2 Emulator v2.9 by Marcelo Dantas
      Build Dec 29 2017 - 08:51:01
       with Lua scripting support
-----------------------------------------
CCP: CCP-DR.60K  CCP Address: 0xe400

RunCPM Version 2.9 (CP/M 2.2 60K)

A>user 1
A>dir
no file
```
`user` switches but the prompt does not change.
```
A>c:
C>dir
C: CR       SUB : ECHO     C   : ECHO     COM : ENVIRON
C: MANDELBR C   : MANDELBR COM : MANDELBR SUB : MYHI     C  
C: MYHI     COM : MYHI     SUB : TE       BKP
```
This is `C1:`.
```
C>exit
EXIT?

C>submit myhi
SUBMIT?

```
`exit` and `submit` are on `A0:` and should be runnable from `C1:`.
```
C>user 0
C>c hello.c
HI-TECH C COMPILER (CP/M-80) V3.09
Copyright (C) 1984-87 HI-TECH SOFTWARE

RunCPM Version 2.9 (CP/M 2.2 60K)

```
Ok. `c.com` does not hang.
```
C>a:
A>dir exit*.*
A: EXIT     ASM : EXIT     COM : EXIT     SUB : EXIT     Z80
A>dir sub*.*
A: SUBMIT   COM : SUBMITD  COM
```
Proof that `exit` and `submit` really are on `A0:`.

MockbaTheBorg, 2017-12-29:
I believe I changed the default CCP to DR's when I was testing ADV (to make sure I was on 100% DR 2.2 compatible mode). This "strange" behavior you are seeing should be the standard (unpatched) DR 2.2 behavior.
Go to globals.h and change the CCP on line 25:
//#define CCP_DR
#define CCP_CCPZ
CCPZ has the behavior you want: Fallback to user0 on A: if the requested executable is not found locally.
(You can try the others just for fun)

closed 2017-12-29 by MockbaTheBorg