MockbaTheBorg/RunCPM #49: Dropping CCPHEAD?

drawkula, 2018-01-24:
Is the CPPHEAD message really needed that often?
Its contents does not change.
Would showing this at most once be enough?

I'll mute it in a local branch if the majority loves to see it frequently...

MockbaTheBorg, 2018-01-31:
Well ... needed it is not. I just printed it all the time the original CP/M also does, for the sake of being faithful to the "real" one.
The original CP/M 2.2 prints the CCP header every time you press ^C on the prompt, and every time you come back from most of the application programs, like MBasic or ASM.
STAT doesn't print it when coming back to the prompt, because STAT uses a simple RET to return to the prompt, and in that case I don't print it either.
Therefore it is just a matter of matching the original. I would keep it for the sake of "correctness", unless you find that it prints when it shouldn't or vice versa, which would be a bug indeed.

drawkula, 2018-01-31:
Probably I spent more time with CP/M3 in "þe olde days" and just don't remember how often CP/M2 did that.

I think it is worth to be muted because it does not give enough information to be worth wasting 2 lines on the screen that often.

I'll mute it in a local branch (yay for Git!) and everyone will be happy.

closed 2018-01-31 by drawkula