MockbaTheBorg/RunCPM #50: RunCPM SD disk problem (long/short file names?)

riwide, 2018-02-01:
Problem: With teensy 3.6, files in A/0 of sd showing duplicated, one normal short name e.g.  asm.com and one false short name _AS~.COM
Scenario:Using RunCPM files of 2018-feb-01, move contents of DISK/A.ZIP to SD under OSX (or Windows 10), and also copy DR 60K ccp to root of SD; take fresh Arduino 1.85  IDE for OSX, install teensyduino, open RunCPM.ino from unzip of github runcpm files. Build and upload Runcpm to teensy 3.6 with above SD card inserted. Terminal shows RunCPM running ok except no files visible on A; extract SD card to PC to move "A" files into folder A/0; re-insert SD card in teensy 3.6 and reboot; RunCPM working OK except see all files in folder A/0 ok and also see all files with a newly created short name, e.g. ASM.COM visible as ASM.COM and _AS~1.COM. This looks like some sort of long/short naming problem. 
Note 1) this error was visible also on windows10 hosted teensy3.6;
Note 2) on some occasions the error may show as only showing first file on disk folder A/0 with normal name, the other files wouldn't be visible. (on this occasion after teensy reboot to get screen shot, I only saw this flavour of error, could not get screen shot of error showing as full directory with duplicates.
Note 3) SD card is 4gb type 4 micro sd hc sandisk formatted as FAT 16

riwide, 2018-02-01:
Problem: With teensy 3.6, files in A/0 of sd showing duplicated, one normal short name e.g.  asm.com and one false short name _AS~.COM
Scenario:Using RunCPM files of 2018-feb-01, move contents of DISK/A.ZIP to SD under OSX (or Windows 10), and also copy DR 60K ccp to root of SD; take fresh Arduino 1.85  IDE for OSX, install teensyduino, open RunCPM.ino from unzip of github runcpm files. Build and upload Runcpm to teensy 3.6 with above SD card inserted. Terminal shows RunCPM running ok except no files visible on A; extract SD card to PC to move "A" files into folder A/0; re-insert SD card in teensy 3.6 and reboot; RunCPM working OK except see all files in folder A/0 ok and also see all files with a newly created short name, e.g. ASM.COM visible as ASM.COM and _AS~1.COM. This looks like some sort of long/short naming problem. 
Note 1) this error was visible also on windows10 hosted teensy3.6;
Note 2) on some occasions the error may show as only showing first file on disk folder A/0 with normal name, the other files wouldn't be visible. (on this occasion after teensy reboot to get screen shot, I only saw this flavour of error, could not get screen shot of error showing as full directory with duplicates.
Note 3) SD card is 4gb type 4 micro sd hc sandisk formatted as FAT 16
Note 4) with scenario on note 2 above , i found on subsequent testing that dir shows 1 file, but pip b:=a:*.* will find all files with proper name and all the false duplicates with _AS~.COM style of name

MockbaTheBorg, 2018-02-01:
I duplicated the tildes above so they don't become a strikeout of the text ... but there's only one tilde on the file name,

MockbaTheBorg, 2018-02-02:
I have done a lot of tests here last night trying to break it on the Teensy, all unsuccessful.
It has to be something related to the SD library, but I cannot find a way to replicate.
I would like to know (unfortunately) what would be the result of buildind it all from scratch on a Windows machine, if you have one.
Following these steps:
1. Make sure that Documents\\Arduino\\libraries is empty.
2. Reinstall Arduino (I have 1.8.5 here, but shouldn't matter if a newer one is used).
3. Reinstall Teensyduino (I have 1.41 here).
4. Copy the Teensy SD Library to Documents\\Arduino\\library to make sure it supersedes the Arduino one.
5. Rebuild RunCPM from scratch.

I have to ask for this, but it would at least certify that all the hardware is good, and the issue is contained to the Arduino on OSX environment.

Unfortunately I have only trial-and-error to work on at the moment. But we will get to the bottom of this.

riwide, 2018-02-02:
Will do more testing on windows, also send you,upload, a zip of my sd
image. Be over next few days

MockbaTheBorg, 2018-02-04:
Hi Richard, I have found the root cause of the issue.
By looking at the debug log file you sent me I see that the terminal emulator used to communicate with the Teensy is sending a CRLF sequence when you press the Enter key.
My (very simplistic) character reading routine closes the current line either on receiving CR or LF, just like CP/M 2.2 does, so in your case the CR closes the line and the LF goes into the keyboard buffer.
DR's CCP checks for a character waiting in the keyboard buffer after every file is printed by the DIR command, so after printing ASM.COM it checks the buffer, finds the LF there and terminates the DIR command.
This is working as designed.
The solution is to make sure the terminal emulator sends only one character when you press the Enter key. Can be either CR or LF, either will work to close the current command line.
This should solve the issue and let you print full DIR listings when using the original DR CCP.

closed 2018-02-04 by MockbaTheBorg

riwide, 2018-02-05:
Thanks, checked it works for me, it does. Puzzled how I ever had the
symptom of each file twice, one short name and one long, for all files.