MockbaTheBorg/RunCPM #51: Unable to load CP/M CCP. CPU halted.

leonardoelias300, 2018-04-12:
Hello. I'm trying to run it on a teensy 3.5 but without success.
I change the SD type, A folder, 0 subfolder... there's something I'm missing?

MockbaTheBorg, 2018-04-12:
Hi Leonardo,

To run on the Teensy you have to:

1. Format the SD card as FAT32
2. Create a "A" (uppercase, to be sure) folder on that SD card
3. Create a "0" (zero) folder inside that "A" folder
4. Copy contents of the distribuition disk (A.zip) inside that folder (optional)
5. Copy the CCP file that you intend to use onto the SD root folder (you can copy them all)

This should do it.
Please let me know if you have tried this.

Thanks,
Marcelo.

MockbaTheBorg, 2018-04-12:
This is how my folder looks like:

[image]

leonardoelias300, 2018-04-12:
Thank you! Now its working fine... I forget to copy the CCP files to root folder! Now I'll explore!!! :)

closed 2018-04-12 by leonardoelias300