MockbaTheBorg/RunCPM #52: May need to add ENABLE_VIRTUAL_TERMINAL_PROCESSING

drobri, 2018-05-22:
Hello, I tried runcpm.exe and it runs under Windows 7 except for processing of ANSI terminal escape sequences.  Upon investigation it seems there is a flag that must be set for the console host to filter and process the escape sequences ( ENABLE_VIRTUAL_TERMINAL_PROCESSING).  I even tried the another method with a program called "ansicon.exe" and it did not work.  I discovered this issue trying the TE.COM editor and TURBO Pascal for CPM-80.  The ANSI escape sequences just get echoed to the console.  I think this flag can be added to the function void _console_init(void)  in abstraction_vstudio.h
Here is the Microsoft Reference URL on Console Virtual Terminal Sequences:  https://docs.microsoft.com/en-us/windows/console/console-virtual-terminal-sequences

drobri, 2018-05-28:
I have been testing under Windows 7 and the ENABLE_VIRTUAL_TERMINAL_PROCESSING flag is not implemented.  It seem this is only a Windows 10 supported item.

For whatever reason I was unable to get the open source ansicon program to set the ANSI processing filter for RunCPM.exe.  https://github.com/adoxa/ansicon

Also the ConEmu program works very well with RunCPM and I was able to get the TE editor and TURBO Pascal to work full screen. https://conemu.github.io/

I think I may have not enabled ansicon correctly and will try and see if I can get it working with RunCPM.exe

closed 2018-05-28 by drobri

MockbaTheBorg, 2018-05-29:
Awesome! ... sorry for the long delay as I was travelling.
I am running Windows 10, and by default it is translating ANSI sequences. It always did.
Long time ago when I started developing RunCPM I also used W7, and I think ConEMU was the one I was using.
Great to know that you were able to get it running. Good news!