MockbaTheBorg/RunCPM #53: Latest Windows 10 upgrade, 1803, breaks TE.COM

MockbaTheBorg, 2018-05-30:
Hi All,

I am noticing something strange here, maybe you have seen the same and may be able to help.
I have tried many different things with no results. Here is the issue:
After I had to reinstall Windows 10, and Visual Studio 2017, when I built RunCPM it won't accept ESC-X to exit TE.COM (distributed on disk A:).
ESC-N and ESC-A seem to work, but not ESC-X.
Maybe I drew a blank, maybe I am not seeing something obvious, but if one of you has seen the same, let me know. I want to find out what is causing it.

Thanks,
Marcelo.

MiguelVis, 2018-05-30:
That part of the ui is very simple in TE:

```
switch(toupper(getchr()))
{
	case 'N'   : run = MenuNew(); break;
	case 'O'   : run = MenuOpen(); break;
	case 'S'   : run = MenuSave(); break;
	case 'A'   : run = MenuSaveAs(); break;
	case 'B'   : MenuAbout(); ++menu; break;
	case 'H'   : MenuHelp(); ++menu; break;
	case 'X'   : run = stay = MenuExit(); break;
	case K_ESC : run = 0; break;
	default    : ask = 0; break;
}
```

I think you should debug the input / status keyboard routines in RunCPM to see if there is something wrong related to the last Windows 10 update.

By the way, TE has been already improved in recent versions.

May be is worth for you to check the last version in my github [repository] https://github.com/MiguelVis/RetroProjects)

MockbaTheBorg, 2018-05-30:
Found the issue ... a call to _getch(), which is the "conin" BIOS call for CP/M, somehow returns zero.
So when I press "X", TE correctly shows "Confirm Y/N:" and waits for keypress, that keypress comes back \0, and it goes back to the menu.
It was happening so fast that I was not able to see it.
Now I am trying to figure out why the extra \0, so I can fix it.

MiguelVis, 2018-05-30:
Check if _kbhit() returns true before you call _getch() when it returns '\0'. May be you get a clue if you do that. Good luck!

MockbaTheBorg, 2018-05-30:
_kbkit() was returning a valid waiting character for the \0  ... I have applied an ugly fix (workaround) for now. Will research more on it as I have time.
Closing for now.

closed 2018-05-30 by MockbaTheBorg