MockbaTheBorg/RunCPM #54: RunCMP on Arduino

pavelzampach, 2018-06-03:
It's great! Return to good old times of Speccy/80KB + home made disk drive.
As Win terminal seems very good Kitty (http://www.9bis.net/kitty/) with key remapping (PageUp/Down, arrows... can use almost standard WordStar command for cursor moving).
As CP/M text editor is ZDE (https://sites.google.com/site/vdeeditor/Home/vde-files) recommanded - very good stuff.
Once more - thanks for RunCMP

MockbaTheBorg, 2018-06-05:
Hi Pavel,

Thanks for giving RunCPM a try. Hope you have lots of fun with it.
Feel free to try many programs on it, and let me know if anything breaks.

Thanks,
Marcelo.

closed 2018-06-05 by MockbaTheBorg

pavelzampach, 2018-07-04:
Hi Marcelo,

one question and one small note to RunCPM:

* copying and renaming files is slow (14K file is copyed about 2 min), is
it only my problem (bad formatted SD catd etc.)?

* on address http://www.chstercius.cz/runcpm are some tips and original
programs

Best regards
Pavel Zampach

kklis, 2018-07-08:
Hi Pavel,
I don't think it's your SD card. I just tested copying a 60kB file and it took about 3 minutes on Arduino Due and about a minute on Teensy 3.6. I suspect that the problem is in the CP/M routines which copy data byte by byte and as far as I know don't use any buffering (Marcelo can probably correct me here if I'm wrong). On a regular PC the problem cannot be spotted because the emulation is a lot faster and disk operations take fractions of seconds.