MockbaTheBorg/RunCPM #58: [FAQ?] Graphics? Yes we can!

drawkula, 2018-08-13:
Well... there is no standard for graphics with CP/M but there are [SIXELs](https://en.wikipedia.org/wiki/Sixel)!

 And because this kind of graphics is just like printing with a dot matrix printer, CP/M programs easily can do it if connected to a terminal capable of sixel graphics:

[image]

Above snapshot was taken from RunCPM in an XTerm on Debian9. There are more sixel enabled terminals and it should work similar, if you connect RunCPM on an Arduino-DUE to one of them.

These files...
```
(yeti@kumari:16)/home/wrk/runcpm/run/C/1$ ls -l MB6EL.*
-rw-r--r-- 1 yeti yeti  896 Aug 13 16:41 MB6EL.C
-rw-r--r-- 1 yeti yeti 3712 Aug 13 16:41 MB6EL.COM
-rw-r--r-- 1 yeti yeti  128 Aug 13 16:13 MB6EL.SUB
```
...are attached as: [mb6el.zip](https://github.com/MockbaTheBorg/RunCPM/files/2283914/mb6el.zip)

MockbaTheBorg, 2018-08-17:
Damn this is cool !!!
I never thought RunCPM could do it ... :)

closed 2018-08-17 by MockbaTheBorg

drawkula, 2018-08-17:
Sixel graphics behaves like a colour dot matrix printer but I did not use use colour to keep the example simple.

The Mandelbrot image is calculated in fixed point arithmetic to get an acceptable speed. With a maximum resolution of 1/4092 that sure is not able to yield deep zooms.

But I think it is ok as example and for wetting the appetite for more experiments.
:-)

MiguelVis, 2018-08-20:
@drawkula wrote:

> Well... there is no standard for graphics with CP/M

Oh, yes, we already have a standard.

Its name is [GSX](https://www.seasip.info/Cpm/gsx.html) (Graphics System eXtension) from Digital Research.

See also some [GSX Screen Shots](http://toastytech.com/guis/gsx.html) here.