MockbaTheBorg/RunCPM #61: Cannot compile

wx4cb, 2018-09-07:
Trying to compile latest download (as of today) and i get an overflow for the RAM array.

Not sure where it's coming from I've tried compiling for the uno (the actual target) but also for the mega and the due and they all get the same thing

D:\Arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\Arduino\hardware -hardware C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages -tools D:\Arduino\tools-builder -tools D:\Arduino\hardware\tools\avr -tools C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages -built-in-libraries D:\Arduino\libraries -libraries C:\Users\Charles Blackburn\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10804 -build-path C:\Users\CHARLE~1\AppData\Local\Temp\arduino_build_416580 -warnings=none -build-cache C:\Users\CHARLE~1\AppData\Local\Temp\arduino_cache_784956 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.1.1 -prefs=runtime.tools.avrdude.path=C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -prefs=runtime.tools.avr-gcc.path=C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -verbose C:\Users\Charles Blackburn\Documents\Arduino\RunCPM\RunCPM\RunCPM.ino
D:\Arduino\arduino-builder -compile -logger=machine -hardware D:\Arduino\hardware -hardware C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages -tools D:\Arduino\tools-builder -tools D:\Arduino\hardware\tools\avr -tools C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages -built-in-libraries D:\Arduino\libraries -libraries C:\Users\Charles Blackburn\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10804 -build-path C:\Users\CHARLE~1\AppData\Local\Temp\arduino_build_416580 -warnings=none -build-cache C:\Users\CHARLE~1\AppData\Local\Temp\arduino_cache_784956 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.1.1 -prefs=runtime.tools.avrdude.path=C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -prefs=runtime.tools.avr-gcc.path=C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -verbose C:\Users\Charles Blackburn\Documents\Arduino\RunCPM\RunCPM\RunCPM.ino
Using board 'uno' from platform in folder: C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21
Using core 'arduino' from platform in folder: C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21
Detecting libraries used...
"C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino" "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\variants\standard" "C:\Users\CHARLE~1\AppData\Local\Temp\arduino_build_416580\sketch\RunCPM.ino.cpp" -o "nul"
"C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino" "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\variants\standard" "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\libraries\SPI\src" "C:\Users\CHARLE~1\AppData\Local\Temp\arduino_build_416580\sketch\RunCPM.ino.cpp" -o "nul"
"C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino" "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\variants\standard" "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\libraries\SPI\src" "-IC:\Users\Charles Blackburn\Documents\Arduino\libraries\SD\src" "C:\Users\CHARLE~1\AppData\Local\Temp\arduino_build_416580\sketch\RunCPM.ino.cpp" -o "nul"
Using cached library dependencies for file: C:\Users\CHARLE~1\AppData\Local\Temp\arduino_build_416580\sketch\main.c
"C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino" "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\variants\standard" "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\libraries\SPI\src" "-IC:\Users\Charles Blackburn\Documents\Arduino\libraries\SD\src" "C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\libraries\SPI\src\SPI.cpp" -o "nul"
"C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino" "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\variants\standard" "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\libraries\SPI\src" "-IC:\Users\Charles Blackburn\Documents\Arduino\libraries\SD\src" "C:\Users\Charles Blackburn\Documents\Arduino\libraries\SD\src\File.cpp" -o "nul"
"C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino" "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\variants\standard" "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\libraries\SPI\src" "-IC:\Users\Charles Blackburn\Documents\Arduino\libraries\SD\src" "C:\Users\Charles Blackburn\Documents\Arduino\libraries\SD\src\SD.cpp" -o "nul"
"C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino" "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\variants\standard" "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\libraries\SPI\src" "-IC:\Users\Charles Blackburn\Documents\Arduino\libraries\SD\src" "C:\Users\Charles Blackburn\Documents\Arduino\libraries\SD\src\utility\Sd2Card.cpp" -o "nul"
[... cut ...]
cpu.h:14: error: expected ')' before 'volatile'

Multiple libraries were found for "SD.h"
 Used: C:\Users\Charles Blackburn\Documents\Arduino\libraries\SD
 Not used: D:\Arduino\libraries\old_SD
 Not used: D:\Arduino\libraries\SD
Using library SPI at version 1.0 in folder: C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\libraries\SPI 
Using library SD at version 1.2.2 in folder: C:\Users\Charles Blackburn\Documents\Arduino\libraries\SD 
exit status 1
overflow in constant expression

drawkula, 2018-09-07:
> Not sure where it's coming from I've tried compiling for the uno (the actual target) but also for the mega and the due and they all get the same thing

Arduino-Uno and -Mega aren't able to run RunCPM.

wx4cb, 2018-09-07:
that still doesnt explain why i get the same thing compiling for the duemelinove 

D:\Arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\Arduino\hardware -hardware C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages -tools D:\Arduino\tools-builder -tools D:\Arduino\hardware\tools\avr -tools C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages -built-in-libraries D:\Arduino\libraries -libraries C:\Users\Charles Blackburn\Documents\Arduino\libraries -fqbn=arduino:avr:diecimila:cpu=atmega328 -ide-version=10804 -build-path C:\Users\CHARLE~1\AppData\Local\Temp\arduino_build_416580 -warnings=none -build-cache C:\Users\CHARLE~1\AppData\Local\Temp\arduino_cache_784956 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.1.1 -prefs=runtime.tools.avrdude.path=C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -prefs=runtime.tools.avr-gcc.path=C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -verbose C:\Users\Charles Blackburn\Documents\Arduino\RunCPM\RunCPM\RunCPM.ino
D:\Arduino\arduino-builder -compile -logger=machine -hardware D:\Arduino\hardware -hardware C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages -tools D:\Arduino\tools-builder -tools D:\Arduino\hardware\tools\avr -tools C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages -built-in-libraries D:\Arduino\libraries -libraries C:\Users\Charles Blackburn\Documents\Arduino\libraries -fqbn=arduino:avr:diecimila:cpu=atmega328 -ide-version=10804 -build-path C:\Users\CHARLE~1\AppData\Local\Temp\arduino_build_416580 -warnings=none -build-cache C:\Users\CHARLE~1\AppData\Local\Temp\arduino_cache_784956 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.1.1 -prefs=runtime.tools.avrdude.path=C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -prefs=runtime.tools.avr-gcc.path=C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -verbose C:\Users\Charles Blackburn\Documents\Arduino\RunCPM\RunCPM\RunCPM.ino
Using board 'diecimila' from platform in folder: C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21
Using core 'arduino' from platform in folder: C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21
Detecting libraries used...
"C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_DUEMILANOVE -DARDUINO_ARCH_AVR   "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino" "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\variants\standard" "C:\Users\CHARLE~1\AppData\Local\Temp\arduino_build_416580\sketch\RunCPM.ino.cpp" -o "nul"
"C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_DUEMILANOVE -DARDUINO_ARCH_AVR   "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino" "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\variants\standard" "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\libraries\SPI\src" "C:\Users\CHARLE~1\AppData\Local\Temp\arduino_build_416580\sketch\RunCPM.ino.cpp" -o "nul"
"C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_DUEMILANOVE -DARDUINO_ARCH_AVR   "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino" "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\variants\standard" "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\libraries\SPI\src" "-IC:\Users\Charles Blackburn\Documents\Arduino\libraries\SD\src" "C:\Users\CHARLE~1\AppData\Local\Temp\arduino_build_416580\sketch\RunCPM.ino.cpp" -o "nul"
Using cached library dependencies for file: C:\Users\CHARLE~1\AppData\Local\Temp\arduino_build_416580\sketch\main.c
"C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_DUEMILANOVE -DARDUINO_ARCH_AVR   "-IC:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\a [... cut ...]
cpu.h:14: error: expected ')' before 'volatile'

Multiple libraries were found for "SD.h"
 Used: C:\Users\Charles Blackburn\Documents\Arduino\libraries\SD
 Not used: D:\Arduino\libraries\old_SD
 Not used: D:\Arduino\libraries\SD
Using library SPI at version 1.0 in folder: C:\Users\Charles Blackburn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\libraries\SPI 
Using library SD at version 1.2.2 in folder: C:\Users\Charles Blackburn\Documents\Arduino\libraries\SD 
exit status 1
overflow in constant expression

drawkula, 2018-09-07:
> that still doesnt explain why i get the same thing compiling for the duemelinove

Sure it does!
The Arduino-Duemilanove is nearly an Uno but with an older and bigger bootloader.
The Duemilanove is not the Due!

—▷ [Arduino Hardware Overview](https://www.arduino.cc/en/Main/Products?from=Main.Hardware)
—▷ [Board Secs](https://www.arduino.cc/en/Products/Compare)

wx4cb, 2018-09-07:
[image]

So which one is it in here then? cus I can't find it, and i looked in the board manager and can't find it

drawkula, 2018-09-07:
[image]

MockbaTheBorg, 2018-09-09:
Hi wx4cb,

Unfortunately, to emulate a 64K RAM CP/M machine I need a chip which has that much RAM available solely to hold RunCPM's memory array, plus more, to contain the variables required to run RunCPM itself. And also enough Flash to contain the code.
As Drawkula explained, any board based on the ATmega168(1KB) or the ATmega328 (2KB) will not have near enough memory to run any of it.
Maybe you would be able to run it on an Atmel based board if you modified RunCPM's code and implemented extra external memory onto these boards. I did a test once with serial memory, but it was absurdly slow and unfortunately I lost it when I lost a laptop.
The best small architecture to run RunCPM is the ARM, it is fast and (usually) has a lot of RAM.
You can go with the Arduino DUE (not the duemilanove) or a Teensy (the 3.5 or a 3.6).
I would dare say that a chip to run RunCPM would require at least 128K SRAM and 128K Flash on it. Though I cannot still guarantee any chip will do.
So far we have got it to run only on the Due or on the Teensy (other than some crazy chips I tested here which end up expensive and so complicated that I won't dare release into the public. Believe-me, you don't want to run RunCPM on a US$500 FPGA board when a $28 Teensy gives you even better performance. :)
If you want to try it on an ARM based processor I would recomment the Teensy 3.5, it is absurdly fast, absurdly small and still very cheap.
Hope this helps.
Let us know if you give it a try or need any help doing it.

Cheers,
MockbaTheBorg.

closed 2018-09-09 by MockbaTheBorg

Nurb432, 2018-09-09:
For what its worth, while a larger format, the Chinese DUE clone boards 
are even cheaper than that ( ~ 16 USD. ).  ( its what i have running 
RunCPM )  Tho i did have to add a SD module, which i already had from 
another project..