MockbaTheBorg/RunCPM #62: support for esp8266?

aldolo69, 2018-09-14:
will you ever try to compile for esp cpu?

drawkula, 2018-09-14:
Loosely related: https://github.com/SmallRoomLabs/cpm8266

closed 2018-09-14 by aldolo69

MockbaTheBorg, 2018-09-22:
A quick look at the Sparkfun ESP32 Thing shows a lot of SRAM and flash. Being it programmed with the Arduino IDE, I think there's no big porting to do, just figure out how to connect the SD card.
This one already has a TF Card slot on it. Should be an easy fit.

https://www.ebay.com/itm/1pcs-ESP32-WROVER-4MB-PSRAM-TF-CARD-WiFi-Module-Bluetooth-development-board/183401249438

BTW: I just ordered one ... :)

drawkula, 2018-09-22:
I've ordered [boards based on ESP32s without the PSRAM](https://www.ebay.de/itm/132767536480) a week ago. The chinese racing camels will take a while to deliver them to me.

RunCPM should fit in the board without the PSRAM good enough and attaching a SD card socket probably will need no resistors or level shifters [like on the ESP8266](https://hive-project.de/board/download/file.php?id=3508&mode=view).

aldolo69, 2018-09-26:
in my mind sd card access will be via a transfer program. is it possible to map an sdcard directly as a cp/m disk? maybe it could, but will be the sdcard usable outside the emulator? I'm planning to use https://github.com/Jose-Barreto/SuperDir as a starting point for the transfer program.

drawkula, 2018-09-26:
> in my mind sd card access will be via a transfer program. is it possible to map an sdcard directly as a cp/m disk?

—▷ https://github.com/MockbaTheBorg/RunCPM#getting-started

drawkula, 2018-09-27:
What about opening an ESP32-port issue before the discussion of the new port gets into the busy phase?

drawkula, 2019-01-09:
Quoting @MockbaTheBorg:
 
> https://www.ebay.com/itm/1pcs-ESP32-WROVER-4MB-PSRAM-TF-CARD-WiFi-Module-Bluetooth-development-board/183401249438

I got 2 * 2 Boards looking like these from 2 different orders from the same seller.

When the 1st 2 arrived (they have no branding, logo or such) I thought they were [LOLIN32 Pro](https://web.archive.org/web/20181026145939/https://wiki.wemos.cc/products:lolin32:lolin32_pro) boards or clones of them but they lack the 2 LEDs near to the battery connector.

The 2nd order gave me 2 identically looking boards with a TTGO label on the bottom side but so far I was not able to find them in the wild wild web.

Which subtype of TTGO boards are they?
Which pin drives the LED?