MockbaTheBorg/RunCPM #64: DIR confusion...

drawkula, 2018-09-25:
```
CP/M 2.2 Emulator v3.1 by Marcelo Dantas
      Build Sep 24 2018 - 19:14:00
       with Lua scripting support
-----------------------------------------
CCP: INTERNAL v1.4  CCP Address: 0xf400

A0>e:
E0>dir
A: VIDATT   Z80 : WS       COM : WS       OVR : WSCHANGE COM
A: WSCHANGE OVR : WSCHHELP OVR : WSHELP   OVR : WSMSGS   OVR
A: WSPRINT  OVR : WSPRINT  TST : WSREADME BAK : WSREADME TXT
A: WSSHORT  OVR : WSU      COM
E0>█
```
The drive E directory listing lines should be prefixed with `E:`.

I found screenlogs where other CCPs did it how I expected it to be e.g. in: https://github.com/MockbaTheBorg/RunCPM/issues/48#issuecomment-353753914

MockbaTheBorg, 2018-09-27:
This is definitely a bug, and I can replicate it here.
The internal CCP is buggy for sure, I would not recommend it for "production".
I will take a look though and try to find the issue.

MockbaTheBorg, 2018-09-27:
This one should be fixed now.
Please test the latest commit.

drawkula, 2018-09-27:
```
CP/M 2.2 Emulator v3.1 by Marcelo Dantas
      Build Sep 27 2018 - 16:14:32
       with Lua scripting support
-----------------------------------------
CCP: INTERNAL v1.4  CCP Address: 0xf400

A0>c:
C0>user 1
C1>dir
C: CR       SUB : ECHO     C   : ECHO     COM : ENVIRON     
C: MANDELBR C   : MANDELBR COM : MANDELBR SUB : MB6EL    C  
C: MB6EL    COM : MB6EL    SUB : MB6EL    ZIP : MYHI     C  
C: MYHI     COM : MYHI     SUB : TE       BKP
C1>e:
E1>user 0
E0>dir
E: VIDATT   Z80 : WS       COM : WS       OVR : WSCHANGE COM
E: WSCHANGE OVR : WSCHHELP OVR : WSHELP   OVR : WSMSGS   OVR
E: WSPRINT  OVR : WSPRINT  TST : WSREADME BAK : WSREADME TXT
E: WSSHORT  OVR : WSU      COM
E0>█
```
...looks fixed now.
Thanks!

closed 2018-09-27 by drawkula