MockbaTheBorg/RunCPM #65: `user S`? Only a HiTech-C hiccup?

drawkula, 2018-09-25:
I've HiTech-C installed in `C/0` and keep my C workspace in `C/1`.
The location is defined in `C/1/ENVIRON`:
```
C1>type environ
HITECH=0:C:
```
Compiling stuff is done with simple `*.sub` files like:
```
C1>type myhi.sub
; create myhi.com
c myhi.c
myhi
;
```
Watching the drive/user structure with `watch 'ls -d ?/?'` in an own terminal window while compiling stuff shows, that the directory `C/S` is created while compiling...
```
C1>c myhi.c
HI-TECH C COMPILER (CP/M-80) V3.09
Copyright (C) 1984-87 HI-TECH SOFTWARE

C1>myhi
No meltdown, no specte.
No rowhammer, no brainslammer!

CP/M!
;-)

C1>█
```
...and not while running the result. This was typed manually (no `submit` files were harmed for this example!) so `submit` does not seem to be related to the appearing of the `C/S` directory. I'm currently using the internal CCP but I remember having seen `C/S` while using CCPZ too.

Is HiTech-C the only software creating strange drive/user directories?
Does this happen only on RunCPM's posix port?
Please have a look at your drive/user trees.

So far it looks like having no other side effects than generating this strange user area directory. I can live with that but if it shows up in other contexts too, narrowing it down may become easier.

MockbaTheBorg, 2018-09-27:
Yeah ... I have seen this. Looks like some software would attempt to access user area 28 it seems, maybe to detect the type of CP/M running? 
I have seen user area 'S' appear on my drives for some reason, but to be honest, it did not really bother me much. I had it added to my "won't be fixed" list.
Unless, of course, it becomes a problem to someone.

drawkula, 2018-09-27:
Just good to know that this is not PEBCAK... \o/

closed 2018-09-27 by drawkula

MockbaTheBorg, 2018-09-27:
LOL ... when I went to look at the code I found this:

```
void _SetUser(uint8 user) {
	userCode = user & 0x1f;	// BDOS unoficially allows user areas 0-31
							// this may create folders from G-V if this function is called from an user program
							// It is an unwanted behavior, but kept as BDOS does it
#ifdef USER_SUPPORT
	_MakeUserDir();			// Creates the user dir (0-F[G-V]) if needed
#endif
}

```