MockbaTheBorg/RunCPM #66: ESP32 Port

MockbaTheBorg, 2018-10-02:
My ESP32 boards will be arriving in a few days.
Let's see if this is going to work right out of the box, or if it will require a lot of coding to happen.
Or even if it will not be at all possible.
Stay tuned.

MockbaTheBorg, 2018-10-08:
Now off to extensive testing ... the thing seems to not have (at least with no code changes) support to analogWrite.
Also, it runs at 80MHz, a bit slow, but I saw somewhere it can be heavily overclocked, though I have no idea how to do it yet.
Fun times lie ahead.

drawkula, 2018-10-08:
AnalogWrite still is broken in this Arduino core.

https://github.com/espressif/arduino-esp32#development-status mentiones:

> Most of the framework is implemented. Most noticable is the missing analogWrite.

And PSRAM is said to be slower than the internal RAM. Can it be disabled?

MockbaTheBorg, 2018-10-08:
mySD went inside Documents/Arduino/mySD folder

Looks like it is always running at 240MHz? I have a feeling that it runs slower than the Teensy.
Gotta run more tests.
The DUE, poor DUE, got left behind.

MockbaTheBorg, 2018-10-08:
Now ... I think we should implement BDOS calls to make use of the other functionalities of the ESP32, like BlueTooth, WiFi and Real Time Clock.

drawkula, 2018-10-08:
I've no idea where `mySD.h` hides. This line should find it indepently of upper/lower-case usage. No output, no file... :-(
```
(yeti@kumari:25)/home/wrk/runcpm/src/RunCPM$ find . -iname mysd.h
```

The good news: I got a refund for the wrongly delivered variant of the ESP32 board and don't need to send it back. Maybe I'll find time to test it tomorrow. If I get `mySD.h` from somewhere... :-)

MockbaTheBorg, 2018-10-08:
Cool ... keep me posted.
I have added some website references to RunCPM's readme.

drawkula, 2018-10-09:
The Board is labeled `ESP32 DEVKITV1`.
There is no vendor name like NodeMCU, DOIT, Lolin or such.

[image]

It has the 2 rows with 15 pins pinout and uses an ESP-WROOM-32 (so no PSRAM).

[image]

Unluckily it only leaves 1 row of holes on a breadboard so I wired a fullsize SD card socket directly to it using F/F jumper wires.

[image]

It is only a socket, no resistors or other stuff is included.

[image]

I'll rethink that when the next types of boards arrive.

The first in action snapshot using a "Hackaday amber" XTerm:
 
[image]

MockbaTheBorg, 2018-10-09:
I am starting to think about a new BDOS call (243 I think) for supporting the ESP32.
We would build a record on 0x80 with more information about what the calls does, it can be wifi, BT, RTC, then call BDOS 243 to make it happen. So it would be a single BDOS entry for ESP32 support.

drawkula, 2018-10-09:
For some stuff, the buffer at 0x80 will not be large enough (e.g. when doing something with longer URLs) but then a pointer to a string could be dropped there instead of the whole URL.
A way to get software directly from the net would be like magic... \o/
Being able to save/backup stuff via WLAN would be nice too!

----

But now for somethig completely different:

Are you planning to mention RunCPM/ESP32 in [/r/ESP32](https://www.reddit.com/r/ESP32), [/r/CPM](https://www.reddit.com/r/CPM), Hackaday, ...

New users may have new ideas too... :-D

MockbaTheBorg, 2018-10-09:
My idea is that the 0x80 buffer would be only for passing one byte with the "sub-call" ID and multiple pointers to the data to be sent/received. Then a C program could allocate whatever memory and go from there.
And yes, definitely a "DOWNLOAD.COM", which we could call to download files from the network.
The possibilities are infinite (which is a number around 65536).

MockbaTheBorg, 2018-10-09:
Now, regarding social network ... lol ... I suck at these ... I avoid going to forums and suck to promote the software because I would then have no time to follow up.
RunCPM in on hackaday, I go there maybe every 6 months only. 😄

drawkula, 2018-10-10:
> My idea is that the 0x80 buffer would be only for passing one byte with the "sub-call" ID and multiple pointers to the data to be sent/received.

Ok.

> The possibilities are infinite (which is a number around 65536).

:-)

drawkula, 2018-10-11:
```
$ git branch
* local-ccpz-64k-115200bd
  master
$ git log --max-count=1
commit d0ae20704ad23837e25f9618287cfbe11022a70f
Author: yes! :-P
Date:   Thu Oct 11 12:42:51 2018 +0000

    My local changes...
    
    * RunCPM/RunCPM.ino
      ** console speed 115200Bd
    * RunCPM/globals.h
      ** use CCPZ
      ** silence CCPHEAD mesage
      ** use 64k RAM
```
The following tests were run on the `ESP32 DEVKITV1` board connected to `screen` (with SIXEL support) in an `xterm` on Debian9/AMD64.

So far I see no trouble with ...
- [X] WordStar4
- [X] Hi-Tech C - fixed point arithmetic Mandelbrot Set program with ASCII output
- [X] Hi-Tech C - fixed point arithmetic Mandelbrot Set program with SIXEL output
- [X] [ACK](https://github.com/davidgiven/ack) C - fixed point Mandelbrot Set program cross compiled for CP/M-8080 with `ack` on Linux.
- [X] [ACK](https://github.com/davidgiven/ack) C - floating point Mandelbrot Set program cross compiled for CP/M-8080 with `ack` on Linux.
- [X] Complete rebuild of [[Peter Schorn]'s SPL compiler](https://schorn.ch/cpm/zip/spl.zip) from [AltairZ80's Programming Languages download page](https://schorn.ch/altair_6.php).
  The rebuild was run on the ESP32 using the binaries of a previous rebuild done on RunCPM/Posix copied over to RunCPM/ESP32's SD card.
  The freshly rebuilt SPL then is used to build a lot of SPL programs.
  So this is an epic opera of `M80` and friends featuring SPL in Z80 major. :notes:

Bonus material included...
- [X] [Deceleration!](https://www.xkcd.com/303/)
  I'm sooo spoiled by the speed of RunCPM/Posix on my notebook...
  This puts the [Ommmmmmm...](http://pixdaus.com/files/items/pics/6/10/278610_204039a138c28d58538dfc33237d3ed4_large.jpg) back into computing... :wink:

MockbaTheBorg, 2018-10-12:
Ran a few side-by-side tests here between a Teensy 3.5 (overclocked to 168MHz) and the ESP32.

1 - The serial being 115200, on the ESP32 it still seems to be much slower than on the Teensy.
2 - The disk on the Teensy looks much faster, applications load faster on the Teensy.
3 - Then on running ZEXALL, the ESP32 goes faster.

So I am concluding: 
The Teensy is good for disk intense apps with many user interactions and screen changing.
The ESP32 is good for processing horsepower.

At least this is what it looks like so far.

drawkula, 2018-10-14:
I'm still waiting for more ESP32 boards to arrive. When playing with them and looking at the SD card speeds (of e.g. MicroPython, uLisp, ...) maybe it turns put, that all of them are slower than expected. If not, a closer look at the differences may help.

drawkula, 2018-10-15:
My fullsize SD breakout is nice because it has both sets of names of the SD card's signals. The microSD breakout I'm using now only has the SDIO names, so a table like this may be helpful in the documentation or source not only for me... ;-)
```
SPI     ESP32   SDIO
------------------------
cs      13      dat3
mosi    15      cmd
miso    2       dat0
sck     14      clk
```

MockbaTheBorg, 2018-10-18:
The ESP32's are definitely good for RunCPM.
Right now I think the Teensy 3.6 is better, even though the ESP32 cpu is faster.
But on the ESP32 we can implement things like the RTC and WIFI, which will move away from CP/M compatibility, but into some more useful development with all the languages CP/M have to offer.
I ended up buyng 4xESP32-TTGO-T1 and 2xLolin32-pro. I will certainly find an use for each one of them.

drawkula, 2018-10-18:
The SD speed really is subterran.
Rebuilding the whole SPL stuff takes hours on the ESP32 and that seems to be caused mainly by slow loading of the binaries.
Probably even the serial floppy drive of the C128 was faster with CP/M.
I think after a while of playing with other stuff on the ESP32 ideas to get faster loading times will show up.
Maybe even the bottleneck still is in the Arduino core and we'll see improvements with future updates.

I didn't find time today to look closer at MicroPython but testing MicroPython's SD card speed will be one of the things to be done as soon as possible.

MockbaTheBorg, 2018-10-20:
Yup ... this is my feeling. They may have a bad SD library.
The serial is also slow it seems.
I have no time to test anything else for at least two more weeks (working in Chicago).

drawkula, 2019-01-06:
The SD card access is much faster now.

Rebuilding SPL [mentioned above with links](https://github.com/MockbaTheBorg/RunCPM/issues/66#issuecomment-429084086) now rebuilds in roughly 15 minutes on an old slow 2G µSD.

The current version:
```
CP/M 2.2 Emulator v3.7 by Marcelo Dantas
Arduino read/write support by Krzysztof Klis
      Build Jan  6 2019 - 12:45:57
--------------------------------------------
CCP: CCP-CCPZ.64K    CCP Address: 0xf400
BOARD: DOIT Esp32 DEVKIT V1
Initializing SD card.

A>█
```

...aaaaand: HNY! ;-)