MockbaTheBorg/RunCPM #68: STM32 Port

MockbaTheBorg, 2018-11-12:
Hi All,

I have pulled my STM32 board from the drawer and gave it a try on the Arduino.
It seems to work fine with code from this project: https://www.stm32duino.com/
Source code here: https://github.com/rogerclarkmelbourne/Arduino_STM32
SD Library here: https://github.com/stm32duino/STM32SD (I have not tested this yet)
There's a good chance this will work for running RunCPM.

I have tried compiling RunCPM for it but got some errors, also, the code needs defines to use different settings when building for this board. Not sure when I will be able to port it, but if anyone wants to give it a try, check pages the above.

I am not using the guy's STM32duino boot loader, as I feel it is not needed, you can program the STM board from Arduino using the on-board st-link. Needs to install the ST-Link driver btw.

I was able to load a led-blink program onto it, which is already some progress.

Cheers,
Mockba.

MockbaTheBorg, 2018-11-12:
Tried this one as well: https://github.com/stm32duino/Arduino_Core_STM32
No luck so far.

MockbaTheBorg, 2018-11-12:
I was able to port RunCPM to the STM32F4DISCOVERY board.
It is still unstable, but with promising results.
The steps to implement are:

1 - Install Arduino STM32 core from https://github.com/stm32duino/Arduino_Core_STM32 via the package manager.
2 - Install **FatFs** and **STM32SD** libraries via the library manager.
3 - Edit the **stm32f4xx_hal_conf.h** file for the board and uncomment **#define HAL_SD_MODULE_ENABLED**
4 - Edit the **ffconf_default_68300.h** file from the STM32SD library and set **#define _FS_TINY 1**
5 - Build, install.
6 - Plug a serial adapter to PA2 (TX) and PA3 (RX)
7 - Press reset (black) button.

RunCPM should show the greeting.

At this point I am not 100% sure of the correct operation of the file system, so be aware that file corruption can occur.

MockbaTheBorg, 2018-11-16:
Further investigation leads me to think there's a memory leak on the SD card handling.
Multiple DIR commands lead to a crash ... 

So far I was not able to replicate.

MockbaTheBorg, 2018-11-19:
STM32 board back to the drawer for now. Maybe I will revisit in the future if there's free time.

closed 2018-11-19 by MockbaTheBorg

MockbaTheBorg, 2018-11-22:
Port complete.

linker3000, 2019-06-23:
Any tips for compiling for the STM32F411 Nucleo? 

I have followed the instructions for setting up the discovery board (above)
 and found/confirmed every settings change, but when I compile with 'Board Nucleo-64' and 'part number Nucleo F411RE', the compilation fails at "PD13 was not declared in this scope"..

```
#if defined _STM32_DEF_ // STM32 boards
  SdFatSoftSpiEX<PC8, PD2, PC12> SD; // MISO, MOSI, SCK
  #define SDINIT PC11 // CS
  #define LED PD13  <<<<<< 
```
Is this setup missing a pin definition file?

Thanks

MockbaTheBorg, 2019-06-24:
Hi Nigel,

I am not sure how the Nucleo-64 is laid out.
Port PD13 is what I had defined for the board I have, which might be
different on yours.
Take a look at the schematic of your board, check for which one is the LED
port.
A quick look at one schematic I have found on ST's website seems to
indicate that it can be either PA5 or PB13. (configured by jumpers maybe)
Indeed the processor doesn't seem to have a port D.
Also, I am not sure which pin will be used for the SD card's CS signal. You
will need to test a demo design first, to make sure you have access to the
data in the SD card, then try the same pin and initialization code for
RunCPM.
As I do not have one of these boards, I can't say for sure.
It was also not clear to me how much RAM memory it has. You need more than
64K or ram, as RunCPM allocates 64K already for the CP/M operating system,
and more for extra variables and stuff.
The Arduino DUE has 96K on it, so if your board has at least that you
should be fine.
Let me know how it goes for you.

Cheers,
Marcelo.

linker3000, 2019-06-24:
Thanks for the reply.

Got it...if the board has that port/pin, the support libs would have 
defined it.

I think I've found the pin for the LED - I'll start with a blinky program!

Board has 128K RAM, so let's see if I can improve my programming skills 
and get this going...

linker3000, 2019-06-24:
Working on the STM32F411 Nucleo!

STM code section below. Once running, the tty comes through the embedded ST-Link interface apparently fine and I can see the console in the Arduino IDE or using minicom.
See the note in the code about the SD card activity LED.

`#if defined _STM32_DEF_ // STM32 boards

    // F411RE uses MISO = PA6 (D12), MOSI = PA7 (D11), SCK = PA5 (D13) 
    // (See note about LED below), CS = PB6
    SdFatSoftSpiEX<PA6, PA7, PA5> SD; // MISO, MOSI, SCK
    #define SDINIT PB6 // F411RE uses CS = PB6 (D10)

    #define LED PA8 // STM32F411 Nucleo has an LED on PA5 (SCK), which will flash during SD access. 
                  // LED reassigned here to a diffent pin so we don't drive the SCK line.
                  // Stick an LED + resistor on it if you want two LEDs for SD access!
                  // PA8 is Arduino pin D7

    #define LEDinv 0 // 0=normal 1=inverted
    #define BOARD "STM32F411 Nucleo"`