MockbaTheBorg/RunCPM #71: RunCPM 3.7 for Win does exit after DIR

guidol70, 2018-12-04:
Today I did download v3.7 of RunCPM and the actual A.ZIP,
but wheater the content of the old A.ZIP or the new A.ZIP
RunCPM 3.7 on Windows does exit for me after gigving the command DIR :(
`
CP/M 2.2 Emulator v3.7 by Marcelo Dantas
      Build Nov 29 2018 - 01:41:18
-----------------------------------------
CCP : CCP-ZCP2.60K    CCP Address: 0xe400

RunCPM Version 3.7 (CP/M 2.2 60K)

A>DIR
1STREAD .ME

C:\TEMP\RunCPM_Win>
`
I didnt got this with a v3.7 compiled under armbian-linux on a Pi-Style SBC.

guidol70, 2018-12-04:
Ahh - sorry I did miss the Info about the new CCP-Version for CCP-ZCP2.60K.
Now the v3.7 for Windows doesnt exit
https://github.com/MockbaTheBorg/RunCPM/commit/9fc6c16b2e584b9ffd5f107914293d76c3882356

For me it can be closed :)

MockbaTheBorg, 2018-12-05:
Thanks for the info.
Yes I was going to say the same.
I had to do a few (many) changes to the code to increase compatibility with the various CCP's, and this was indeed going to break compatibility with the previous versions here and there.
Glad to know that you got it sorted out.

closed 2018-12-05 by MockbaTheBorg