MockbaTheBorg/RunCPM #72: Cant compile RunCPM v3.7 for Arduindo DUE

guidol70, 2018-12-04:
I downloaded the Master today and did try to compile (with Arduino IDE 1.8.7) for the Arduino DUE.
First I did see I missed the SDFat Library and installed the v1.0.7 via "Manage Library"

But while compiling the IDE has some problems mit the SDFAT EX from the RunCPM.ino

Please read attached .txt file
[Arduino_DUE_ErrMsgs.txt](https://github.com/MockbaTheBorg/RunCPM/files/2645203/Arduino_DUE_ErrMsgs.txt)

MockbaTheBorg, 2018-12-05:
Sorry I should have documented this better.
On the RunCPM.ino, where you set up the details for each board, I left these notes:
```
// SDCard/LED related definitions
//   SdFatSoftSpiEX and SdFatEX require changes to SdFatConfig.h:
//     #define ENABLE_EXTENDED_TRANSFER_CLASS 1
//     #define ENABLE_SOFTWARE_SPI_CLASS 1
```
You need to edit the SdFatConfig.h file of the SD Library and make those changes.
Those two values are 0 by default.

Give it a try and let me know if you succeed.

guidol70, 2018-12-05:
OK - I did find the SdFatConfig.h on my computer as
C:\Users\guido\Documents\Arduino\libraries\SdFat-1.0.7\src\SdFatConfig.h
and did edit the 2 parameters.
Now I could compile the RunCPM.ino **BUT now** it seems to have a problem after loading the CCP from the SDCard:
**CP/M 2.2 Emulator v3.7 by Marcelo Dantas
Arduino read/write support by Krzysztof Klis
      Build Dec  5 2018 - 09:11:54
--------------------------------------------
CCP: CCP-DR.60K    CCP Address: 0xe400
BOARD: ARDUINO DUE
Initializing SD card.
Unable to initialize SD card. CPU halted.**

I did test it with different Cards (Real SD or MicroSD with adapter) from 1-4GB and FAT/FAT32
These cards did run with v3.3 on my SD-Card-Hat for the Arduino DUE without changing SDCard-Pin Numbers in the RunCPM.ino

I also tested with old CCPs and new CCPs on the SDCard.
My DUE is - at this state - blink a heartbeat at the "L"-LED

MockbaTheBorg, 2018-12-05:
If you got no compilation errors then this is a good sign.
Now the problem may be related to the SD card connection.
Most probably the SD card CS pin.
I am using a shield called "Arduino Wireless SD Shield". (https://store.arduino.cc/usa/arduino-wirelss-sd-shield)
This one I have has the CS pin set to 4, but others can have different pins. You can see I set SDINIT to 4 for the Arduino, so the call to SD.begin below that uses it.
At this point you need to make sure you format the SD card with FAT16 (FAT32 probably works as well) and place all the files and folder onto the root of the formatted file.
Another test is to make sure Arduino can actually access your SD card, regardless of RunCPM, so you can use one of the SdFat library tests, like SdInfo or SoftwareSpi.
If they can read your card fine, then you may need to check the schematics of your SD card adapter to see which pins are used.
I have just recompiled v3.7 and applied to my Arduino to test.

guidol70, 2018-12-05:
I checked the CS-Pin of my itead.cc Stackable SD Card Shield V3.0 and this also do use Pin D4 for SD_CS
I did use this adapter and sdcard  before with up to RunCPM v3.3 without problems.
I formatted again the card as FAT(16) (on Windows 10 and Windows XP) and did put (like before) the files at the root of the SDCard.

I did test SdInfo with SD_CS 4- but did get the error
type any character to start
error: cardSize failed
SD errorCode: 0X50,0X0

The adapter also uses the corect Pins D11 MOSI, D12 MISO and D13 SCK
[Sch_sd_shield_3.0_datenblatt.pdf](https://github.com/MockbaTheBorg/RunCPM/files/2647838/Sch_sd_shield_3.0_datenblatt.pdf)

guidol70, 2018-12-05:
One little small success with SdInfo :)
I did the issues for SDFat on Github and found
https://github.com/greiman/SdFat/issues/111

After setting in SdInfo.ino the Line
from:   if (!sd.cardBegin(SD_CHIP_SELECT, SD_SCK_MHZ(50)))
to: if (!sd.cardBegin(SD_CHIP_SELECT, SD_SCK_MHZ(25)))

the SDInfo-Sketch did work:
init time: 2 ms
Card type: SD2
Manufacturer ID: 0X74
OEM ID: JE
Product: SDC
Version: 1.0
Serial number: 0X2F16904D
Manufacturing date: 5/2011

So now I have to find this init speed in the Library which is used by RunCPM :)

guidol70, 2018-12-05:
Haha - SUCCESS!!
I did found 
SPISettings settings = SD_SCK_MHZ(50)
in the SdFat.h  (NOT SdFatConfig.h)

Changed this into
SPISettings settings = SD_SCK_MHZ(25)
and did recompile.

But the same result - SD Card init failed....

Then I found similiar lines with
spiSettings = SPI_FULL_SPEED

I did give it a try and replaced ALL 
from: SPI_FULL_SPEED
to: SD_SCK_MHZ(25)   
and also up to SD_SCK_MHZ(40) did work :)   

And after the next compile I do get:
CP/M 2.2 Emulator v3.7 by Marcelo Dantas
Arduino read/write support by Krzysztof Klis
      Build Dec  5 2018 - 11:59:26
--------------------------------------------
CCP: CCP-ZCP2.60K    CCP Address: 0xe400
BOARD: ARDUINO DUE
Initializing SD card.
RunCPM Version 3.7 (CP/M 2.2 60K)
A>

So now - maybe - I only have to find which one the bad one was....

MockbaTheBorg, 2018-12-05:
This is interesting ... So looks like SdFat has speed issues with your combination of adapter/card. I don't remember having to change speed configurations using my wireless shield, but it all depends on shield design I guess.
I apologize for it happening as I changed to SdFat, but at the end I think the change is positive, as SdFat will be one single library for all the RunCPM ports, which simplifies the coding.
Right now I am working on porting RunCPM to an Altera FPGA board, which normally uses SdFat as well. So let's see.
I will close this one, feel free to open any other issue you may find. And enjoy RunCPM. :)

closed 2018-12-05 by MockbaTheBorg

guidol70, 2018-12-05:
Info from the SdFat-Library Autor Bill Greiman at 
https://github.com/greiman/SdFat/issues/117#issuecomment-444488987

> SPI_FULL_SPEED is defined as:
#define SPI_FULL_SPEED SD_SCK_MHZ(50)

The result will be the highest speed less than or equal to 50 MHz supported by your board. On Due this will be 48 MHz.

For SD_SCK_MHZ(40), the Due SPI library will choose 32 MHz which is the max supported speed <= 40.

I have no problem running modern SD cards on Due at 48 MHz. This depends on using a bare SD socket.

I can't tell what speed your SD shield will support on Due. Also some very old SD cards run at a max speed of 25 MHz.

I can't help more since I don't have your shield and SD card.

>