MockbaTheBorg/RunCPM #73: Certain errors seem to cause files to disappear

mbramwel, 2018-12-12:
I have verified this with 2 CCP.  I have used the internal as well as CCP-CCPZ.60

If I cause certain I/O errors to occur, files seem to disappear.  After trying a few things, I discovered that my files were not truly disappearing, the User area was changing to "1".

How to test:
Create a system with at least 1 drive.
Do a DIR A: to verify that you see all of the files.
Try to switch to a non-existant drive letter.  Example:  C:
You will get a bdos error.  Hit any key to put you back to A:
Do a DIR, where did the files go??
You have been put into USER 1.
Type USER 0 and poof!  The files are back again.

I have verify that more than 1 CCP has this anomoly.
Is this an emulation problem or *normal* for CPM?

I am running Centos 7.6, make posix build
I have all user areas created  A/0, A/1, A/2, etc...

drawkula, 2018-12-12:
On...
```
$ lsb_release -sir
Debian
9.6
```

```
CP/M 2.2 Emulator v3.6 by Marcelo Dantas
      Build Nov 22 2018 - 09:18:33
-----------------------------------------
CCP : CCP-CCPZ.64K    CCP Address: 0xf400

A>dir
1STREAD .ME   |  ASM     .COM  |  BDOS    .ASM  |  BDOS    .LUA
BDOS    .SUB  |  BDOSEQU .LIB  |  CAL     .COM  |  CCP     .ASM
CCP     .SUB  |  CCPZ    .SUB  |  CCPZ    .Z80  |  CLEAN   .SUB
DDT     .COM  |  DISKDEF .LIB  |  DISPLAY .LIB  |  DO      .COM
DUMP    .ASM  |  DUMP    .COM  |  ED      .COM  |  EXIT    .ASM
EXIT    .COM  |  EXIT    .SUB  |  EXIT    .Z80  |  HELLO   .LUA
INFO    .COM  |  INFO    .SUB  |  INFO    .Z80  |  LOAD    .COM
LU      .COM  |  LUA     .COM  |  LUA     .SUB  |  LUA     .Z80
LUAINFO .LUA  |  MAC     .COM  |  MAKEFCB .LIB  |  MBASIC  .COM
MLOAD   .ASM  |  MLOAD   .COM  |  MLOAD   .DOC  |  MOVCPM  .COM
OPCODES .TXT  |  PIP     .COM  |  PMA     .DOC  |  PMARC   .COM
PMARCEXT.DOC  |  PMCAT   .COM  |  PMCHGMX .MAC  |  PMEXE2  .COM
PMEXT   .COM  |  PMSET   .COM  |  PMSFX2  .COM  |  READ    .ME 
RSTAT   .COM  |  RSTAT   .SUB  |  RSTAT   .Z80  |  STAT    .COM
SUBMIT  .COM  |  SUBMITD .COM  |  SYSGEN  .COM  |  TE      .BKP
TE      .COM  |  UNARC   .COM  |  UNCR    .COM  |  USQ     .COM
XSUB    .COM  |  Z2HDR   .LIB  |  Z3BASE  .LIB  |  Z3HDR   .LIB
Z80ASM  .COM  |  Z80ASM  .PDF  |  Z80CCP  .ASM  |  Z80CCP  .SUB
ZCPR2   .ASM  |  ZCPR2   .SUB  |  ZCPR3   .ASM  |  ZCPR3   .SUB
ZEXALL  .COM  |  ZEXDOC  .COM  |  ZSID    .COM  |  ZTRAN   .COM
A>p:
Bdos Error on P : Select
A>dir
1STREAD .ME   |  ASM     .COM  |  BDOS    .ASM  |  BDOS    .LUA
BDOS    .SUB  |  BDOSEQU .LIB  |  CAL     .COM  |  CCP     .ASM
CCP     .SUB  |  CCPZ    .SUB  |  CCPZ    .Z80  |  CLEAN   .SUB
DDT     .COM  |  DISKDEF .LIB  |  DISPLAY .LIB  |  DO      .COM
DUMP    .ASM  |  DUMP    .COM  |  ED      .COM  |  EXIT    .ASM
EXIT    .COM  |  EXIT    .SUB  |  EXIT    .Z80  |  HELLO   .LUA
INFO    .COM  |  INFO    .SUB  |  INFO    .Z80  |  LOAD    .COM
LU      .COM  |  LUA     .COM  |  LUA     .SUB  |  LUA     .Z80
LUAINFO .LUA  |  MAC     .COM  |  MAKEFCB .LIB  |  MBASIC  .COM
MLOAD   .ASM  |  MLOAD   .COM  |  MLOAD   .DOC  |  MOVCPM  .COM
OPCODES .TXT  |  PIP     .COM  |  PMA     .DOC  |  PMARC   .COM
PMARCEXT.DOC  |  PMCAT   .COM  |  PMCHGMX .MAC  |  PMEXE2  .COM
PMEXT   .COM  |  PMSET   .COM  |  PMSFX2  .COM  |  READ    .ME 
RSTAT   .COM  |  RSTAT   .SUB  |  RSTAT   .Z80  |  STAT    .COM
SUBMIT  .COM  |  SUBMITD .COM  |  SYSGEN  .COM  |  TE      .BKP
TE      .COM  |  UNARC   .COM  |  UNCR    .COM  |  USQ     .COM
XSUB    .COM  |  Z2HDR   .LIB  |  Z3BASE  .LIB  |  Z3HDR   .LIB
Z80ASM  .COM  |  Z80ASM  .PDF  |  Z80CCP  .ASM  |  Z80CCP  .SUB
ZCPR2   .ASM  |  ZCPR2   .SUB  |  ZCPR3   .ASM  |  ZCPR3   .SUB
ZEXALL  .COM  |  ZEXDOC  .COM  |  ZSID    .COM  |  ZTRAN   .COM
```
Strange! I see my files.
...just to verify user switching still to work:
```
A>user 1
A1>user 0
A>█
```
I'm a bit busy now, so I didn't have the patience to build the newest RunCPM for this test.

MockbaTheBorg, 2018-12-12:
I may have broken something with the new libraries. Checking ...

MockbaTheBorg, 2018-12-12:
I have tried it here with both CCPs, both on Linux and Windows and with TPA sizes of 60K and 64K.
No re-creation. It seems to work fine all the time.

Are you using the latest code from Github?

For running on Linux the only thing to check would be to make sure the folder letters are uppercase, and also if the CCP's are upper case, though this would not have any interference with user area selection.

If you are not running v3.7, please give it a try, as I have replaced the SD library, and now rename and truncate are working much faster.
Just make sure you read the GitHub Readme as a few things have changed, or example, user areas are now mandatory, no more non-user-area disks.

Let me know if this helps... if you get the same behavior with the latest code, then we need to find a way to debug it.

IMPORTANT: when updating to the latest code, make sure you get the latest CCP's as well, as ZCPR2 and ZCPR3 have also been updated.

mbramwel, 2018-12-12:
Interesting.  You must have made some very_recent changes.  It did a git pull and re-built the binary.
I am seeing different results now.

Since have the old binary from a week ago, I can easily see the change in behaviour.

Now it does not do strange things with the BDOS error and auto-changing the user area to 1.

What I have noticed:
The A drive is now searched, even if you are on another drive (this is good!)
The newer RunCPM and CCP Internal now give A0> as the prompt.  (Last week it was just A>)
The user area is displayed as part of the prompt, examples:   A0>  A1>  Ae> Af>

FYI: I made a BASH script to create the main-folders and sub-folders with a stub file in each

for letter in {A..P}
do

  # Create Drive Folders  eg: ./A  ./B  ./C
  if [ ! -d $letter ]; then
    echo "Create Drive $letter"
    mkdir "$letter"
  fi

  # Create user areas 0..9 & A..F for in each Drive Folder eg: ./A/0  ./A/1  ./A/2
  for user in {{0..9},{A..F}}
  do
    if [ ! -d $letter/$user ]; then
      echo "- $letter/$user"
      mkdir "$letter/$user"
    fi

    # create stub file in each folder/user  eg: ./A/0/DRIVE.A0
    info="Drive=$letter: User=$((0x$user))"
    echo $info > "$letter/$user/DRIVE.$letter$((0x$user))"
  done

done

mbramwel, 2018-12-12:
oops!  The markdown causes the previous message to display the BASH script in an odd way.

[http://web1.foxhollow.ca/makeDrives](url)

I have placed the bash script on my server.
Feel free to use it in any way you want.

Thanks for everything!

drawkula, 2018-12-12:
Use three backtics on an own line to wrap code or indent all code lines by 4 spaces:
```
$ mkdir -pv {A..D}/0
```

MockbaTheBorg, 2018-12-12:
Cool ... yes, I have made these latest changes just few days ago.
The behaviors you are seeing are CCP dependent. The vanilla DR CCP won't do any of that.

Glad to see that it is all working fine now.

closed 2018-12-12 by MockbaTheBorg