MockbaTheBorg/RunCPM #74: Is there a blog to discuss RunCPM?

mbramwel, 2019-01-07:
I just received a TTGO board. I compiled RunCPM and it seems to work.

I created a ESP32_telnetd.h and with a small change to the arduino_abstraction, I am now able to telnet into the ESP32/RunCPM board.

Seems to work, nothing has failed yet.  The EXIT (was halt the emulator) command does a network close and waits for another inbound connection.

**Is there a blog somewhere that we can post code and make comments?**  
Github seems to be the wrong tool for that type of activity.

**$telnet 192.168.31.139**

Trying 192.168.31.139...
Connected to 192.168.31.139.
Escape character is '^]'.

CP/M 2.2 Emulator v3.7 by Marcelo Dantas
Arduino read/write support by Krzysztof Klis
      Build Jan  6 2019 - 22:27:16
'--------------------------------------------'
CCP: INTERNAL v1.5    CCP Address: 0xe400
BOARD: TTGO_T1
Initializing SD card.

RunCPM Version 3.7 (CP/M 2.2 60K)

A0>info
RunCPM System Information v1.0

RunCPM Version 3.7 running on ESP32
CCP is Internal Version 1.5
CCP loads at address E400h
BDOS is at address EC00h
BIOS is at address FE00h
60160 bytes available for applications

RunCPM Version 3.7 (CP/M 2.2 60K)

A0>exit

Connection closed by foreign host.

drawkula, 2019-01-07:
Yayyyy... telnetting into CP/M... \o/

> I created a ESP32_telnetd.h and with a small change to the arduino_abstraction, I am now able to telnet into the ESP32/RunCPM board.

Can you make a PR for this?

> Is there a blog somewhere that we can post code and make comments?
> Github seems to be the wrong tool for that type of activity.

I don't think fragmentation of information is a good idea. So looking for a way how to keep blogging and documentation here would be my preferred choice.

ForthHub is using the issues of an extra repository as a kind of forum.
—▷ https://github.com/ForthHub/discussion
Let's have a look how they do it.

PRs have discussions too.
Can issues and PRs be cloned like the Wikis which just are git-repos too?
If not, that'd would be a showstopper and a GitHub based Wiki would be the better choice.

Doing it with Git and keeping it clone-able is better for the survival of the information.

Summary: Probably we can discuss code and have blogging here in many ways. We just need to define how to do it.

mbramwel, 2019-01-07:
In the zip file:  ESP32_telnetd.README

 

Notes from Mark Bramwell

re: ESP32_telnetd.h

 

This will turn the ESP32/RunCPM into a network node.

While the network connection is alive, the serial terminal is also live.

A serial terminal connection is not required, the ESP32 will run standalone from battery.

 

changes from stock RunCPM

 

RunCPM: 

  #include "ESP32_telnetd.h"

 

  #ifdef ESP32

  _clrscr();

  initializeWiFi();

  #endif

 

  Also increased serial speed to 115200 (makes a big difference!)

 

 

abstraction_arduino.h:

search for "wifi"

keyboard/serial routines now also send/check the network stack

Specific functions changed include:

  int _kbhit(void)

  uint8 _getch(void)

  void _putch(uint8 ch)

  void _clrscr(void)

 

 

ccp.h:

The EXIT command has been changed to force a reboot of the ESP32 instead of just halting

  case 8: // EXIT

   #ifdef ESP32

   wifi_disconnect_sessions();

   ESP.restart();

   #endif ESP32

 

 

ESP32_telnetd.h is a new file

mbramwel, 2019-01-07:
I’ll give the code a check tomorrow evening to make all is good.

I’ll have to figure out how to properly do github without embarrassing myself too badly.

 

If you would like to play with the code right away:  http://web1.foxhollow.ca/RunCPM/

 

Edit ESP32_telnetd.h

Put in your own SSID + password

 

You can select DHCP or STATIC

Uncomment this line:  // #define use_static_ip

Then look in the same file for the lines that look like:  

 

// If you have requested a STATIC IP for your ESP32

#ifdef use_static_ip

IPAddress local_IP(192, 168, 31, 50);

IPAddress gateway(192, 168, 31, 1);

IPAddress subnet(255, 255, 255, 0);

#endif

 

Then Control-U to send it to your ESP32 TTGO

drawkula, 2019-01-07:
> I’ll give the code a check tomorrow evening to make all is good. I’ll have to figure out how to properly do github without embarrassing myself too badly.

Don't worry! I periodically screw up my own repositories and then need to pull them from GitHub/GitLab again to restart from a safe state. Shit happens! Life has too many other tasks that are far more important than becoming a never failing git expert...

> If you would like to play with the code right away: http://web1.foxhollow.ca/RunCPM/

I'll look at it but maybe your polished version will be there faster than my results.
We'll see...

Stay ommmmmmPtimistic! \o/

drawkula, 2019-01-07:
I imported the ZIP as new branch into [my RunCPM clone](https://github.com/drawkula/RunCPM/tree/telnetd-mbramwel).

drawkula, 2019-01-07:
```
(yeti@kumari:15)~$ telnet doit-esp32-devkit-v1-1
```
After clearscreen:
```
CP/M 2.2 Emulator v3.7 by Marcelo Dantas
Arduino read/write support by Krzysztof Klis
      Build Jan  7 2019 - 07:13:50
--------------------------------------------
CCP: INTERNAL v1.5    CCP Address: 0xe400
BOARD: DOIT ESP32 DEVKIT V1
Initializing SD card.

RunCPM Version 3.7 (CP/M 2.2 60K)

A0>info
RunCPM System Information v1.0

RunCPM Version 3.7 running on ESP32
CCP is Internal Version 1.5
CCP loads at address E400h
BDOS is at address EC00h
BIOS is at address FE00h
60160 bytes available for applications

RunCPM Version 3.7 (CP/M 2.2 60K)

A0>user 1
A1>c:
C1>do myhi
SuperSUB V1.1
Directory full error on line number: 5

RunCPM Version 3.7 (CP/M 2.2 60K)

C1>█
```
Ok... so far, so good.
I get some directory full errors when trying to compile stuff but I had this yesterday too without the telnetd changes. Maybe a waggly connection in the jumper wires or a bitchy µSD card or it just needs a filesystem check. Definitely not a telnetd related problem.

I need to narrow that down before anything else.

drawkula, 2019-01-07:
I've put my WiFi config in an own header which is added to `.gitignore`:
https://github.com/drawkula/RunCPM/blob/local%2Btelnetd/.gitignore#L62

There is an empty example for this file instead:
https://github.com/drawkula/RunCPM/blob/local%2Btelnetd/RunCPM/esp32_wifi_config.h.example

https://github.com/drawkula/RunCPM/blob/local%2Btelnetd/RunCPM/ESP32_telnetd.h#L12-L20

Maybe we'll have a better idea how to set SSID and PASSWORD without fearing to publish it via git.

MockbaTheBorg, 2019-01-31:
Sorry guys I have been disconnected for some time.
You know, sometimes life gets in the way.
What did I miss?

drawkula, 2019-01-31:
> What did I miss?

Telnetting to RunCPM.
It works!

I threw it in the corner because of SD card access errors. Meanwhile I have some more SD breakouts (still untested) but I'm not expecting them to behave better because I had write failures on the Wrover32 board with onboard SD slot too which should have those pullup resistors. And from other projects I remember comments that the resistors should not really be needed in a pure 3.3V setup. I'm waiting for the right moment and mood to retry it but that's not a telnet related issue.

MockbaTheBorg, 2019-01-31:
Cool ... I will go read about it.

MockbaTheBorg, 2019-02-28:
This one is not really an issue anymore right? Can I close it?

(we need that forum)

mbramwel, 2019-03-01:
Yep.  Feel free to close it.  The entry was more of an announcement.

 

I will have another version in a few weeks to support slightly different networking mode.

At that time we can make a new entry.

MockbaTheBorg, 2019-03-01:
Cool. Closing this one then.

closed 2019-03-01 by MockbaTheBorg

guidol70, 2021-11-19:
@mbramwel @drawkula @MockbaTheBorg 
Today I found this announcement and tested with the normal v5.7 of RunCPM and didnt got problems with File-access by now :)

At the command prompt I got a double-CR so I do get
```
A0>
A0>

```
and at the start of opening a file in Wordstar I seem to get another char which prevent the initial edit mode :(

@mbramwel 
Did you ever create another network-mode? ;)

[image]

guidol70, 2021-11-21:
> At the command prompt I got a double-CR (when pressing Enter/Return) so I do get
> 
> ```
> A0>
> A0>
> ```
> also after a DIR command and at the start of opening a file in Wordstar I seem to get another char which prevent the initial edit mode :(

because I didnt got this problem with a telnet-connect from linux/armbian I found the problematic setting in puTTY / KiTTY - which has to be disabled to get it working right ;)

[image]

guidol70, 2021-12-06:
If you get distorted output from the ESP32-Telnet-Server on a Telnet-Client 
(like the FabGL Network Terminal) then change in the ESP32-Telnet-Server 
**ESP32_telnetd.h**

from
```
// start listening on the WiFi IP
server.begin();
server.setNoDelay(true);

```
to
```
// start listening on the WiFi IP
server.begin();
// server.setNoDelay(true);
```

this is because of the **Nagle algorithm**:
`The purpose of the Nagle algorithm is to reduce the tcp/ip traffic of small data packets sent over the network by combining some small sending messages and then sending all the messages at once. The disadvantage of this method is that it delays the sending of a single message until a sufficiently large TCP-packet is assembled`