MockbaTheBorg/RunCPM #78: Windows RunCPM folder can't be on Dropbox

MustBeArt, 2019-03-24:
On Windows, RunCPM seems unable to cope with the RunCPM folder being on Dropbox. On startup, we get:

Bdos Error on A : Select

I'm not currently set up to recompile on Windows, but I think I can see the problem.

In abstraction_vstudio.h, around line 126, we have:

int _sys_select(uint8 *disk) {
	return((uint8)GetFileAttributes((LPCSTR)disk) == 0x10);
}

I checked the file attributes returned by the WIN32 API using Python, and it seems that the folder in Dropbox returns 0x30 instead of 0x10. The attributes are a bit mask, so that's 0x20 | 0x10, where 0x10 is the "I'm a directory" bit and 0x20 is the "archive" bit.

I think the problem would go away if the above lines of code were replaced by:

int _sys_select(uint8 *disk) {
	return(((uint8)GetFileAttributes((LPCSTR)disk) & 0x10) != 0);
}

MustBeArt, 2019-03-24:
I've since tried this fix and it seems to work.

MockbaTheBorg, 2019-03-26:
This one:
```
int _sys_select(uint8 *disk) {
return((uint8)GetFileAttributes((LPCSTR)disk) & 0x10);
}
```
Should work too, right? Can you give it a try?
I am travelling and won't be able to touch RunCPM (unless on Windows) for weeks.

Thanks!

MockbaTheBorg, 2019-03-26:
Tried it here ... seems to work fine.
I will upload shortly. Thanks a lot!

closed 2019-03-26 by MockbaTheBorg

MustBeArt, 2019-03-26:
Since _sys_select() is always called in a boolean context, yes that should work the same.

However, with a bit more testing (and more careful reading of the WIN32API docs) I see it's not correct in the failure case. We need something like this:
```C
int _sys_select(uint8 *disk) {
    uint32 attr = GetFileAttributes((LPCSTR)disk));
    return(attr != INVALID_FILE_ATTRIBUTES && (attr & FILE_ATTRIBUTE_DIRECTORY) != 0);
}
```

I've compiled and tested it (briefly), and it now works in both failure and success cases.

MockbaTheBorg, 2019-03-28:
Cool ... I have pushed the change regardless as it looks fine (apart from the extra parentheses on the second line ... :)
I will test more when I get back from my trip.