MockbaTheBorg/RunCPM #80: Integrated debugger

tomxp411, 2019-06-03:
Hi, I've been looking through the source code, getting an idea of RunCPM's strucutre and the BIOS emulation, and I noticed what looks like a debug monitor in the CPU module.

What I can't figure out is how to break into this debug module, so I can use this to test and debug 8080/Z80 code using the debugger. 

Is there a way to do this from within the emulator, or should I build a custom version of the code with a way to break in to the debugger?

MockbaTheBorg, 2019-06-05:
Hi Tom,

Sorry this was a "beta" functionality and I actually never documented it.
If you build RunCPM in Debug mode (on Visual Studio) or with -D DEBUG on Linux, you should be able to press ^D at any time RunCPM is waiting for console input to enter debug mode.
Then from there, press ? to see what commands are available.
I have used it a few times to debug internal code.
Give it a try and let me know if it works for you.

Thanks,
Marcelo.

MockbaTheBorg, 2019-06-05:
PS: It doesn't seem to be working with the internal CCP, I will have to fix that. But with the DR CCP (and probably the others) it works fine.

tomxp411, 2019-06-05:
Thanks. I’ll check it out today.

tomxp411, 2019-06-05:
Thank you, Marcelo. That worked well. I couldn't find the trigger for that in the code, but now that I know what I'm looking for....

Line 269 in cpm.h:
```
	case 0x09:				// 3 - CONIN - Console input
		SET_HIGH_REGISTER(AF, _getch());
#ifdef DEBUG
		if (HIGH_REGISTER(AF) == 4)
			Debug = 1;
#endif
		break;
```

I can see what's going on here (you're implementing BIOS calls in this method), but this code doesn't actually seem to be executed. (Setting a breakpoint in Visual Studio never hits this code.) Is this because I'm using an external CCP?

MockbaTheBorg, 2019-06-06:
I tested earlier today with the DR CCP and it worked fine.
If you are on the prompt you might need to press ctrl-d then enter.

Cheers,
Marcelo

closed 2019-06-07 by MockbaTheBorg

tomxp411, 2019-06-07:
Thanks, Marcelo. 

Yes, I have to press ENTER for the ^D to process on the command line. It processes instantly inside of an application. 

The debugger does work with the internal CCP, but since the internal CCP isn't processing any Z80 instructions, the debugger can't be used at the CP/M command prompt. It can be triggered while BASIC is running (I'm using that as my test case.)

As a suggestion... I changed the code to 28 on my system, so Control+\ would trigger the debugger. This allows me to use  Control+D in applications... otherwise, that's a problem. =)

Again, thanks! 
-Tom