MockbaTheBorg/RunCPM #81: STAT (on STM build) does not see all drives + reports same free size for all

linker3000, 2019-06-24:
STAT on my STM build doesn't consider a disk as present unless it's explicitly referenced, or has been 'visitied'. In any event, the free space for all disks seems to return the same value for all disks - but not always the same value, so, for example, A: is reported as different sizes (see examples below)

Here, I run STAT after walking round or referencing a few disks and doing a few RESETs. The system has drives A-D (all upper case), each with a '0' subfolder. The drives have different things on them.

`
CP/M 2.2 Emulator v3.7 by Marcelo Dantas
Arduino read/write support by Krzysztof Klis
      Build Jun 24 2019 - 19:29:15
--------------------------------------------
CCP: CCP-DR.60K    CCP Address: 0xe400
BOARD: STM32F411 Nucleo
Initializing SD card.

RunCPM Version 3.7 (CP/M 2.2 60K)

A>stat
A: R/W, Space: 898k
A>stat b:
Bytes Remaining On B: 898k
A>stat
A: R/W, Space: 898k
B: R/W, Space: 898k
A>

[RESET]

CP/M 2.2 Emulator v3.7 by Marcelo Dantas
Arduino read/write support by Krzysztof Klis
      Build Jun 24 2019 - 19:29:15
--------------------------------------------
CCP: CCP-DR.60K    CCP Address: 0xe400
BOARD: STM32F411 Nucleo
Initializing SD card.

RunCPM Version 3.7 (CP/M 2.2 60K)

A>stat b:
Bytes Remaining On B: 898k
A>stat
A: R/W, Space: 898k
B: R/W, Space: 898k
A>
B>a:stat
A: R/W, Space: 898k

[RESET]

CP/M 2.2 Emulator v3.7 by Marcelo Dantas
Arduino read/write support by Krzysztof Klis
      Build Jun 24 2019 - 19:29:15

CP/M 2.2 Emulator v3.7 by Marcelo Dantas
Arduino read/write support by Krzysztof Klis
      Build Jun 24 2019 - 19:29:15
--------------------------------------------
CCP: CCP-DR.60K    CCP Address: 0xe400
BOARD: STM32F411 Nucleo
Initializing SD card.

RunCPM Version 3.7 (CP/M 2.2 60K)

A>stat b:
Bytes Remaining On B: 898k
A>stat
A: R/W, Space: 898k
B: R/W, Space: 898k
A>d:
D>dir
D: ZORK1    COM : ZORK1    DAT : ZORK2    COM : ZORK2    DAT
D: ZORK3    COM : ZORK3    DAT
D>e:
Bdos Error on E : Select

RunCPM Version 3.7 (CP/M 2.2 60K)

D>dir
D: ZORK1    COM : ZORK1    DAT : ZORK2    COM : ZORK2    DAT
D: ZORK3    COM : ZORK3    DAT
D>a:
A>stat
A: R/W, Space: 962k
D: R/W, Space: 962k
A>b:
B>a:stat
A: R/W, Space: 962k
B: R/W, Space: 962k
D: R/W, Space: 962k
B>
`

MockbaTheBorg, 2019-06-25:
Hi Nigel,

Yes, STAT is finicky, because it expects real (physical) disks to exist in
the system, as it collects sector occupation bitmaps which won't exist in
RunCPM, as the filesystem is actually "hooked" and replaced by calls to the
underlying (Linux, Windows or Arduino) filesystem.
I have kept STAT for historical completeness, and added RSTAT, which
"kinda" replicates the behavior of STAT, or at least the part that works.
(Source code provided)
I made a short comment on thigs like RSTAT and MOVCPM, as these programs,
even though I kept them on the distribution, need real physical disks (or
disk images) to work correctly.
Take a look at the read-me file here (if you didn't already):
https://github.com/MockbaTheBorg/RunCPM

Good to know that you are actually able to run stuff already. This is cool.
One more board added to the compatibility list... :)

Cheers,
Marcelo.

linker3000, 2019-06-25:
Thanks Marcelo. Yes, I had read the comment on STAT etc., I was just concerned that operation was inconsistent rather that a binary 'works' or 'doesn't work'.

closed 2019-06-25 by linker3000