MockbaTheBorg/RunCPM #83: Arduino Compile Error

plainpylut, 2019-08-05:
Platform: Arduino IDE ver 1.8.9
Board: ESP32
OS: Windows 10 with latest updates
Issue: Compiling runCPM produces the following error code:
In file included from C:\Users\<myuser>\Documents\Arduino\Sketches\RunCPM\RunCPM\RunCPM.ino:50:0:

C:\Users\<myuser>\AppData\Local\Temp\arduino_build_483789\sketch\abstraction_arduino.h: In function 'bool _sys_extendfile(char*, long unsigned int)':

C:\Users\<myuser>\AppData\Local\Temp\arduino_build_483789\sketch\abstraction_arduino.h:171:29: warning: **comparison of unsigned expression < 0 is always false** [-Wtype-limits]

     if (f.write((uint8_t)0) < 0) {
                                      ^
Can you change the _sys_extendfile() routine so it functions correctly?
My changed routine is attached; however, modify it any way you desire to make if functional.

[_sys_extendfile.txt](https://github.com/MockbaTheBorg/RunCPM/files/3468698/_sys_extendfile.txt)

Thanks for developing this!  I'm still waiting on my ESP32 board.  Then I want to test  runCPM on it.
Garry

MockbaTheBorg, 2019-08-06:
Hi Garry,

Thanks for using RunCPM.

This is strange.
I have just compiled RunCPM for my Arduino Due and didn't get any warnings. I am using Arduino 1.8.9 too, which I have just installed. SdFat by Bill Greiman version 1.1.0.
Did you make any change to any configuration that you would remember? Maybe to compile a different project? Maybe you're not using the regular Greiman's SdFat library?
This is the definition of f.write() inside the library:
```
  /** Write a single byte.
   * \param[in] b The byte to be written.
   * \return +1 for success or -1 for failure.
   */
  int write(uint8_t b) {
    return write(&b, 1);
  }
```
So it is supposed to be less than 0 if there's an error.
The return value of this function is not unsigned, so nothing to fix there.
Please check it out and let me know. I am happy to assist more if needed.

Cheers,
Marcelo.

plainpylut, 2019-08-06:
Marcelo,
Correct, however, when I compile original (unmodified) RunCPM, using SDFat library with 1 file modified, I get a warning that is critical.  It shows the code will never be executed.

Here is what I did:
Copy extracted RunCPM distro into my Sketch subdir with name RunCPM
Start Arduino IDE
Manage Libraries -> Install "SDFat library version 1.1.0".
Edit SdFatConfig.h in actual disk install location, per this info in RunCpm.ino:
//   SdFatSoftSpiEX and SdFatEX require changes to the following lines on SdFatConfig.h:
//     #define ENABLE_EXTENDED_TRANSFER_CLASS (from 0 to 1 - around line 71)
//     #define ENABLE_SOFTWARE_SPI_CLASS (from 0 to 1 - around line 87)
Restart Arduino IDE
Preferences -> "Show verbose output during compiling" -> Compiler warnings -> "ALL"
Load RunCPM sketch
Compile Sketch
The following "Warning" is displayed:
In file included from C:\Users\kraemerg\Documents\Arduino\Sketches\RunCPM\RunCPM\RunCPM.ino:49:0:

C:\Users\kraemerg\AppData\Local\Temp\arduino_build_251570\sketch\abstraction_arduino.h: In function 'bool _sys_extendfile(char*, long unsigned int)':

C:\Users\kraemerg\AppData\Local\Temp\arduino_build_251570\sketch\abstraction_arduino.h:171:31: warning: comparison of unsigned expression < 0 is always false [-Wtype-limits]

     if ((f.write((uint8_t)0)) < 0) {

                               ^
The Warning states the "If statement will never be executed (entered)!

 Casting the f.write() return value stops the warning.
				if ((int)(f.write((uint8_t)0)) < 0) {
					result = false;
					break;
				}
I'm not sure if this is the correct way to fix the issue.

Thanks

Garry

plainpylut, 2019-08-06:
Marcelo,
This looks to be the way to fix the routine:
bool _sys_extendfile(char *fn, unsigned long fpos)
{
	uint8 result = true;
	File f;
	unsigned long i;
  int zero = 0;
	digitalWrite(LED, HIGH^LEDinv);
	if (f = SD.open(fn, O_WRITE | O_APPEND)) {
		if (fpos > f.size()) {
			for (i = 0; i < f.size() - fpos; ++i) {
				if ((f.write((uint8_t)0)) < zero) {
					result = false;
					break;
				}
			}
		}
		f.close();
	} else {
		result = false;
	}
	digitalWrite(LED, LOW^LEDinv);
	return(result);
}
Declaring a local integer value 'zero' and initializing it makes sure it is an integer.
Now the logical test using the f.write return value properly compares the integer variable 'zero'
I still don't understand exactly why '0' is unsigned, unless it is treated as a character.
I have never run into this in 20 years of software engineering.
Garry

MockbaTheBorg, 2019-08-08:
Yes, this is very strange ... if the compiler is not dumb (which seems to be the case) it should just interpret the numeric value as having the same type as to the expression to the left, being it a constant.

But anyways, one more compiler quirk that we have to work around. I will make the change to the official code branch, as I hate warnings too. :)

closed 2019-08-08 by MockbaTheBorg

MockbaTheBorg, 2019-08-08:
I have changed the code to:

if (f.write((uint8_t)0) != 1)

This should work as the library states it returns 1 for success.
So no more warning, please run some tests if possible.