MockbaTheBorg/RunCPM #84: Kermit for file transfers

moxhamj, 2019-08-07:
Now have Kermit working for file transfers to and from a PC, and also for transfers between to ESP32 boards running RunCPM. There are quite a few additions to the code, particularly with adding in handling for iobyte. Hopefully these can be added in some sort of 'official' way. They can all be added with conditional #ifdef ESP32 so as to not affect any other code. There is a small bug in the existing code where iobyte gets reset on return to the A> prompt so changing it with STAT was not working. Fixed by moving the initial iobyte declaration to a 'cold boot' part of the code, ie just before the signon message. Below is a copy of all the notes I made along the way. 
Pinout for SD card https://github.com/espressif/arduino-esp32/tree/master/libraries/SD
MISO IO19, MOSI I023, CLK IO18, CS IO5, 3V and 0V. These are all in a group apart from the 3V3. (ESP32 pins are all 3V3)
for arduino sized boards they have a reset when uploading the program. The smaller boards need the reset button pushed prior to uploading
globals.h - commented out USE_PUN and USE_LST for the moment
changed the esp32 pin definitions for the arduino form factor board "ESPDUINO-32"
#elif defined ESP32 // ESP32 boards
  SdFatSoftSpiEX<19, 23, 18> SD; // MISO, MOSI, SCK Some boards use 2,15,14,13, other 12,14,27,26
  #define SDINIT 5 // CS
  #define LED 2 // TTGO_T1=22 LOLIN32_Pro=5(inverted) DOIT_Esp32=2 ESP32-PICO-KIT=no led, Espduino pin 2
  #define LEDinv 0
  #define BOARD "ESPDUINO-32"
added in two new serial ports (the ESP32 has 3 hardware uarts, and later could possibly add software uarts as well)
  Serial1.begin(SERIALSPD,SERIAL_8N1, 26, 25);    //Baud rate, parity mode, RX, TX  ESP32
  Serial2.begin(SERIALSPD,SERIAL_8N1, 17, 16);    //Baud rate, parity mode, RX, TX  ESP32  
commented out //  _RamWrite(0x0003, 0x3D);   in cpm.h as this is resetting the iobyte on a warm boot, which means that STAT could change the iobyte but as soon as stat finished, it got changed back again
now have this line in the RunCPM tab
  _clrscr();
  _RamWrite(0x0003, 0x3D); // IOBYTE set 
  _puts("CP/M 2.2 Emulator v" VERSION " by Marcelo Dantas\r\n");
Then extensive changes to abstraction_ardunio.h so that all IO is redirected as per the iobyte. Example of the code is for console out
void outCON(uint8_t ch)
{
  uint8_t iobyte;
  iobyte = iobyteCON(); // 00, 01 10, 11
  switch (iobyte) {
    case 0: deviceTTYout(ch);
            break;
    case 1: deviceCRTout(ch);
            break;
    case 2: deviceBATout(ch);
            break;
    case 3: deviceUC1out(ch);
            break;                                 
  }
}

If you run STAT VAL: it prints out the list of all the 13 physical devices that can be connected (the table would suggest 4x4=16 but some are repeated)
CON: = TTY: CRT: BAT: UC1:
RDR: = TTY: PTR: UR1: UR2:
PUN: = TTY: PTP: UP1: UP2:
LST: = TTY: CRT: LPT: UL1:
In our case the important thing is the console is connected to the CRT, which is the first serial port 
Run STAT DEV: and it lists the current layout eg CON: CRT:
To change this, run STAT CON:=TTY: and now the console changes from Serial to Serial 1
It is very helpful at this point to have a D9 to USB adaptor, and also a terminal program such as Teraterm
Next were lots of experiments with the RDR port. It is possible to do things like PIP TEST.TXT=RDR:
however, this does not work for large files over about 30K  as CP/M runs out of Z80 memory and then goes to write all the data to the disk and then the buffer overflows. 
Tried adding a very large buffer and slow baud rates but I think the ESP32 memory can be used for better things.
Outputting data works fine too, eg PIP PUN:=TEST.TXT but it did need code to poll the number of bytes still available in the buffer and put in a 100ms delay if the buffer is getting full.
The next experiment is with Kermit. There is a "generic" version of Kermit that uses iobyte. The kermit documentation says that it defaults to PTR being the port with the following settings:
SET PORT xxx    input from      output to
   CRT            CRT:            CRT:
   PTR            PTR:            PTP:
However, lots of experiments(which are much easier on a simulation) suggest that this documentation is not correct, and that in fact the default setting for Kermit is to redirect CON to BAT
This is done by changing the two lower bits for iobyte from (our default of 01 which is CRT) to 10 which is BAT
Putting a debug on _kbhit shows that Kermit alternately polls the default port then the BAT port, so you can abort the transfer with a keypress (not sure which one though)
So for debugging at this point, attach a D9 to USB to Serial1 and open another terminal and use STAT CON:=TTY: to test the Serial One port, and then STAT CON:=BAT to test the Serial Two port
Once that is working, then replicate the whole thing with another ESP32 board. Connect these two with a D9 male to male crossover cable (pin 2 to 3, 3 to 2 and 5 to 5)
and it is then possible to run kermit on both boards and transfer files
KERMIT SEND TEST.TXT and KERMIT RECEIVE TEST.TXT
Kermit pauses every 10 secs or so to write data out to the disk and some packets look like they are resent
Text files are fine. Sending MBASIC.COM did not seem to work, the file was corrupted. However, using UNLOAD MBASIC.COM and sending MBASIC.HEX and then LOAD MBASIC.HEX all worked fine
For debugging it makes sense to leave the USB programming cable and use a terminal program (or even the arduino terminal program). This default is Serial Zero port.
For using the board standalone, TTY on Serial One may be better. It may be worth copying Grant Searle's idea of iobyte being set by the first character that comes in a port, CRT or TTY
see Grant Searle's page http://searle.hostei.com/grant/cpm/index.html discussion using two serial ports and switching using iobyte
Further thoughts, the ESP32 has Wifi and Bluetooth. These could ?? be CP/M 'devices'. The ESP32 bluetooth example is only a few lines.

MockbaTheBorg, 2019-08-08:
I wonder if transferring MBASIC.COM fails because of 8th bit stripping?
I wrote some programs long ago which are supposed to switch the console between 7 and 8 bits, they are called CONSOLE7 and CONSOLE8 (source code is there).
This is for when I was trying to have a zmodem transfer work between RunCPM and Putty. Which failed basically because I was never able to find a zmodem implementation, so I stuck to xmodem.
Give it a try there ...

MockbaTheBorg, 2019-08-08:
Btw ... I have created a discord channel for RunCPM.

https://discord.gg/7aQAGG

I know Discord is a gaming platform, but I use it on a daily basis for both gaming and other projects, so it would make it easier for me to be available there.

drawkula, 2019-08-08:
> Btw ... I have created a discord channel for RunCPM.

:-(

I don't think that is a good idea. Spreading RunCPM related discussions over too many media will result it being incomplete everywhere. And stuff like Discord is less under our control than Git and friends.

What about a GitHub/Gitlab-Wiki instead? It just would be a repository too so storage would be redundant and no evil provider could kill al the talk there.

MockbaTheBorg, 2019-08-08:
We can keep these here for our "offline" discussions, I thought of Discord to be more of an "online" channel, which GitHub lacks.
I have extreme OCD (just kidding), so seeing issues marked as open gives me itches ... lol.
Well ... between this and code compilation warnings ... I might actually have some level of OCD.

drawkula, 2019-08-08:
Wouldn't IRC be easier to log and convert if something worth to be keept happens there? And those who are interested in CP/M sure can live with a pure text chat...

https://xkcd.com/1782

moxhamj, 2019-08-09:
Unload and Load seems to be a universal solution to the 7th bit problem. One thing that is nice about having several serial ports is I can swap between various terminal emulations and real terminals - eg STAT CON:=BAT:
In terms of OCD and issues being "open", you can close this one as it is 'resolved'. I'm working now on building several little terminals. I have a tiny 1" OLED display that only costs a few dollars and runs on the I2C bus - fantastic for quick testing. Working on larger VT100 terminals using various LCD screens. Today managed to get a PS2 keyboard remapped so the arrow and Home/End/Insert keys output Wordstar commands. 
So you can close this issue as resolved. 
The one that isn't fully resolved is moving in the official ESP32 SD driver into RunCPM so all the SD cards work properly. I have a feeling based on the demo code it might also run faster. I guess I have a bit of OCD too :)

MockbaTheBorg, 2019-08-09:
This is pretty cool. You should blog about it. (or brag about it, :))
The ESP32 boards are really cool. I think they can be considered the most useful platform RunCPM runs on.

closed 2019-08-09 by MockbaTheBorg

moxhamj, 2019-08-13:
Update re Kermit. Using the generic version as no need to change things. At one end
KERMIT SEND MYFILE.TXT     and at the other end
KERMIT RECEIVE MYFILE.TXT
But sometimes it would work and sometimes not. Of course, it helps to read the instructions (and to use some debug code).  The answer is on page 16 of the Kermit Users Guide:
"The key point is to start the receiving end first -- some microcomputer Kermits do not include a timeout facility, and if the receiver is not ready to receive when the sender first sends, there will be a protocol deadlock."
Another thing buried in the instruction manual is that if you do start the Kermit programs in the wrong order and both programs are hanging, hit Enter on the sending Kermit. This wakes the receiver up by resending the start packet. 
[KERMIT.zip](https://github.com/MockbaTheBorg/RunCPM/files/3494941/KERMIT.zip)