MockbaTheBorg/RunCPM #85: Console redirection and adding extra ports on the ESP32

moxhamj, 2019-08-11:
This isn't really an 'issue' so can be closed by Mr Mockba, but I thought I might write some notes about the code that was changed to enable Kermit file transfers.

1) The board I am using is called ESPDUINO-32 and if you google this and select images, you can see it is an arduino form factor. I specifically went for the one with the large USB socket as this is plugged and unplugged many times and is mechanically stronger than the small ones. This particular board is not listed in the arduino IDE but it seems to compile ok with many of the ones that are listed. 

2) Near the beginning of the RunCPM module are some ifdef's, I changed the pins for the SD card to 19,23 and 18, the CS pin is 5, and the led is on pin 2.
```
#elif defined ESP32 // ESP32 boards
  SdFatSoftSpiEX<19, 23, 18> SD; // MISO, MOSI, SCK Some boards use 2,15,14,13, other 12,14,27,26
  #define SDINIT 5 // CS
  #define LED 2 // TTGO_T1=22 LOLIN32_Pro=5(inverted) DOIT_Esp32=2 ESP32-PICO-KIT=no led, Espduino pin 2
  #define LEDinv 0
  #define BOARD "ESPDUINO-32"
```
3) in void setup() I added the two new serial ports (the ESP32 has three uarts. Just a thought here, the DUE has 4 ports so this idea could also work on that board too). 
```
#ifdef ESP32  
  Serial1.begin(9600,SERIAL_8N1, 26, 25);    //Baud rate, parity mode, RX, TX  ESP32
  Serial2.begin(9600,SERIAL_8N1, 17, 16);    //Baud rate, parity mode, RX, TX  ESP32
#endif  
```
4) For hardware I used Max3232 modules available on ebay. Run from the 3V3 pin. In keeping with the old-skool spirit, computers have male D9 plugs and devices (modems, terminals) are female D9. The female Max3232 modules are much more common - I had to search a bit to get the male ones. I made a crossover cable with two female sockets and pin 5 to 5, 2 to 3 and 3 to 2. 

5) I got a copy of generic Kermit from Columbia University. It is in the CP/M archives. This needs no modification at all. If anyone has problems getting this I'll post a copy here.

6) Below is a wall of code. This goes in the abstraction_arduino.h file. It needs to go above the console abstraction functions (I think this is because functions need to be declared before you use them, and this applies in .h files but not .ino files. ). Once this code is copied in, test everything still compiles ok. None of this should affect compilation as it isn't linked yet to any of the existing code. 

```
// ***************************** Physical Device List **************************************
//CON: = TTY: CRT: BAT: UC1:
//RDR: = TTY: PTR: UR1: UR2:
//PUN: = TTY: PTP: UP1: UP2:
//LST: = TTY: CRT: LPT: UL1:

// TTY

void deviceTTYout(uint8_t ch)  // same as _putch 
{
  Serial1.write(ch); // same as _putch but with buffer overflow protection
  while (Serial1.availableForWrite() < 20) {
      delay(100); // at 9600 baud, ESP32 output buffer is 128 bytes, if stop when only 20 chars left in the output buffer, and pause 100ms, then have about 115 left as sending about 1 per ms.
  }
}

uint8_t deviceTTYavailable(void) {  // same as _kbhit
  return(Serial1.available());
}

uint8_t deviceTTYin(void) {  // same as _getch
  while (!Serial1.available());
  return(Serial1.read());
}

uint8_t deviceTTYinEcho(void) { // same as _getche
  uint8_t ch = deviceTTYin();
  deviceTTYout(ch);
  return(ch);
}

// CRT

void deviceCRTout(uint8_t ch)  // same as _putch 
{
  Serial.write(ch); // same as _putch but with buffer overflow protection
  while (Serial.availableForWrite() < 20) {
      delay(100); // at 9600 baud, ESP32 output buffer is 128 bytes, if stop when only 20 chars left in the output buffer, and pause 100ms, then have about 115 left as sending about 1 per ms.
  }
}

uint8_t deviceCRTavailable(void) {  // same as _kbhit
  return(Serial.available());
[... 283 lines cut ...]
  //Serial.write(ch);
  //return(ch);
  return echoCON(); // redirect based on iobyte
}

void _putch(uint8_t ch) {
  //Serial.write(ch);
  outCON(ch); // redirect based on iobyte
}

void _clrscr(void) {
  Serial.println("\e[H\e[J"); // done once at startup so left as is
}
```

8) Looking back through the code. I think that is pretty much it. I did end up adding in a tiny bit of code in the cpm.h function to redirect to the reader device, however in the code above the reader device doesn't do anything. But this does give an idea how to add in other devices. 
```
#ifdef ESP32
    HL = inRDR();
#else    
    HL = 0x1a; // ^Z end file
#endif
``` 

9) One small bug I am working on - kermit works rock solid between two boards, but when using teraterm with a kermit transfer, it seems to have many dropped packets. This is true at baud rates from 1200 to 115200. I have tried different Max3232 modules, different USB to serial devices and different boards. I haven't tried different terminal programs. This may not be such an issue, because the *huge* advantage of RunCPM over all other real and emulated CP/M systems is that it uses Fat32 files on the SD card. Thanks++ to Mr Borg for this :)

MockbaTheBorg, 2019-08-12:
I will try to merge this onto the main code. Cool stuff.

closed 2019-08-12 by MockbaTheBorg

moxhamj, 2019-08-12:
Sorry, one more bit that needs adding from the bdos. The punch output needs to be redirected. This bit is a bit messy as I simply commented out the part that redirects to files. ?? redirect to files except on the esp32. Or put in an elif if the disk redirect is not selected. I'm not sure about how to add this. In any case the generic version of Kermit sends output to the punch device, and receives data via a redirected console device. 
```
	case 4:
//#ifdef USE_PUN
//		if (!pun_open) {
//      pun_dev = _sys_fopen_w((uint8*)pun_file);
//			pun_open = TRUE;
//		}
//		if (pun_dev)
//			_sys_fputc(LOW_REGISTER(DE), pun_dev);
//#endif
#ifdef ESP32
    outPUN(LOW_REGISTER(DE));  
#endif    
		break;
		/*
		C = 5 : Printer output
		*/
	case 5:
//#ifdef USE_LST
//		if (!lst_open) {
//			lst_dev = _sys_fopen_w((uint8*)lst_file);
//			lst_open = TRUE;
//		}
//		if (lst_dev)
//			_sys_fputc(LOW_REGISTER(DE), lst_dev);
//#endif
#ifdef ESP32
    outLST(LOW_REGISTER(DE));  
#endif    
		break;
```