MockbaTheBorg/RunCPM #86: Tiny Terminal

moxhamj, 2019-08-11:
I built a tiny terminal using a 128x64 oled and a PS2 keyboard and an Arduino Uno. It is good enough for STAT and MBASIC and Kermit and other simple CP/M testing. I'm building a bigger display for Wordstar etc. It is pretty simple to put together with some jumper wires. 
I know there are some CP/M versions out there that combine display and CP/M but I kind of like the idea of a generic box with a D9 socket and small display and PS2 socket and you can use it on different machines.
This issue can be noted as closed once read :)
Code below. 
```

// Tiny terminal Arduino with I2C display, keyboard and RS232
// PS2 keyboard powered by 5V, data pin 2, clk pin 3
// I2C display on SCL and SDA ‎A4 (SDA), A5 (SCL)
// Max3232 or Max2321 module powered by 5V, on Tx and Rx pins 0 and 1 (disconnect Rx when programming)
// for keyboard, // https://github.com/techpaul/PS2KeyMap/
// for display, #include <Adafruit_GFX.h> and #include <Adafruit_SSD1306.h>
// Scrolling screen, backspace but no VT100 cursor movement
// 21x6 display 
// US keyboard but can change this if needed to UK, French or German (see code).
// outputs an ascii code for keypress. 9600 baud
// also output the control characters eg ^A outputs ascii 1, ^B ascii 2. 
// arrow and home keys mapped to wordstar ^S^D etc
// caps lock, num lock work as do leds on the keyboard when press these
// function keys not mapped as don't map to any ascii values, could potentially map to ascii 128 to 255

/*  KeyboardPS2AdvancedSimpleTest.ino

  PS2KeyAdvanced library example

  Advanced support PS2 Keyboard to get every key code byte from a PS2 Keyboard
  for testing purposes.

  IMPORTANT WARNING

    If using a DUE or similar board with 3V3 I/O you MUST put a level translator
    like a Texas Instruments TXS0102 or FET circuit as the signals are
    Bi-directional (signals transmitted from both ends on same wire).

    Failure to do so may damage your Arduino Due or similar board.

  Test History
    September 2014 Uno and Mega 2560 September 2014 using Arduino V1.6.0
    January 2016   Uno, Mega 2560 and Due using Arduino 1.6.7 and Due Board
                    Manager V1.6.6

  This is for a LATIN style keyboard using Scan code set 2. See various
  websites on what different scan code sets use. Scan Code Set 2 is the
  default scan code set for PS2 keyboards on power up.

  Will support most keyboards even ones with multimedia keys or even 24 function keys.

  The circuit:
   * KBD Clock (PS2 pin 1) to an interrupt pin on Arduino ( this example pin 3 )
   * KBD Data (PS2 pin 5) to a data pin ( this example pin 4 )
   * +5V from Arduino to PS2 pin 4
   * GND from Arduino to PS2 pin 3

   The connector to mate with PS2 keyboard is a 6 pin Female Mini-Din connector
   PS2 Pins to signal
    1       KBD Data
    3       GND
    4       +5V
    5       KBD Clock

   Keyboard has 5V and GND connected see plenty of examples and
   photos around on Arduino site and other sites about the PS2 Connector.

 Interrupts

   Clock pin from PS2 keyboard MUST be connected to an interrupt
   pin, these vary with the different types of Arduino

  PS2KeyAdvanced requires both pins specified for begin()

[... 515 lines cut ...]
  if (serialHead >= serialBufferSize) {
    serialHead = 0;
  }
}

uint8_t readSerialBuffer()
{
  uint8_t ch;
  ch = serialBuffer[serialTail];
  serialTail++;
  if (serialTail >= serialBufferSize) {
    serialTail = 0;
  }
  return ch;
}

boolean availableSerialBuffer()
{
  boolean availableBuffer = false;
  if (serialTail != serialHead) { 
    availableBuffer = true;
  }
  return availableBuffer;
}
```

drawkula, 2019-08-11:
[\o/
Creative Compulsive Disorder!
Definitely!
;-D](https://vimeo.com/80973511)

MockbaTheBorg, 2019-08-12:
This is freeking AWESOME!!! .. :)

closed 2019-08-12 by MockbaTheBorg