MockbaTheBorg/RunCPM #87: Speed tests

moxhamj, 2019-08-13:
Has anyone done any speed tests? I'm wondering what the ESP32 runs at in terms of a real Z80 clock speed. It seems a little slow (but then, I got a bit spoiled with Grant Searle's Cyclone II board which runs like a 50Mhz Z80). On the ESP32, loading up, say MBASIC takes 6 seconds - and I'm wondering how much of this is disk activity and how much is CPU activity. Or maybe I've just forgotten how slow real floppy drives were!
Quick test on mbasic doing 10,000 multiplies in a loop takes 7 seconds. 
The ESP32 reports a clock speed of an insane 240Mhz, and even allowing 20 instructions per Z80 instruction should still give a decent Z80 speed.

MockbaTheBorg, 2019-09-05:
I think most of the time is spent on I/O ... in the Teensy 3.6 is where I observe the fastest I/O speed, using the same uSD card.
I wonder what would be the best method for accurately comparing the speeds with a real Z80, other than just running things and using a chronograph.

MockbaTheBorg, 2019-09-20:
RunCPM on Windows has a PROFILE define, which when defined allows for timing the execution of CP/M programs ... we could develop something similar for timing it on other platforms. But on a regular Z80 we might need something different.

closed 2019-09-20 by MockbaTheBorg

MockbaTheBorg, 2019-09-30:
So I would say that on our emulations we are reaching 22Mhz or more of equivalent Z80.

moxhamj, 2019-10-01:
22Mhz? That is very impressive!

MockbaTheBorg, 2019-10-01:
Indeed ... I was comparing to this board I am eyeing on eBay, which runs CP/M at 22Mhz. 
My Arduino DUE took 2 minutes and 45 seconds to run the same test. Very slow, comparatively, but still faster than a real Z80 running at 4Mhz.
The ESP32 boards I would say are performing similarly to the 22Mhz one.
They are faster than the Teensy 3.5 (overclocked) and definitely slower than the Teensy 3.6 (overclocked) which is the fastest one, outperforming the 22MHz Z80.
I will probably test on more devices here as I get them to run.

MockbaTheBorg, 2019-10-02:
My Zeta2 (8Mhz) did it in 2 minutes and 32 seconds. A little faster than the Arduino Due.
If we overclock the Due it will probably match.

mecparts, 2019-10-05:
Having just got RunCPM running on a Teensy 4.0, I was wondering how fast it was. It ran this test in under 8 seconds. If I'm doing the math correctly, that's 143MHz. I am blown away...

MockbaTheBorg, 2019-10-05:
WOW!!!

Damn I need to get myself one.

mecparts, 2019-10-05:
It would have been nice to have a microSD connector and been able to use SDIO, I agree. But I guess there's only so much they could do in 18x36mm. (and for what they've delivered for $20, I'm prepared to forgive not getting everything I would have liked!) There's even pads on the bottom of the board that look like you could solder a microSD connector to... but apparently it's a sham. Based on my not very extensive 10-minute web search, you're supposed to solder wires to it and use an off board adaptor.

Since the el cheapo microSD adapter I had was SPI only, that's the route I took. Used the stock SPI pins: SCK(13), CS(10), MOSI(11) and MISO(12). The SdFat library took a bit of massaging as it apparently doesn't support the Teensy 4.0 just yet (neither SDIO nor SPI)

```
SdFat/src/SpiDriver/SdSpiTeensy3.cpp
***************
*** 23,31 ****
   * DEALINGS IN THE SOFTWARE.
   */
  #include "SdSpiDriver.h"
! #if defined(__arm__) && defined(CORE_TEENSY)
  // SPI definitions
  #include "kinetis.h"
  
  //------------------------------------------------------------------------------
  void SdSpiAltDriver::activate() {
--- 23,33 ----
   * DEALINGS IN THE SOFTWARE.
   */
  #include "SdSpiDriver.h"
! #if defined(__arm__) && defined(CORE_TEENSY) && !defined(__IMXRT1062__)
  // SPI definitions
  #include "kinetis.h"
+ #endif
+ #if defined(__arm__) && defined(CORE_TEENSY)
  
  //------------------------------------------------------------------------------
  void SdSpiAltDriver::activate() {

```

There would obviously have to be more done here if you wanted to go the SDIO route...

I also had to add a few bits in RunCPM.ino  as well to support the Teensy 4.0 and the SPI adapter. And it didn't like running at whatever default speed the SdFat library was trying either; 50MHz didn't work, but 25MHz did.

```
***************
*** 21,26 ****
--- 21,32 ----
    #define LED 22 // TTGO_T1=22 LOLIN32_Pro=5(inverted) DOIT_Esp32=2 ESP32-PICO-KIT=no led
    #define LEDinv 0
    #define BOARD "TTGO_T1"
+ #elif defined __IMXRT1062__ // Teensy 4.0
+   SdFat SD;
+   #define SDINIT 10
+   #define LED CORE_LED0_PIN
+   #define LEDinv 0
+   #define BOARD "TEENSY 4.0"
  #elif defined CORE_TEENSY // Teensy 3.5 and 3.6
    SdFatSdio SD;
    #define SDINIT
***************
*** 114,119 ****
--- 120,127 ----
        _puts("\nFile System initialization failed.\n");
        return;
      }
+ #elif defined __IMXRT1062__ 
+   if (SD.begin(SDINIT, SD_SCK_MHZ(25))) {
  #else
    if (SD.begin(SDINIT)) {
  #endif
```
There may well be better ways to handle all of the above; what you see is just what I came up with 2am this morning after discovering RunCPM only a few hours earlier! Still, it was good enough to get an A0> prompt within a few minutes of running in the door with my brand new Teensy this afternoon, so I'm content at the moment.

MockbaTheBorg, 2019-10-10:
My RC2014 running at 7.37 MHz: 1 min 42 sec. Got it out the back of a shelf just to run this code ... :)

mecparts, 2019-10-10:
Ampro LB+ @ 4MHz: 4 minutes 51 seconds. I'm not sure what surprised me more: a) that I still had the board and a couple of 5.25" floppy drives, b) that they all worked after gathering dust in the basement for the last 28 years, c) that I still had the last set of backup floppies and knew where they were, or d) that every single disk read without a single error. But playing with some of this stuff at 140MHz+ is going to be fun!

MockbaTheBorg, 2019-10-10:
Holy cow ... those boards never show up on eBay ... I look like every other week ... lol
I have it emulated on the M.E.S.S.

wrljet, 2019-10-10:
I bought one of the Ampro Little Boards back when they were new, '83 or '84 I guess.  Assembled the kit, and it didn't work.  Couldn't figured out why.  In 2013 I stumbled upon it, discovered a bad IC socket was the problem, and got it working.  I still had the disk(s) and used a drive from a vintage PC.
Played with it a while and then sold it on eBay 12/2013 for US$157.
-Bill

mecparts, 2019-10-10:
I've had my LB since the mid 80s. It saw use for many years alongside a Cromemco Z2H, after the Xerox 820-I, and before the first PC. I've dug up everything except the Project Board. Manuals, source disks, my notes, the lot. Reading some of my old notes, I realise I'd forgotten just how, umm, "interesting", a lot of the old hard drives were to work with.

Just finished installing my last modified BIOS, NovaDOS and ZCPR3.3. Ahh, the memories. Am I going to try plugging the old 40MB ST-251 back in? Must... resist... get back to Teensy and RunCPM!

And to all those physicists trying to figure out where all the missing mass in the universe is, a whole bunch of it is still sitting in that Z2H. It's black, it's heavier than sin, and probably never leaving this house - it's way too heavy for me to move by myself!