MockbaTheBorg/RunCPM #88: BDOS function 17: Search for First Name Match with '?' in drive byte errors out

mecparts, 2019-10-16:
I'm having a problem with RunCPM's implementation of BDOS function 17 (Search for First Name Match). Unless I'm not reading the code correctly (which is entirely possible), it doesn't seem to allow for the 'search for files in all user areas on the default disk' operation.

Quoting from the description of BDOS function 17 in "The Programmer's CP/M Handbook" by Andy Johnson-Laird (which I've found to be quite reliable over the years):

> The first byte of an FCB normally contains the logical disk number code. A value of 0 indicates the default disk, while 1 means disk A, 2 is B, and so on up to a possible maximum value of 
> 16 for disk P. However, if this byte contains a '?', the BDOS will search the default logical disk and will match the file name and type regardless of the user number.

With a disk number code in the first byte, the call behaves as I expect it to. But when I try it with a '?' in there to search for matching files in all user areas on the default disk, it crashes out with a "Bdos Error on A : Select" error.

I've looked at the code for _SearchFirst and it calls _SelectDisk with the value in the first FCB byte and doesn't seem to allow for a '?' there (and so would seem to be the source of the select error).

Apologies in advance if this just a PEBCAK issue. This is the first time I've looked at CP/M in many, many, many years and so I could be missing something incredibly obvious.

The reason I'm looking at this is that I was trying to get ZCPR3's FINDF utility working. It searches for files by going to the BIOS and examining directory sectors (which of course won't work under RunCPM) and I'd thought I could accomplish the same thing by using BDOS 17 & 18 (find first and find next).

If it matters, this is on a Teensy 4.0 so I'm working with the Arduino end of things. But if there actually is a problem, I think it's happening long before control gets into the Arduino abstraction level.

MockbaTheBorg, 2019-10-16:
This is definitely working as designed. Not pebcak.
I will take a look later today to see if I can implement like this.

Thanks,
Marcelo.

MockbaTheBorg, 2019-10-22:
So ... I did check the code to try to make this happen and it doesn't look trivial. Will take me some time to figure out a way to do it.

Therefore I cannot give an ETA right now.

mecparts, 2019-10-22:
In a way, I'm happy to hear you say that it's non trivial. I had had a look at doing it and while it sounded fairly simple at a high level, implementing it without having an actual CP/M directory track to search in appeared to be non trivial. I wasn't sure if that was due just to me not being familiar with the code or that it was in fact not easy (probably some of both). Getting all the ZCPR3.3 segments set up correctly on a cold boot was easy in comparison.

At the highest level, it's simple enough:

```
if fcb[0] is '?' then
  search for file in any user area on current default disk
else
  search for file in current user area on disk in fcb[0]
fi

```
In CP/M, the major difference between the two alternatives is whether or not you look at the first byte of the directory entry (where the user number of the file is stored). In the first case, you don't, and in the second case, you do.

But with user numbers emulated by sub directories, the first case becomes "search every sub directory on the current default disk for this file" and the second case is "search the user number sub directory on the selected disk for this file". Two very different things.

I'll keep poking at things working up from the Arduino abstraction end and see if anything occurs to me there.

MockbaTheBorg, 2019-10-22:
Thanks.

Actually I am wondering here also what should be the behavior for when we
use the find next function like that. After we would get the find first
result, should we assume that the folder (user number) where we found the
file becomes now the new default?
I couldn't find what should be the correct behavior on the CP/M
documentation.

mecparts, 2019-10-22:
I just fired up the Ampro and wrote a quick Turbo Pascal program to see what CP/M did. It just went in the order it found them in the directory sectors, which was what I thought it would do. If I looked for ?FILE??????? and the directory order was 5:FILE1, 3:FILEX, 5:FILE2 that's how they were returned.

MockbaTheBorg, 2019-10-22:
So the idea would be:

Look for (find)first ?:FILE???????.
start looking on user 0 and so on until found on 5:FILE.

As we look for the (find)next ones we would find anything ?:FILE??????? from user 5 on until user 15.

The big problem with this is that I actually use the underlying operating system findFirst and findNext  (or whatever they are named) functions, and they end search at the end of the folder.
I would need to, at end of each folder, see if the FCB holds '?' at the first byte and actually increment the user and start the process again. Is this it?

This might be feasible ... I need to find a way to do it that would actually be easily portable among all the possible platforms.

Any chance you can share that Turbo Pascal program?

Thanks!

mecparts, 2019-10-22:
Okay, this is either both easier and stranger than I originally thought, or I'm doing something incredibly dumb.

Here's the Pascal program I threw together to see exactly how CP/M handles BDOS 17 & 18 with a question mark in the first byte.

```
PROGRAM directory;
   TYPE
     Str2 = STRING[2];
   CONST
      fcb : ARRAY[0..35] OF CHAR='?FILE?      '#00#00#00#00#00#00#00#00#00#00#00#00#00#00#00#00#00#00#00#00#00#00#00#;
      FindFirst = 17;
      FindNext = 18;
      SetDMA = 26;
   VAR
      a : BYTE;
      i,j,dmaIndex : INTEGER;
      c : CHAR;
      dma : ARRAY[0..127] OF BYTE;
      buf : ARRAY[0..15] OF BYTE;

   FUNCTION tohex(b : BYTE) : Str2;
   CONST
      Hex : ARRAY[0..15] OF CHAR='0123456789ABCDEF';
   BEGIN
     tohex := Hex[b DIV 16]+Hex[b MOD 16];
   END;

   BEGIN
      BDOS(SetDMA,ADDR(dma));
      a := BDOS(FindFirst,ADDR(fcb));
      WHILE a in [0..3] DO BEGIN
         dmaIndex := a*32;
         FOR i := 0 TO 31 DO BEGIN
            buf[i] := dma[dmaIndex];
            dmaIndex := SUCC(dmaIndex);
         END;
         FOR i := 0 TO 1 DO BEGIN
            FOR j := 0 TO 15 DO
               WRITE(tohex(buf[i*16+j]),' ');
            WRITE(' ');
            FOR j :=  0 TO 15 DO BEGIN
               c := CHR(buf[16*i+j]);
               IF c IN [' '..'~'] THEN
                  WRITE(c)
               ELSE
                  WRITE('.');
            END;
            WRITELN;
         END;
         a := BDOS(FindNext,ADDR(fcb));
      END;
   END.

```

And when I ran it on a disk that I'd just formatted and scattered some files call FILE around on, this is what I saw:

```
00 46 49 4C 45 20 20 20 20 20 20 20 00 00 00 02  .FILE       ....
2D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  -...............
06 46 49 4C 45 20 20 20 20 20 20 20 00 00 00 02  .FILE       ....
2E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
01 46 49 4C 45 20 20 20 20 20 20 20 00 00 00 02  .FILE       ....
2F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  /...............
02 46 49 4C 45 20 20 20 20 20 20 20 00 00 00 02  .FILE       ....
30 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  0...............
0F 46 49 4C 45 20 20 20 20 20 20 20 00 00 00 02  .FILE       ....
31 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  1...............
0A 46 49 4C 45 20 20 20 20 20 20 20 00 00 00 02  .FILE       ....
32 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  2...............
[... 39 lines cut ...]
In a word, huh?

So I went back and re-read Johnson-Laird:

> However, if this byte contains a '?', the BDOS will search the default logical disk and will match the file name and type regardless of the user number.

Now, that says to me that BDOS is still going to match the name and file type. But that's not what I'm seeing.

So I went and looked up BDOS 17 in Cortesi's _Inside CP/M_. And here's what it had to say:

> Normally, _dr_ is ignored and only files with the current user code are matched. If _dr_ contains a question mark, all directory entries of any user code, and entries of any type, including those not in use, are compared.

That still sounds like it's going to match the file name and type, but it's not quite so clear cut. I could read that to mean that I'm going to get all directory entries ever used.

So then I read the entry from [http://seasip.info/Cpm/bdos.html](http://seasip.info/Cpm/bdos.html) and it said: 

> If the first byte of the FCB is ?, then any directory entry (including disc labels, date stamps etc.) will match. 

Now that sounds like what I'm seeing: a ? in the first byte of FCB is getting me every directory entry, regardless of what's in the file name and type.

So what does all that mean? Well, if I haven't done something dumb, and that's the way BDOS 17/18 are supposed to work with a ? in the drive byte, that would explain why programs like FINDF don't even bother with those BDOS calls and go straight to the directory track; it doesn't really buy you anything to go through BDOS.

If this is how it's actually supposed to work, I honestly don't think it's worth a big investment of time to implement it in RunCPM. If it simply wildcarded the user number and returned files that matched the file name and type, that'd be useful in certain cases. But this? Pfffft! I'm think I'm better off writing a replacement for FINDF that just cycles through all the user subdirectories on each allowable disk and using Find First/Find Next on each subdirectory.

(I'm still half expecting to hear "no, it's supposed to work the way you originally read/though... you just made a rookie mistake in your program")

mecparts, 2019-10-27:
I think I may have spoken too quickly. The more things I get running under ZCPR3.3, the more things I run into that use the search first/search next with the wildcarded user. I guess not everything went straight to the directory tracks.

And from reading the source code of the utilities that are using it, the behaviour of "return all directory entries, regardless of filename matching or even if it's a deleted entry", is indeed the way things are supposed to work.

With a couple of hours of work today, I've got about half of it working on the Arduino side of things. I've had to zero out the 32 bytes of the directory entry in _findnext() before populating it with the filename, type and user number. ZFIND was checking the extent number and would only match on the first (zero) extent of a file. And there are a few tweaks in _SelectDisk() and _FCBtoHostname() but nothing too huge. XDIR is no longer throwing a select error and is showing a directory listing, though it is showing all files with 0K (a consequence of zeroing the directory entry). One thing at a time...

Fingers crossed, I may have the rest of it figured out tonight.

MockbaTheBorg, 2019-10-27:
Cool. Feel free to share your findings and changes, let's make sure they
are portable out of the arduíno, and I will add them to the main code.

Thanks,
Marcelo

mecparts, 2019-10-27:
Will do. I spent the last few hours chasing a problem that wasn't caused by the changes I'm making, but rather a 27 year old bug in ZFIND. I'm glad I've got the Ampro to check against! Fixing, assembling and linking ZFIND was fun as the author used Echelon's ZAS and ZLINK (neither of which I have or can locate) and converting some of its idiosyncrasies was... yes, "fun", I think is the word I'll use in public.

mecparts, 2019-10-28:
Here's the changes. I've included a diff and attached a zip of the 4 modified files.

_SearchFirst() and _SearchNext() call two different functions depending on whether the drive byte is set to the 'all users' wildcard. ('?').

_SelectDisk is modified to select the current default disk if the drive byte is either 0 or a '?'.

_findfirstallusers()  opens the drive directory rather than the user subdirectory, and calls _findnextallusers().

_findnextallusers() looks to see if it has a user directory open. If not, it searches for the next subdirectory that's a single character and is '0'..'9' or 'A'..'F' and opens it. Once it has a user subdirectory open, it calls _findnext().

_findnext() has been augmented to produce a "closer to the real thing while still being phoney" directory entry. Besides populating the user number in the directory entry **if** we're searching all users, it also sets the extent up properly. and does what it can (within the limits of only returning one directory entry per file) for the record count. It also phonies up a block map so utilities that might be aware of sparse files don't get confused.

The file sizes that get reported are correct only for files that would occupy a single directory entry (physical extent) on an actual CP/M system. Larger files report a truncated size. I do have an idea or two about how to have Search First/Search Next return the proper number of directory entries for files that are big enough to need them. But I decided that this was enough to stand on its own for the time being.

I also modified the disk parameter block in cpm.h to allow for more directory entries (up from 256 to 1024). I noticed that most all directory programs were unhappy when there were more than 256 files in a user directory (their buffers overflowed). I also bumped up the disk size to 8M, and zeroed out the directory check size, assuming that we're treating the SD card like a non-removable hard drive rather than a floppy. I also calculate a handful of new globals from data in the disk parameter block so that _findnext() can use them to make a good forgery of a legitimate directory entry.

I don't know how hard the changes in the abstraction layer will be to port. As long as the target system can scan subdirectories, and get the file size, I don't think it'll be too bad.

Here's the diff below. The zip follows at the end.

Wayne

```
--- /Original/abstraction_arduino.h
+++ /Modified/abstraction_arduino.h
@@ -18,6 +18,11 @@
 #define HostOS 0x06
 #endif
 
+#include <ctype.h>
+
+#define BlkSZ 128 // CP/M block size
+#define BlkEX 128 // Number of blocks on an extension
+
 /* Memory abstraction functions */
 /*===============================================================================*/
 bool _RamLoad(char *filename, uint16 address) {
@@ -35,7 +40,7 @@
 
 /* Filesystem (disk) abstraction fuctions */
 /*===============================================================================*/
-File root;
+File rootdir,userdir;
 #define FOLDERCHAR '/'
 
 typedef struct {
@@ -47,6 +52,14 @@
   uint8 cr, r0, r1, r2;
 } CPM_FCB;
 
+typedef struct {
+  uint8 dr;
+  uint8 fn[8];
+  uint8 tp[3];
+  uint8 ex, s1, s2, rc;
+  uint8 al[16];
+} CPM_DIRENTRY;
+
 File _sys_fopen_w(uint8 *filename) {
   return(SD.open((char *)filename, O_CREAT | O_WRITE));
 }
@@ -294,19 +307,59 @@
   uint8 result = 0xff;
   uint8 dirname[13];
   bool isfile;
+  uint8 blocks;
+  uint16 fileRecords,fileExtents;
+  uint32 bytes;
 
   digitalWrite(LED, HIGH^LEDinv);
[... 385 lines cut ...]
+   }
   return(result);
 }
 

--- /Original/globals.h
+++ /Modified/globals.h
@@ -161,6 +161,14 @@
 static uint8  userCode = 0;   // Current user code
 static uint16 roVector = 0;
 static uint16 loginVector = 0;
+static bool    allUsers = false; // true when dr is '?' in BDOS search first
+static uint8   currFindUser = 0; // user number of current directory in BDOS search first on all user numbers
+static uint8   blockShift;       // disk allocation block shift
+static uint8   blockMask;        // disk allocation block mask
+static uint8   extentMask;       // disk extent mask
+static uint16  firstBlockAfterDir; // first allocation block after directory
+static uint16  numAllocBlocks;   // # of allocation blocks on disk
+static uint8   extentsPerDirEntry; // # of logical (16K) extents in a directory entry
 
 #define tohex(x)  ((x) < 10 ? (x) + 48 : (x) + 87)

``` 

[modified.zip](https://github.com/MockbaTheBorg/RunCPM/files/3776736/modified.zip)

MockbaTheBorg, 2019-10-28:
Cool. I will check later today.
I think it is a good idea to move to the hard drive paradigm.
I voided it initially following what I had in a comparison system and never
changed it.
I will check everything and start porting to the other hosts.

Cheers,
Marcelo.

MockbaTheBorg, 2019-10-28:
BTW ... are you still looking for ZAS/ZLINK? I have it.

mecparts, 2019-10-28:
> BTW ... are you still looking for ZAS/ZLINK? I have it.

Thanks; only if it's no trouble (very little of the old Echelon stuff seems to be out there). I worked around what I thought was an issue with the code and got it to compile with ZASM (aka Cromemco' ASMB) and link with plain old L80. I was thrown by a loop by a reference to $MEMRY which I didn't ever recall seeing before but apparently L80 handled it as ZLINK would have. It's funny, the things you forget. Some of them come right back as soon as you stumble over them, but others, it's like "nope, don't remember that at all."

MockbaTheBorg, 2019-10-28:
[J_0.zip](https://github.com/MockbaTheBorg/RunCPM/files/3780218/J_0.zip)

Here you go.

mecparts, 2019-10-28:
Thank you very much; I'm sure that will come in handy at some point.

MockbaTheBorg, 2019-11-08:
I have added this change only to Arduino platform so far. Didn't have a chance yet to port to the others.
Will probably need help to do so.

mecparts, 2019-11-08:
Has anyone tried RunCPM on a Raspberry Pi? I could try getting the POSIX/Linux portion working there if that'll help.

MockbaTheBorg, 2019-11-08:
It should work fine on any Linux. I have tried on the Radxa Pro and on a few FPGA I have (running Linux) and they all work just fine.

It seems to work fine too under Ubuntu-for-Windows (WSL for Windows 10).

mecparts, 2019-11-08:
Up and running on a Raspberry Pi 3B. That was even more painless than getting it running on the Teensy. 6 seconds for the speed test, by the way; 33% faster than the Teensy 4.0. I'm going to print out a copy of abstraction_posix.h and start looking at porting the changes.

mecparts, 2019-11-08:
I have the all users find first/find next working on Linux. Just bashing away at it to make sure I haven't missed anything.

It turned out to be even easier in some ways than on the Arduino side; glob quite happily took a path of the form "A/?/*" and searched all the user sub directories at once. In addition, I've put the code in to have _findnext() return the correct # of directory entries based on the file's size. Directory programs that show file size work properly.

I forked your repository and cloned it locally rather than just downloading a zip this time. I already use git here for other projects so it *should* be easier to integrate my changes now - I hope.

mecparts, 2019-11-09:
Okay, the code seems to be doing what I want it to so I've pushed the changes to my fork. Does it make any sense for me to do a pull request when only 2 of the platforms are supported in my changes or is there another way that would work better?

MockbaTheBorg, 2019-11-09:
That's cool.

Please do a pull request. I will check what you did for Linux and try to do
the same for Windows.

And I will also figure out how to apply a pull request, lol.

mecparts, 2019-11-09:
First time in a long time that I've done a pull request; I had to look up whether it was done from the forked directory or the master. I think I got it right.

MockbaTheBorg, 2019-11-09:
I have merged, made a few modifications and pushed v3.8. Please grab that versions and make further modifications to that one instead.
I have "frozen" 3.7 on my internal backup, so we should no make any changes to code on top of 3.7 anymore.

I will close this when I figure out the F17/F18 patch for the Windows version.

MockbaTheBorg, 2019-11-09:
BTW ... thanks a lot for the help. I need your first/last name to add to the credits ... :)

MockbaTheBorg, 2019-11-12:
@mecparts 
I have cleaned up the code a bit, so please test it out on Arduino and Linux to check if I broke anything.
I tried also to bring at least the mockup directory entry to the Windows port, but it is messing up the files which have more than 1 extension. Considering that it is almost 1am and I didn't sleep for half of last night, I might have overlooked some detail.
Not sure if you have access to a Windows machine, maybe you can give it a second set of eyes?

Thanks a lot!

mecparts, 2019-11-12:
I took a very quick look over the changes and saw a few instances of "yeah, that really should have been there in the first place". :) I'll give it a try tomorrow and let you know how it goes.

I went Windows free a long time ago, and didn't hang on to any of it, I'm afraid. I will take a look at what you've done for the Windows side and see if anything jumps out at me, but that's the best I can do.

Oddly, I distinctly remember replying to your message about needing my name but now I don't see my reply. I even checked some of the other issue threads and didn't find it there either. Anyhow, let me try that one more time.

Regards,
Wayne Hortensius

mecparts, 2019-11-12:
Good news: neither the Arduino nor Linux sides are failing obviously. Everything I tried seemed to work okay except for one oddity on the Linux side (but when I checked I discovered that it was there before the last set of changes). One particular directory program (XDIR3 v2.0) is giving file sizes that are 32K too large for a small percentage of files. All other programs return the correct size. And this is only on Linux; on the Arduino side, XDIR3 works fine. And even on Linux, XDIR3 works; it just reports the file size 32K too large in a few cases. As it's the only directory program with that issue I'm not terribly concerned, but I will keep it in the back of my mind in case something else shows up. My suspicion is that it's an issue with XDIR3's code rather than something in RunCPM.

I did find one extraneous line of code in cpm.h (which I think may have been me). At the very end of the DPB setup there's an extra SPT line. It shouldn't be there, but I don't think it's causing any problems; it's just a extra byte set off the end of the DPB.

I took a look at the Windows code and may have spotted your problem with files that occupy more than 1 directory entry. At the top of _findnext in abstraction_vstudio.h I expected to see this bit of code and didn't:

```
	if (allExtents && fileRecords) {
		_mockupDirEntry();
		result = 0;
	} else {
```

That's what returns the extra directory entries for files > 32K.. If what you're seeing is a maximum file size of 32K being reported, this is probably what's causing it.

Aside from that, the only things I found were a handful of spacing issues; really minor nits. I've updated my fork to include those and to remove the extra DPB line, but not the possible missing code in Window's _findnext, as I wasn't sure if what I found was what you were looking for.

I was able to crash the DR CCP on Linux just by running Turbo Pascal. After exiting TP, the CCP couldn't find any programs. The ZCPR3 CCP didn't have any problems. I don't know if that's an existing or new problem as I'd never built for the DR CCP before this.

MockbaTheBorg, 2019-11-12:
What I normally do is use the DR CCP as it should be the standard for CP/M
2.2 compatibility. If it works there then it should work anywhere.
I have an emulator called Z80emu which I match against to make sure
everything is working fine.
I will check the issue with Turbo Pascal. Which version was it?
Actually, I am thinking of creating a repository with my CP/M folders.
Wondering if this could pose a license issue. But it would be good to have
my folder online so other people could use the same programs I use here. I
have a lot of stuff, more or less organized.
I need to work more on the windows port as well.
Maybe do me a favor: add a little more comments to the posix code that
might help me translate your idea to Windows?
I am traveling for a couple weeks, but I will be working on it from the
hotels at night.

Thanks!

mecparts, 2019-11-12:
It was TP 3.01A. I just started it up, answered Y to include the message file, then exited. CCP builtin commands still worked after that, but any attempt to execute a transient resulted in a warm boot and nothing else.

TP is fairly aggressive in grabbing as much memory as it can for itself. And it likes to overwrite stuff in low memory as well. I did notice that it thinks it's all right to grab part of the CCP's memory. But it does do a warm boot when it exits, so it ought to reload the CCP... ah, found it. It's my fault. I assumed that a warm boot only had to reload the CCP, but it has to do the CCP and BDOS. I've taken out my "if (first_time)" code in _PatchCPM() and now TP doesn't crash DR's CCP. Sorry about that.

I've added some comments in the POSIX version of _sys_findnext(). The gist of it is fairly straightforward:

1. Did the user ask for all extents, and have we already found a file that filled a complete directory entry, and still has records left over?
2. Yes to both? Return the next physical extent (directory entry) and decrease the remaining number of records by how many were used in this entry.
3. No to either? Look for the next file.

Wayne

MockbaTheBorg, 2019-11-12:
Cool. I will give it a try.

Thanks!

mecparts, 2019-11-12:
I've put in a pull request so you can have _PatchCPM() reinitialise everything again.

MockbaTheBorg, 2019-11-12:
Cool... It's been some time since I went that deep into runcpm code. I need
to refresh my memory, lol.

MockbaTheBorg, 2019-11-13:
The missing piece of code seems to have resolved the sector counting problem.
Now I need to figure out how to to the scan on Windows. On Linux it indeed looks much easier. Easier than Arduino actually.

mecparts, 2019-11-13:
Glad to hear the missing bit of code was the solution for Windows.

Yes, the Linux scan was waaay easier than the Arduino one. Being able to say "give me all files of the form "A/?/*" made life so much easier. Python-style yields are bad enough with having to nest them.

Worst case, you can always do the scan on Windows the way I did it on Arduino, with two levels of scanning, one on the subdirectories of the current drive, and the second on the files in the current subdirectory. If you can do it that way you ought be able to reuse the logic the Arduino solution used.

MockbaTheBorg, 2019-11-13:
Yeah ... this is what I am thinking about ... look to me that the Arduino logic is indeed closer to that the Windows one should be.
I remember Windows disk logic was the one that gave me more headache ... but I will give it a try.
Problem is finding the willpower to do it ... lol.

MockbaTheBorg, 2019-11-13:
One strange thing I noticed is that, as I go around my folders doing STAT *.*, sometimes Windows agrees with Linux in terms of block count and free space, sometimes it doesn't, even though they are reading from the same folder.
And same with Arduino, sometimes they agree among them all, sometimes Windows and Arduino give me the same numbers, sometimes Linux and Arduino, and so on.
I have not yet been able to find a reason or explanation.

mecparts, 2019-11-13:
If STAT, like most of the DIRectory utilities I've run across, uses BDOS 27 to get the allocation vector and calculate the free space from that, remember that the *tmpFCB* falls right in the middle of the allocation vector (which is relatively benign, as the allocation vector itself is never actually set by anything). So you'd get different free space results depending on what file was in *tmpFCB* when the allocation vector was fetched.

I did briefly consider cobbling something up using a wildcard first first/find next and populating the allocation vector from that, until I realised that I've already got more than 8MB stored on a few of my drives. Oops...

MockbaTheBorg, 2019-11-15:
I took a stab with a sledgehammer on this issue yesterday. I think the implementation for Windows is now good enough. It was actually simple to implement, and maybe I can use the same format in the future for other versions.
Maybe I will revisit it sometime today to make it proper (it actually stops the search if there's a gap in the user area numbering).
But I have been thinking: 3.9 is probably going to be the last RunCPM version I release as it is. If/when I release 4.0 it will be a full rewrite of some of the elements, especially disk and console (so we can do file transfers via XMODEM/ZMODEM over the console).

closed 2019-11-15 by MockbaTheBorg

mecparts, 2019-11-15:
Good job, Marcelo. In every renovation, be it a house or a program, there's a point where you have to say, "That's close enough. What's next?".

If/when you ever decide to rip out and replace the console code, I've got a little Finite State Machine running in _kbhit() and _getch() that might be worthwhile finding a place for. It translates the VT-100 escape key sequences (the arrow keys, Page up/dn, home/end and ins/del) into Wordstar sequences. So now everything that understands WS sequences "just works" with the cursor keys. And it even turns itself off whenever 8 bit character I/O is enabled, so XMODEM/ZMODEM over the console wouldn't be impacted. It works flawlessly on the Arduino side but the character input on Linux doesn't seem to be as easy to get going (i.e. it didn't work the first time I tried it!) and I had other fish to fry.