MockbaTheBorg/RunCPM #89: BDOS function 15: (Open File): _FCBtoHostname needs to mask off high bit in each character before adding it to host filename

mecparts, 2019-10-31:
This time I bring both a problem and a solution.

I fired up the Dazzlestar disassembler this afternoon and was dismayed to see:

Not found:  DZENGHLP.OVR
Error D at address 1859

Well, argh, foo and bar. The overlay file was certainly there; all of them were. I went down to the old Ampro and tried running DZ there, and it came up with no problems. So I started digging. To make a long story short, I got to the point of printing out all each filename as RunCPM was asked to open it and spotted this:

OPENFILE - A/F/DZENGÈLP.OVR

Look very carefully at where the H should be. It's an accented character, meaning the high bit is set. Dazzlestar was setting the attribute bit of the F5 character (MP/M II: F5' - set to 1 for "unlocked" mode - other programs can use the file) while trying to open the file. I discovered later that wasn't the only place that Dazzlestar was passing attribute bits when trying to open a file; the object of the disassembly would be opened with F1 (public file) set.

So I opened up a BDOS source to see what happened there on a file open there, and when it was checking a character in the file name or type, there was an AND 07FH to mask off the attribute bit before the compare. So I added the same sort of thing into _FCBtoHostname, uploaded it and Dazzlestar was happy. The diff is below; this one's much simpler than the last.

```
--- original/disk.h
+++ modified/disk.h
@@ -78,6 +78,7 @@
 	CPM_FCB *F = (CPM_FCB*)_RamSysAddr(fcbaddr);
 	uint8 i = 0;
 	uint8 unique = TRUE;
+   	uint8 c;
 
 	if (F->dr && F->dr!='?') {
 		*(filename++) = (F->dr - 1) + 'A';
@@ -91,22 +92,24 @@
 
    if( F->dr != '?' ) {
 	   while (i < 8) {
-		   if (F->fn[i] > 32)
-			   *(filename++) = toupper(F->fn[i]);
-   		if (F->fn[i] == '?')
+         c = F->fn[i] & 0x7F;
+		   if (c > 32)
+			   *(filename++) = toupper(c);
+   		if (c == '?')
 	   		unique = FALSE;
    		++i;
 	   }
    	i = 0;
 	   while (i < 3) {
-   		if (F->tp[i] > 32) {
+   		c = F->tp[i] & 0x7F;
+   		if (c > 32) {
 	   		if (addDot) {
 				   addDot = FALSE;
    				*(filename++) = '.';	// Only add the dot if there's an extension
    			}
-   			*(filename++) = toupper(F->tp[i]);
+   			*(filename++) = toupper(c);
    		}
-   		if (F->tp[i] == '?')
+   		if (c == '?')
    			unique = FALSE;
    		++i;
    	}

```

MockbaTheBorg, 2019-11-01:
Cool. I bet that is undocumented.

I will make the change and upload to GitHub in the morning.

mecparts, 2019-11-01:
File attributes bits set in directory entries on CP/M don't surprise me at all; that's always how I've seen them used (SYS, R/O and PUBLIC in P2DOS derivatives). Finding them set in FCBs, yep, that was a surprise. I certainly wouldn't have anticipated seeing it.

Upon reflection, I guess it didn't matter to CP/M *where* the attribute bit was set; it had to be ignored in either case. So the sequence of XOR (HL) / AND 07FH that I've seen makes sense. "Compare the two 7 bit characters from the directory and the FCB, ignoring the attribute bits."

In my head, I can sketch out how to transfer the host OS file attributes to the attribute bits in the directory entry returned by _findnext(). Just that, and a 1 line change in match(), would be enough to get the SYS file attribute working. But R/O and (especially) PUBLIC would required a considerable amount of work in the routines that _bdos() calls, I think. And try as I might, I cannot figure out a way (at least on the Arduino side of things) to actually *set* file attributes in the OS file system, so it's a moot point. It's a pity; I'd *love* to have support for public files.

And in case I haven't said this already, I am really, really, *really* loving RunCPM. It's an amazing piece of software that you've created.

Wayne

MockbaTheBorg, 2019-11-08:
I have added this change to the code. Will upload soon.

closed 2019-11-08 by MockbaTheBorg