MockbaTheBorg/RunCPM #91: _sys_readrand() in abstraction_arduino.h should not extend the file

mecparts, 2019-11-07:
Marcelo, my apologies, but I've found another glitch.

Whenever I "printed" a file in WordStar 4.0, there was a whole lot of SD activity after the print had finished, and the source file size would balloon to just under 8MB.

For some reason I can't fathom, around the time that WordStar 4.0 finishes printing a file, it tries to do a random read to the source file's record # 65530.

Digging into the RunCPM code, _sys_randread() in abstraction_arduino.h would then happily extend the file out to that size. Result: a nearly 8MB file.

Which is shouldn't ought to be doing. Leaving aside the question of why WordStar 4.0 does what it does, a random read past the end of a file should not extend the file, it should return one of the following errors:

- Case 1: record # would be past the EOF but within the bounds of the last partially filled extent: error 1 (reading unwritten data)
- Case 2: record # would be past the bounds of the last partially filled extent but less than 65536: error 4 (seek to unwritten extent)
- Case 3: record # is greater than 65535: 6 (seek past physical end of disk)

I've verified this on the Ampro. (Is it ever handy having that beast to check what CP/M actually does rather than depending on my interpretation of the books I have!)

I think _sys_readrand() should look something like this (ExtSZ is #defined as (BlkSZ * BlkEX)):

```
uint8 _sys_readrand(uint8 *filename, long fpos) {
   uint8 result = 0xff;
   File f;
   uint8 bytesread;
   uint8 dmabuf[BlkSZ];
   uint8 i;
   long extSize;

   digitalWrite(LED, HIGH^LEDinv);
   f = SD.open((char*)filename, O_READ);
   if (f) {
      if (f.seek(fpos)) {
         for (i = 0; i < BlkSZ; ++i)
            dmabuf[i] = 0x1a;
         bytesread = f.read(&dmabuf[0], BlkSZ);
         if (bytesread) {
            for (i = 0; i < BlkSZ; ++i)
               _RamWrite(dmaAddr + i, dmabuf[i]);
         }
         result = bytesread ? 0x00 : 0x01;
      } else {
         if (fpos >= 65536L * BlkSZ)
            result = 0x06;      // seek past 8MB (largest file size in CP/M)
         else {
            extSize = f.size();
            // round file size up to next full logical extent
            extSize = ExtSZ * ((extSize / ExtSZ) + ((extSize % ExtSZ) ? 1 : 0));
            if (fpos < extSize)
               result = 0x01;   // reading unwritten data
            else
               result = 0x04;   // seek to unwritten extent
         }
      }
      f.close();
   } else {
      result = 0x10;
   }
   digitalWrite(LED, LOW^LEDinv);
   return(result);
}
```

There's two sets of changes. The first is just removing the _sys_extendfile() call at the beginning. The second is returning an error appropriate to the requested fpos rather than just 0x06 (which could never actually happen in the original code, as the file had already been extended to the requested record).

Wayne

MockbaTheBorg, 2019-11-08:
For this one I need to figure out the error codes. I remember reading the CP/M error coded somewhere when I was coding this.
And I need to remember the reason for the ExtedFile ... there was some gotcha there which I will have to remember.

mecparts, 2019-11-08:
Extending the file past eof makes perfect sense in the random write function, and that's indeed what CP/M does. But CP/M doesn't extend the file for a random read past the eof.

MockbaTheBorg, 2019-11-08:
Yeah. It's been years since I put it there. I won't remember what was the
quirk I was fighting. LOL

MockbaTheBorg, 2019-11-09:
I don't think this issue exists on Linux and Windows. I would need to test. But if your patch to Arduino solved the problem there, then we can probably close this one too.

mecparts, 2019-11-09:
I took a very quick look at the _sys_readrand() function in both Linux and Windows, and neither of them had a _sys_extendfile() before the _sys_openfile(); that was only only on the Arduino side. I think I may know why. It may come down to the definition of how seek works on the Arduino side vs. Linux and Windows. On the Arduino side, the docs for seek says: "Seek to a new position in the file, *which must be between 0 and the size of the file (inclusive)*" (italics mine). I have a hunch that there's no such restriction on Linux's seek and it may be able to seek beyond the eof. I'm going to think about this for a bit and run a few tests and I'll get back to you.

MockbaTheBorg, 2019-11-09:
Awesome, thank you.

mecparts, 2019-11-09:
Does this sound familiar at all?

You ran into a case where a program issued a Random Read past the end of file, followed up by a Sequential Write. Things didn't work properly on the Arduino side of things until you changed the code in Random Read to extend the file so that the Sequential Write could write at the record number specified in the Random Read.

I figured out what's going on.

When you set up the random access record number in FCB+33..35 and call Random Read, the BDOS computed the CR (Current Record, FCB+32) and *updated the FCB, even if it returned an error code*.

So if you follow up that Random Read with a Sequential Write, CP/M looks at the CR value and writes the record to spot specified in the Random Read.

It's the *write* that extends the file. But it explains perfectly why your initial approach of extending the file on the read likely worked for your case; the program really did want to write to that spot right afterwards. The case I ran across, with WS4 trying to read a ridiculous record that it had no intention of writing to, found the flaw.

So the solution is going to be something like this: DON'T extend the file in _sys_readrand() (and for that matter, in _sys-readseq()). DO update the CR, EX and S2 in _ReadRand() *even if _sys_readrand() returns a non zero error code*. I still have to investigate what CP/M does for error codes 4 (read to unwritten extent) and 6 (seek past physical end of disk).

I've attched the little Turbo Pascal program I wrote to watch what was going on, and a listing from when I ran it on the Ampro. I've still got to test those other 2 cases to see if they update the FCB, but I thought you'd like to know where things stood.

Wayne

[bdos.zip](https://github.com/MockbaTheBorg/RunCPM/files/3827757/bdos.zip)

```
Old TEST.BIN deleted

New TEST.BIN created

3 records written to TEST.BIN. FCB contents
00 54 45 53 54 20 20 20 20 42 49 4E 00 00 00 03   TEST    BIN    
93 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00                  
03 00 00 00                                          

TEST.BIN closed

TEST.BIN opened. FCB contents
00 54 45 53 54 20 20 20 20 42 49 4E 00 00 80 03   TEST    BIN    
93 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00                  
00 00 00 00                                          

ReadRand at EOF result: 1. FCB contents
00 54 45 53 54 20 20 20 20 42 49 4E 00 00 80 03   TEST    BIN    <-- RC doesn't change
93 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00                      (the file has not been extended)
03 03 00 00                                          <-- CR is positioned at the end of file

Wrote record #3 to TEST.BIN. FCB contents
00 54 45 53 54 20 20 20 20 42 49 4E 00 00 00 04   TEST    BIN    <-- RC changes after the write
93 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00                  
04 03 00 00                                          <-- CR is now positioned after the newly written record

ReadRand past EOF result: 1. FCB contents
00 54 45 53 54 20 20 20 20 42 49 4E 00 00 00 04   TEST    BIN    <-- again, RC doesn't change
93 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00                      (the file has not been extended)
09 09 00 00                                          

Wrote record #10 to TEST.BIN. FCB contents
00 54 45 53 54 20 20 20 20 42 49 4E 00 00 00 0A   TEST    BIN    <-- again, now RC changes
93 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00                      (the file has been extended)
0A 09 00 00                                          

TEST.BIN closed

Record #1
01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01                  
01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01                  
01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01                  
01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01                  
01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01                  
01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01                  
01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01                  
01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01                  

Record #2
[... 67 lines cut ...]
E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5                  

Record #9
E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5                  
E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5                  
E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5                  
E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5                  
E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5                  
E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5                  
E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5                  
E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5                  

Record #10
0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A     <-- Record #10, at its requested spot
0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A                  
0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A                  
0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A                  
0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A                  
0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A                  
0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A                  
0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A                  

TEST.BIN closed

```

mecparts, 2019-11-09:
I've finished the testing on the Ampro. Here's the results:

1. A call to BDOS 33 (read random) with a random record number past the eof but within the bounds of the last extent set CR, EX and S2 appropriately. A subsequent call to BDOS 21 (write seq) extended the file and wrote the record at the expected spot.

2. A call to BDOS 33 (read random) with a random record number past the eof and outside the bounds of the last extent set CR, EX and S2 appropriately. A subsequent call to BDOS 21 (write seq) did not produce an error but wrote nothing. I expect BDOS threw up its hands when it couldn't find the corresponding extent in the directory.

2. A call to BDOS 33 (read random) with a random record number of greater than 65535 did not set CR, EX or S2.

My feeling is that we update CR, EX and S2 when _sys_readrand returns 0 or 1. That would most closely mimic what CP/M does. It allows _sys_readrand to perform as a combination of seek() and an eof test for a later _sys_writeseq. At the moment I can't think how to duplicate what happens when a 4 is returned (i.e. update CR, EX and S2 in the read but don't actually write anything later on).

I've made the change locally to _ReadRand to update CR, EX and S2 if 0 or 1 is returned. i then ran the check program on both the Linux and Arduino sides and it produced the same result as on the real hardware.

MockbaTheBorg, 2019-11-09:
Cool.

I am at a science center with the kids. We are sleeping over (boyscouts
thing).
I will check this out tomorrow, then I have to figure out how to make the
changes in Windows.

Please make sure to pull v3.8 and modify from then on.

Thanks a lot!

mecparts, 2019-11-10:
I figured out yesterday how to set your repository as upstream, fetch it, and the merge it into my branch so that it's even. So we're good there.

Having slept on it, I think CR, EX and S2 should be updated when _sys_readrand returns 0, 1 or 4; that's what CP/M does. The fact that a subsequent _sys_writesync() call would create new extents (from the view of the CP/M program).... well, it *would* be doing what was asked, even if CP/M itself wouldn't do that. Being so far already into undocumented territory already, I think as long as the fix results in the programs we're using do what we think they should, we call it a win. And if something else turns up down the road that needs a different behaviour in order for it to work, we deal with that then.

On a completely unrelated note, I made a build for my ancient eee netbook, took the uSD card out of the Teensy and plugged it into the eee. I immediately discovered that the uSD card is *much* faster than it appears on the Teensy and that I definitely have to get SDIO hooked up and working there. I knew SPI was going to be slower, but the difference in speed was astounding.

MockbaTheBorg, 2019-11-10:
Yes ... the Teensy 3.6 is by far the fastest SD card speed I have on my collection. The ESP32's are ok, the Due and other boards are just painful.

And I agree with you... I think we brought RunCPM as close to the original CP/M 2.2 behavior as documentation allows. At this point if we get the correct behavior out of WS4 (I have never "printed" from it), I would say we nail this coffin.

mecparts, 2019-11-10:
I've put the changes in for Arduino, Linux and Windows. The changes for Windows looked to be exactly the same as those for Linux so it *ought* to work. Are you ready for another pull request?

WS4 was happy as soon as I removed the _sys_extendfile calls in the read routines. Updating the CR, EX and S2 fields on #1 and #4 error returns is intended (fingers crossed) to not re-break the program that caused you to put those calls in originally. :)

I sat down and soldered an FFC connector onto the Teensy 4's SDIO pins. Hand soldering at 1mm spacing is pretty much at the limit of my soldering ability but it's on there, no pins are shorted and the Teensy still works. And that was the easy part; now to try my hand at photo etching traces only 1mm apart for the other end of the cable and the uSD card. I just keep telling myself that if it speeds up the SD I/O as much as I hope, it'll all be worth it.

MockbaTheBorg, 2019-11-11:
Sure ... send over the pull request.

To be honest, the only reason I didn't buy a Teensy 4 yet is the lack of a Teensy 3.5/3.6 form factor version. That format is a winner.

closed 2019-11-11 by MockbaTheBorg

mecparts, 2019-11-11:
The pull request is in.

A Teensy 4.1 with the 3.5/3.6 form factor and an onboard uSD card holder is on my Christmas wish list too.

In some ways the 4.0 is definitely a "bleeding edge" experience at times. The USB serial code is, well, a bit problematic (it's called "a work in progress" on the PJRC forum), and there's a problem in the RTC code that meant that you couldn't set the time and have it stick through a power cycle. I've worked around/patched both things, but the USB serial problem caused me to spend many hours chasing non-existent bugs in CP/M programs  and/or RunCPM. But being able to run CP/M, at speed, in 1 square inch of PCB space, just keeps making me smile and shake my head in wonder.

MockbaTheBorg, 2019-11-11:
Yeah ... I am very proud of RunCPM.
I think it is the most interesting piece of code I ever wrote. And the story behind it is also meaningful to me as well.
Makes me happy every time I see someone having fun with it.

wrljet, 2019-11-11:
Thumbs up!

MockbaTheBorg, 2019-11-19:
@mecparts 

Just out of curiosity:
 
Why is STAT *.* showing the file sizes in multiples of 4K ... is this the minimum size of a 8MB hard disk sector?

mecparts, 2019-11-19:
Yes, the new allocation block size is 4K. As to why... it ended up being that either because that was what Johnson-Laird's  "The Programmer's CP/M Handbook" recommended for hard drives and I was referencing that when I was increasing the maximum disk size, or because that's what I had on the Ampro's hard drive and I might have copied the DPB from there - and I probably chose that value all those years ago for the exact same reason. But I'm pretty sure it's the former.

Thinking about it for a moment, there's really no reason in RunCPM that it couldn't be 2K. We're not worried about chewing up space in the directory. We couldn't go to 1K because that limits you to 255 allocation blocks total. But 2K would work. Oddly, the actual allocation block size on the uSD card's FAT file system that I'm using is even larger; I think it's 8K.

On another note completely; I thought I had Public file support working. Set the Public attributes on Turbo Pascal overlays and it happily finds them from whatever user area I'm in.

But then today I tried DazzleStar. Now, I really, really like DazzleStar. It's helped me figure out many things over the years. But at times it is just plain weird. First it was that business with setting file attributes in an FCB and trying to open a file. That was a bit odd. But today's oddity is even stranger. It doesn't appear to use BDOS' file open call to open its overlays. It uses find first - which I hadn't patched to deal with public files. Why? Because it never crossed my mind that I'd have to. And for the life of me I can't figure out why on earth it's doing that and not just doing a straight File Open call for each overlay. I'd say I need to think about this one more, but I'm not sure that it's going to help me figure out why. What I really need is a time machine set for November 1986 so I can ask the author.

MockbaTheBorg, 2019-11-19:
OMG... LOL
Having that time machine would be cool.

I think we can leave the block at 4k, no problem.
I will have to correct the RSTAT I wrote at some point just to make it
match STAT.

Regarding Dazzlestar, it might be using multiple undocumented behaviors,
for some time I have suspected a call to open is not really necessary if
you use find first. This might be the case.
Would need to disassemble CP/M in detail to find out. (Or use the time
machine)

mecparts, 2019-11-19:
What you said about not necessarily needing a call to file open echoes a thought I've had rolling around in my head for a bit. The BDOS doesn't keep track of open files; it doesn't know how many files have been opened, what position to read or write to next, none of that. Everything about a particular file's state is in the FCB. It isn't until a program calls the BDOS with a pointer to an FCB that BDOS knows anything about that file. So in theory, one could (on real hardware), go to the BIOS to read a directory sector, copy out a particular entry, set the 'not modified' bit in S2, zero out the current record and start reading. If the FCB was a duplicate of what the BDOS left it as after an open call, BDOS would have no way of knowing that it wasn't what had done it.

As someone I used to work with always said when presented with an off the wall idea to solve an otherwise intractable problem,  "but that'd be sick and wrong!" But I bet it'd work.

Oh, the pull you merged yesterday has also fixed the issue with XDIR reporting file sizes that were 32K too large. It must have been looking at the extent value, and when an inflated value leaked thru from the previous 'find next', that's what caused it. It may have also contributed to the block count issue you were seeing in STAT for the same reason. The number of directory entries was correct, but the last extent number would be too high.

MockbaTheBorg, 2020-01-04:
@mecparts Hey man ... any chance you can take a look at issue #101 just reported?
I am not sure if I did Boo Boo when importing your fixes.
I am travelling for work, so I won't be able to do original CP/M match for a couple weeks.

Thanks in advance.

mecparts, 2020-01-04:
I didn't remember any recent changes in that part of _ReadSeq or _WriteSeq, and looking through the history for disk.h confirmed that, so that's good. I will sit down, take a longer look and do some reading to triple check that the logic is sound. Another set of eyes, etc.

MockbaTheBorg, 2020-01-04:
Awesome ... thank you so much.
I should be at a hotel room by Monday night, I will be able to check it out
too.