MockbaTheBorg/RunCPM #95: 2 issues while compiling main.c on older linux-arm device (and solved)

guidol70, 2019-11-13:
Today I did try to compile RunCPM 3.8 on a older ARM-device with a MX51 (ARM7) CPU
a Efika MX Smarttop, which only got a Kernel 2.6.31.14.27 while using debian squeeze:
Linux smarttop 2.6.31.14.27-efikamx #1 Sun Apr 29 21:42:25 EDT 2012 armv7l

Another drawback was that only liblua5.1-dev was available :(

The first compiling stages did work fine, but with the main.c there where
2 problems:

1.)
======================================================================
gcc -Wall -O3 -fPIC -std=gnu99 -Wno-unused-variable -c main.c
disk.h:190: error: ‘for’ loop initial declarations are only allowed in C99 mode
disk.h:190: note: use option -std=c99 or -std=gnu99 to compile your code 
======================================================================
So I changed the compiler-flags in RunCPM-master/RunCPM/Makefile.posix to :
# Flags to pass to the compiler - add "-g" to include debug information
CFLAGS = -Wall -O3 -fPIC -std=gnu99 -Wno-unused-variable -g

2.)
======================================================================
In file included from main.c:33:
abstraction_posix.h: In function ‘_kbhit’:
abstraction_posix.h:146: error: ‘POLLRDBAND’ undeclared (first use in this function)
abstraction_posix.h:146: error: (Each undeclared identifier is reported only once
abstraction_posix.h:146: error: for each function it appears in.)
abstraction_posix.h:146: error: ‘POLLRDNORM’ undeclared (first use in this function)
======================================================================
for this I had to edit the part where POLLRDBAND and POLLRDNORM appears to:
int _kbhit(void) {
        struct pollfd pfds[1];

        pfds[0].fd = STDIN_FILENO;
        pfds[0].events = POLLIN | POLLPRI;

        return (poll(pfds, 1, 0) == 1) && (pfds[0].revents & (POLLIN | POLLPRI));
}

After these 2 changes main.c did compile on this old system.
Do you know / can think of which disadvantages the 2nd change will do to the app?

MockbaTheBorg, 2019-11-13:
Hi Guido,

I have added:
```
#ifndef POLLRDBAND
#define POLLRDBAND 0
#endif
#ifndef POLLRDNORM
#define POLLRDNORM 0
#endif
```
to the very end of posix.h. Right before the last #endif
This should make it build fine for this old platform as well as for any that do have these defined.

Thanks a lot,
Mockba the Borg

closed 2019-11-13 by MockbaTheBorg

guidol70, 2019-11-13:
this did work for the 2nd issue. but for the 1st issue I do need to add 
-stf=gnu99
to the compiler options

MockbaTheBorg, 2019-11-13:
Yes.

As that one seems to be compiler specific I will leave as is.

What is this old platform you tried?
Getting things off the drawers huh?
:)

guidol70, 2019-11-13:
Yes ist from the drawer, because here in Turkey I dont get not much new hardware :(
Its a Genesi Efika MX Smarttop: https://genesi.company/products/efika

MockbaTheBorg, 2019-11-13:
Yeah, I know how it is. When I was in Brasil, back in the days, you had to either build your own computer or get some junk out from universities and such.
Cool ... you just made me want to go through the drawers (again) and see where else I can load RunCPM on.
:)