MockbaTheBorg/RunCPM #96: Zmodem transfers over Telnet using ZMP on RunCPM

mecparts, 2019-11-15:
Totally NOT an issue, but I just had to tell someone about this.

[image]

I hooked up an ESP8266 board (with some software on it that let it act as an AT command capable modem) to one of the serial ports on the Teensy4. With some machine specific additions to RunCPM's BIOS jump tables to add TTY serial in status, TTY serial out status and TTY setup calls, and after writing an overlay file for ZMP15... I've got Zmodem transfers working on RunCPM. ZMP's not quite sure what to make of baud rates above 32K but it manages just fine in the end.

MockbaTheBorg, 2019-11-15:
LOL ... nice ... coincidently I am working on RunCPM right now, trying to figure out the Windows port of the function 17/18.

This is really cool ... I always wanted to get serial transfer working over the regular console. Even if XMODEM only.
I did a few tests, but nothing reliably successful so far.

mecparts, 2019-11-15:
Early on I took a very brief look at what it would take to get rz & sz working on RunCPM. It quickly became apparent that it wouldn't be straightforward. The gulf between what BDS/C could handle and what rz & sz needed was quite wide. I may return to it at some point but it's definitely a long term project. This on the other hand has only taken a couple of days to get to the "hey, it actually worked!" stage. I wanted something to make it easier to transfer files to the Teensy, and this should do the job.

closed 2019-11-15 by mecparts