MockbaTheBorg/RunCPM #98: Internal CCP not optimal

MockbaTheBorg, 2019-11-24:
Hey everyone,

Maybe you want to give the internal CCP a try? Let me know what works as intended and what fails?
I am willing to make it work better. I is a good alternative for microprocessors as its code will sit in flash and not RAM.
Suggestions to code improvements are also welcome.

Thanks a lot,
Marcelo.

drawkula, 2019-11-24:
I'm currently not CP/M-ing on MCUs. But let's torture `BorgCCP` a bit on Debian10/AMD64. It is https://github.com/MockbaTheBorg/RunCPM/commit/e6a0850bad16ae9f5d80902965cb84226eea4de4 with 64k TPA and empty CPPHEAD.
```
CP/M 2.2 Emulator v3.9 by Marcelo Dantas
      Build Nov 24 2019 - 19:31:19
-----------------------------------------
CCP : INTERNAL v1.5   CCP Address: 0xf400

```
Will `submit` work as expected?
```
A0>c:
C0>user 1
C1>type myhi.c
#include "stdio.h"
main()
{
        puts("No meltdown, no specte.");
        puts("No rowhammer, no brainslammer!");
        puts("\nCP/M!");
        puts(";-)");
}

C1>type myhi.sub
; create myhi.com
c myhi.c
myhi
;

C1>do myhi
SuperSUB V1.1

C1>; CREATE MYHI.COM
C1>C MYHI.C
HI-TECH C COMPILER (CP/M-80) V3.09
Copyright (C) 1984-87 HI-TECH SOFTWARE

C1>MYHI
No meltdown, no specte.
No rowhammer, no brainslammer!

CP/M!
;-)

C1>;
```
Ok... using SuperSUB named `do` here it looks ok.
```
C1>submit myhi 

C1>; CREATE MYHI.COM
C1>C MYHI.C
HI-TECH C COMPILER (CP/M-80) V3.09
Copyright (C) 1984-87 HI-TECH SOFTWARE

C1>MYHI
No meltdown, no specte.
No rowhammer, no brainslammer!

CP/M!
;-)

C1>;
```
Good ol' `submit` is happy too.
Now the big stuff: Rebuild [Peter Schorn]'s SPL:
```
C1>h:
H1>user 2
H2>submit makespl
```
(((Lots of lines not pasted in here.)))
Ended without error.
So a new SPL compiler and all the libraries should be rebuilt now.
Do it again with the rebuilt toolchain, this time using SuperSUB:
```
H2>do makespl
```
(((Lots of lines not pasted in here.)))
Ended without error.
```
H2>█
```
So submit files are working from other user areas and other drives than A.
Let's call it "Happy End!" for this test.
\o/

MockbaTheBorg, 2019-11-24:
Awesome!!! So looks like my guava didn't fall too far from the tree.

MockbaTheBorg, 2019-12-04:
Anyone else wants to give it a try?

closed 2020-01-23 by MockbaTheBorg