MockbaTheBorg/RunCPM #101: Suspicious incrementing of record number

lkesteloot, 2020-01-04:
The `_ReadSeq()` and `_WriteSeq()` functions both have this code to increment `cr`:

```
++F->cr;
if (F->cr > MaxCR) {
        F->cr = 1;
        ++F->ex;
}
if (F->ex > MaxEX) {
        F->ex = 0;
        ++F->s2;
}
if (F->s2 > MaxS2)
        result = 0xfe;
```

The `MaxCR` constant has the value 128, which means that `cr` will go from 0 to 128 (inclusive), then wrap to 1. Is that right? I've looked at two other CP/M emulators and they both do the more obvious thing, which is to go from 0 to 127 then wrap to 0 (as you do for `ex` and `s2`). All three have the same sequential behavior, but yours will stay one record longer on the previous extent, which may matter if the user code checks the extent number, or changes the `cr` value at that moment. I've not checked the code of BDOS, so I don't know which matches the original.

MockbaTheBorg, 2020-01-04:
@lkesteloot I won't have a chance to check this for some time as I am travelling for work, but I have asked for help through another related issue we had.
I might have messed this part of the code fixing something else, I will find out.
In the meantime, the original BDOS code is included on disk A, maybe you want to check it out?
seqdiskread is at line 2118 and there you might see the related code. A quick look there made me think there's indeed something wrong.

Thanks!!!

lkesteloot, 2020-01-04:
Okay, I think the original logic is as follows:

```
1: sequential read:
2:     if record number to read > record count in extent:
3:         get next extent
4:         reset record number to read to zero
5:     seek and read record
6:     increment record number to read
```

So the increment that happens _after_ the read (on line 6) does not check for overflow. It lets the record number go to 128, as you do. But the check for overflow happens _before_ the read, and not with 128 explicitly but with the number of records in the extent.

It looks like your logic is closer to the original, since it allows the record number to actually be 128 between reads. It would be nicer if you never actually used that value for a seek, though, but that's more of an internal detail probably.

Up to you whether to fix this, or even consider it a bug, but perhaps document it since it _looks_ like a bug!

mecparts, 2020-01-04:
Here's my first thoughts on this:

Here's what Cortesi had to say in his section on the FCB and what the CR byte contains:

> 20h: The Current Record: The byte 20h into the FCB is used by the BDOS as a count of the number of standard records that have been read or written in this extent of the file.
> 
> You are expected to set the current record number to 00h when opening a file for sequential access. Then the first access will produce the first record of the extent. The current record byte will by incremented to 01h, which will cause the next service request to access the second record, and so on. When the current record equals the record count (on input) or the capacity of the extent (on output), the BDOS knows that the extent is finished.
> 

So the way I read that, after each sequential write/read, RC is the number of 128 byte records that have been written/read from the current extent. Based on that, I would expect the value of CR to increment from 1 up to 128, then recycle to 1 when the first record of the next extent has been accessed.

I wrote a couple of short little programs to test out what sequential write & read left in the RC and EX fields of an FCB and ran them on my Ampro LB. After creating/opening a test file, I zeroed out RC and did a sequential write/read. After the first write/read, RC was 1 and EX was 0, as you'd expect. After accessing 128 records, RC was 128 and EX was 0. After then  the129th access, RC was 1 and EX was 1.

I then repeated the same exercise on my RunCPM system and got identical results. RC was 1 after the first access, 128 after a full extent of records had been accessed, and then recycled to 1 after the first access in the next extent.

At first blush, it appears to me that RunCPM is doing exactly what CP/M does.

On first reading, Cortesi does seem to contradict himself on what CR should contain when he describes what the Sequential Read & Write calls do. But I've long since learned to think very carefully about what he's saying, because upon further reflection I've always found him to be correct; I just need to understand exactly what it is he's saying. Here's part of what he says about Sequential Read:

> The 128-byte record at position _cr_ of the extent described by the FCB is read and placed in the current file buffer. The _cr_ field is incremented. If it then equals _rc_, the entire extent has been read; the directory entry describing the next extent of the file is copied into the FCB and _cr_ is set to 00h.

Again at first blush, that would imply to me that after reading the 128th and final record in an extent that the CR field would contain 00h, which doesn't match what I saw. However, if the final phrase "the directory entry describing the next extent of the file is copied into the FCB and _cr_ is set to 00h" is supposed to happen not at the _end_ of the 128th access, but at the _beginning_ of the 129th access, then it makes sense and does match what I saw.

The description of Sequential Write makes this interpretation even more likely. Writing the 128th record of the first extent is not going to create a new, second, empty directory entry. That's going to wait until the 129th record is actually written.

For myself, I'm not really troubled by _ReadSeq & _WriteSeq comparing CR to 128 rather than RC.  By the time RunCPM's code does that, it already knows that the read or write has succeeded and that there was data read from or written to the file (and therefore that CR and EX were within range). If a read was attempted beyond the EOF, _sys_readseq will have indicated an error the current record count will remain unchanged.

I'll keep mulling this over for a bit but my initial reaction is that it's working as it should.

MockbaTheBorg, 2020-01-04:
Awesome.
Yes this is my suspicion too.
Thanks for double checking.

lkesteloot, 2020-01-05:
Right, so we're all in agreement that RunCPM's behavior externally matches real CPM's. It might still be even nicer to put the wrapping code before the read (as in the read CPM) rather than after, since the current code actually tries to read "extent 0 record 128", which is numerically correct but not logically correct. It should first wrap to "extent 1 record 0", then do the seek/read. (I realize the same record will be read either way.)

closed 2020-01-23 by MockbaTheBorg