MockbaTheBorg/RunCPM #102: Anyone playing with the Teensy 4.0?

MockbaTheBorg, 2020-02-12:
Hi all,

I got a couple Teensy4 and the Audio breakout for a synth project, but this breakout also has a uSD card slot on it, so I thought of trying RunCPM on it as well.
I am wondering if anyone has had the chance to use it like that?
The uSD might be slow, I know, but worth a try.

Please let me know!

Thanks!!!

mecparts, 2020-02-17:
Back on the air after a prolonged bout of hardware hacking. Yes, I'm still using my Teensy 4.0. Last week I added a proper RS232 serial port to the Teensy and hooked up a homebrewed Wemos D1 mini based WiFi modem to it. Sure, I could have skipped the whole RS232 bit and just hooked the mini right up to a serial port on the Teensy, but I'm also building another modem for the Ampro and figured I might as well go full retro on both of them. And there's just something about seeing 8 LEDs blinking behind red acrylic that's very nostalgic. I'm trying to resist adding a little MP3 board in there for faux dialling and connection sounds.

And just a few minutes ago I completed the first test of a parallel printer port. A 16 bit I2C I/O expander to some good old 74LS buffers, a DB25 to Centronics cable, a venerable Juki daisy wheel, and now the Teensy can print like it's 1988.

Throw in a first generation Raspberry Pi running a customised version of PiGFX and you've got a very, very, _very_ fast CP/M system with a colour ANSI terminal (with GSX graphics extensions), a daisy wheel printer and a modem faster than anything I could have dreamed of back in the day. I think it's time to lay out a PCB so I can get SDIO to the uSD card working and put it in a case. Mind you, if it works out like I hope, that case will be small enough to bolt to the VESA mount on the back of the monitor. The Little Board could have a Teensy Board little sibling.

MockbaTheBorg, 2020-02-17:
Daaaamn, this is awesome.
This project deserves a website.
Are you making one?

I just got the teensy and audio boards two days ago, but I have been so
sick that I couldn't write a single line of code.

Maybe later today I will try loading it there and making the teensy+audio
board a self contained solution.

I wonder how much slower the audio board sd is, when compared to the native
4-bit sd.

Cheers,
Marcelo

mecparts, 2020-02-17:
I hadn't thought of doing a website, to be honest. But that doesn't preclude sharing some pictures of the prototype right now.

Here's the main board. The DB25F and DE9M on the left are mounted in a quickie 3D printed holder. A MAX3232 chip provides the RS232 level conversion. An MCP23017 I2C I/O expander handles the I2C <-> Centronics conversion and a 74LS245 and 74LS05 convert to/from 3.3V to TTL levels. I went with LS rather than HCT or ACT because they're feeding signals off board and TTL doesn't require the protection that CMOS would.
[image]

And here's looking across the main board at the Pi that acts as the terminal. The Pi is running a bare metal (no O/S) ANSI terminal emulator derived from PiGFX. The Pi is powered from the same USB supply that powers the Teensy board (the old Pi doesn't need many mA and the adapter has lots of capacity). A standard LCD monitor and USB keyboard are plugged into the Pi. The idea is to use the upper half of the VESA mount to hold the eventual Teensy PCB. You could use a Pi Zero (but not, from what I've read, a Pi Zero W) in place of the old Pi that I did.
[image]

Here's the WiFi modem prototype. I've brought out RTS and CTS for flow control, but didn't bother hooking DSR, DCD, et al up to anything other than the LEDs. Again, a MAX3232 provides the RS232 level conversion. The status LEDs are driven by a 74ACT245. The final version will use a Wemos D1 mini in place of the D1. More quickie 3D printed mounts hold the DE9F and the red acrylic faceplate. For looks I'd love to get a hold of an old Hayes Smartmodem case to mount the final board in, but they're metal except for the end pieces and that wouldn't do good things to the WiFi.
[image]

And finally, the nostalgia shot of the modem's status LEDs. 
[image]
 
My gut feeling on the speed of the uSD with SPI is that if we sent this setup back in time to the late 80s, we'd be more than happy with how fast the "disk" I/O is. Fast forward to today and it feels a bit sluggish. Usable, but sluggish. I'm looking forward to trying out SDIO. Will I get the board laid out and made before Paul brings out a Teensy 4.1 with onboard uSD hooked up to SDIO?

Wayne

MockbaTheBorg, 2020-02-17:
This is REALLY COOL.

I am having a hard time here initializing the SD card.
How did you initialize yours?

Thanks!

mecparts, 2020-02-17:
The only change I've made since I [originally patched things to get it working](https://github.com/MockbaTheBorg/RunCPM/issues/87#issuecomment-538615794) back in October was to remove the SD_SCK_MHZ(25) from the SD.begin(SDINIT) call. The el cheapo SD adapter I had started with was the bottleneck, and once I changed over to something better I was able to let the library figure out the speed it liked. You could try adding the speed parameter to your SD.begin call and seeing if the uSD on the audio board is likewise not quite up to snuff.

MockbaTheBorg, 2020-02-18:
Looks like the latest (1.1.1) Greinman SdFat now works with the Teensy 4.0.
After I loaded it all seems to work.
And damn this thing is fast. Like REALLY fast.

mecparts, 2020-02-18:
There's a new SdFat? Excellent. I can retire my patched version.

Yeah, the speed is really something, isn't it? \<insert evil grin\>

MockbaTheBorg, 2020-02-18:
Yup ... I am amazed that it runs on the Teensy 4.0 as fast (if not slightly faster) as on my i7 2.6G laptop.
It is really awesome.
I will close the issue as Teensy 4.0 (with the audio board) is officially added now.
Thanks for the help man! And please, get the details of your project into hackaday or something.
It is coming up amazing. 👍

closed 2020-02-18 by MockbaTheBorg

mecparts, 2020-06-24:
[A new/old computer is born.](https://github.com/mecparts/TeensyBoardZ80)

Finally, this
[image]
has become this
[image]
and has been put into this
[image]
to become this
[image]
which is capable of doing this.
[image]

You may not have intended RunCPM to emulate a whole CP/M computer, but that doesn't mean that it can't be transformed into the beating heart of one!