MockbaTheBorg/RunCPM #105: Discord Invite is no longer working on the Github page.

scott8, 2020-03-16:
Hey!
I wanted to chat and the Discord invite is expired apparently. 

Scott

MockbaTheBorg, 2020-03-16:
I will create another one.

MockbaTheBorg, 2020-03-16:
Pls see if this works:
https://discord.gg/eHPkgj

scott8, 2020-03-16:
Thanks!
See you on Discord.

Scott

MockbaTheBorg, 2020-03-16:
Cool, see you there.
I need to figure out a way to make a discord invite that won't expire.

Welcome to the group.

closed 2020-03-16 by MockbaTheBorg

xpunkt, 2020-03-16:
scott8 skrev den 2020-03-16 03:44:
> Thanks!
> See you on Discord.

metoo :-)

scott8, 2020-03-16:
Thanks!
I found this:
https://support.discordapp.com/hc/en-us/articles/208866998-Invites-101
"To create a *permanent invite, *select *"never"* under *"expire after" *
and *"no limit"* under *"max uses."* By doing so, you'll generate a link
that has *infinite uses* and won't become obsolete over any period of time.
However, it's important to remember that *the only way anyone can access
your server* is through an *invite link* so be careful with who you give it
to!"