MockbaTheBorg/RunCPM #106: sd not defined in this scope

huskeyw, 2020-03-24:
compiling for DUO and get SD not defined in this scope

huskeyw, 2020-03-24:
fixed, switched to the adafruit lib version of sdfat..

MockbaTheBorg, 2020-03-24:
This problem had been seen before, you need to make a small modification to greiman's sdfat config.

huskeyw, 2020-03-24:
no it was the wrong sdfat from library manager.. needed the adafruit one..
then everything worked. the changes were already there when installed.
thanks for looking into it thought..

to be clear there are two sdfat libraries and had to install the adafruit
one instead of the standard sdfat
William "Steve" Huskey
Managing Member Dreadnought Brewing LLC

MockbaTheBorg, 2020-03-24:
Awesome. 
I originally used Bill Greiman's library because of its higher speed and compatibility with all systems, and I see now there's an Adafruit fork of it.
Glad to know it is working for you. Have fun, and feel free to join us on the Discord channel if you want.

Cheers!!!
Marcelo.

closed 2020-03-24 by MockbaTheBorg