MockbaTheBorg/RunCPM #107: Arduino Due + SD Adapter not working

tomxp411, 2020-04-23:
It appears the SDFat support has changed for the Arduino Due, and RunCPM needs to initialize the card differently than previously.

The two changes that need to be made are:

1. DO NOT use SdFatEX. Instead, line 4 of due.h reads: 
`SdFat SD;`
2.  Run the SD card at 25MHz. At line 86 of RunCPM.ino, I inserted: 

```
#elif defined board_due
  _puts("Initializing Arduino Due SD card.\r\n");
    if (SD.begin(22, SD_SCK_MHZ(25))) {

```
With those two changes, everything started working. 

This was tested with this Arduino SD Card Adapter: 
https://www.amazon.com/gp/product/B07QQBBL6S
(Note: this interface seems to require 5V power, even though the card uses 3.3v. There is an on-board regulator that I'm guessing is stepping down the voltage.)

The changes are in my fork at https://github.com/MockbaTheBorg/RunCPM

mecparts, 2020-04-23:
When I started working with RunCPM on a Teensy 4.0, I used that same type of adapter. I found that it too wouldn't work until I changed the SD.begin call to limit the speed to 25Mhz. 

Switching to using a bare bones breakout board - [DigiKey part #CN0023-ND](https://www.digikey.com/product-detail/en/chip-quik-inc/CN0023/CN0023-ND/5978271) (no level shifters, no regulator, just the socket on a little board with all the lines brought out to a header) - solved the speed limitation. The same processor, with the same uSD card, has been working at full speed for several months now.

So I suspect that the speed limitation is down to that particular type of adapter, rather than anything to do with the type of processor.

tomxp411, 2020-04-23:
Thanks. I’m actually switching to a Teensy and Grand Central as soon as they arrive. 

In the meantime, I’m thinking it would still be a good idea to switch to using the SdFat class (not the Ex version) since softSIO support is not needed on the Due, as well as document the setup a little better, with options for SD initialization.  Ispent hours trying to track down this issue, and someone who’s not that familiar with Arduino would likely have given up.

mecparts, 2020-04-23:
I've actually only ever used SdFat (it's a case of "let's try this, okay this works, don't mess with it").

I'm looking forward to the release (next month, with any luck) of the Teensy 4.1. It has the Teensy 3.6 form factor with an onboard uSD holder hooked up to the SDIO lines.

huskeyw, 2020-04-25:
> It appears the SDFat support has changed for the Arduino Due, and RunCPM needs to initialize the card differently than previously.
> 
> The two changes that need to be made are:
> 
> 1. DO NOT use SdFatEX. Instead, line 4 of due.h reads:
>    `SdFat SD;`
> 2. Run the SD card at 25MHz. At line 86 of RunCPM.ino, I inserted:
> 
> ```
> #elif defined board_due
>   _puts("Initializing Arduino Due SD card.\r\n");
>     if (SD.begin(22, SD_SCK_MHZ(25))) {
> ```
> 
> With those two changes, everything started working.
> 
> This was tested with this Arduino SD Card Adapter:
> https://www.amazon.com/gp/product/B07QQBBL6S
> 
> And connected according to this diagram:
> https://forum.arduino.cc/index.php?topic=132130.0
> (VCC on the adapter is connected to 5V on the SPI header. CS is tied to ground.)
> 
> The changes are in my fork at https://github.com/tomxp411/RunCPM

thank you that fixed seeing the drive now to understand how to get it to see the files on A drive.. it shows blank after unzipping the adrive to a folder called A

MockbaTheBorg, 2020-04-25:
You need to put the files on a sub folder called "0" under the folder
called "A", which is the CP/M user 0.
CP/M 2.2 would allow 16 user "areas" inside the disk.

Cheers,
Marcelo

huskeyw, 2020-04-25:
> You need to put the files on a sub folder called "0" under the folder called "A", which is the CP/M user 0. CP/M 2.2 would allow 16 user "areas" inside the disk. Cheers, Marcelo
> […](#)

thanks, I reread the docs.. assumed it was zipped that way and did not look at it.. fixed now

MockbaTheBorg, 2020-12-21:
@tomxp411 
Any chance you can give your GC4 a try with version 4.5?

Thanks!!!

tomxp411, 2020-12-29:
Sure. I'll load it up today. (Sorry I didn't respond sooner; I missed the email and just noticed it as I was cleaning my inbox.)

MockbaTheBorg, 2020-12-29:
Awesome! Thank you.
Please check if you can get it to work using either the esp32 method (init SPI first and then SD) or the native SD method.
I am trying to simplify as much as possible the code to reduce the amount of #ifdefs.
The time is coming when I will consider RunCPM a "finished project". :)

tomxp411, 2020-12-29:
The SDFAT error caught me;, but I found your fix for it in the Readme. 

The other error was BufferedPrint.h - it's missing. I had to add the StreamLib library to get that. 
I don't see mention of that in the documentation, so that should probably be added to the build instructions.

tomxp411, 2020-12-29:
[image]

I will investigate. It may be a matter of setting the correct pins for the GC's SD Card.

MockbaTheBorg, 2020-12-29:
Cool ... thanks a lot.

tomxp411, 2020-12-29:
Found the problem. It was the pin assignment. 
I added GC.h, which uses the correct pins:

```
// ADAFruit Grand Central
#ifndef DUE_H
#define DUE_H

SdFat SD;
// The GC uses SPI1. CS pin is SDCARD_SS_PIN
#define SDINIT SDCARD_SS_PIN, SD_SCK_MHZ(SDMHZ)
#define SDMHZ 50
#define LED 13
#define LEDinv 0
#define BOARD "ADAFRUIT GRAND CENTRAL"
#define board_gc
#define board_analog_io
#define board_digital_io

#endif

```

So, to sum up:

- add StreamLib to the dependency list in the install directions. 
- add this new include file. 
- add directions that line 9 #include "hardware/xxx.h" needs to be modified to reference the board in use. 

I'm working on a PR right now.

tomxp411, 2020-12-29:
PR sent.

MockbaTheBorg, 2020-12-29:
Cool ... I will put it all together and send up v4.7.
Thanks a lot.

closed 2020-12-29 by MockbaTheBorg