MockbaTheBorg/RunCPM #108: GSX support?

polluks, 2020-04-27:
Take a look at http://www.vcfed.org/forum/showthread.php?73271-GSX-for-CP-M-2-2

MockbaTheBorg, 2020-04-27:
Hi there polluks,

If you look at closed issue #58, there's some stuff done there by @drawkula using sixels.
GSX is also brought up there.
However, we need to keep in mind that RunCPM doesn't emulate a CP/M computer, but the CP/M environment itself, there is no hardware, no physical ports, no monitor, therefore there's no native terminal to run graphics.
This simplicity is actually what makes it portable, as this is exactly what allows it to run unmodified on so many different platforms. 
With this in mind though, GSX is supported on regular hardware by using a BDOS call (115), which could be implemented on a build of RunCPM for Windows for example, driving an emulated graphics "panel".
One would have to emulate a graphics terminal, one of those Tek or HP, and receive the RunCPM BDOS calls and treat them.
I have never used or had access to a graphics capable terminal on CP/M back in the days. To me CP/M was always a 80x25 green-screen type of thing. Therefore it never crossed my mind implementing it.
The closest I did to graphics on CP/M was on an Apple II, where I would load a line chart onto the 6502 memory area and make a hardware call to make the Z80 switch back to it momentarily until someone pressed the space bar. Wild times.
So, answering possible questions you might have:
1. Does it have support for GSX? Currently no.
2. Are you interested/going to implement? Currently, no.
3. Can it be done? Definitely, though not portable.
4. Would you help me do it? By providing as much information as I can regarding the insides of RunCPM, definitely!

Cheers,
Marcelo

mecparts, 2020-04-27:
Well, actually...

[image]

(image from my modified PiGFX ANSI terminal with GSX support)

I started off tinkering with GSX support several months ago. I just didn't think it'd be of much interest to anyone else.

My approach was to build the GSX BDOS call right into RunCPM. That way, you wouldn't need GENGRAF.COM to tack the BDOS 115 interceptor and GSX.SYS onto each and everyone of your GSX'ified programs.

With that done, I also put the GIOS driver code into RunCPM as well. I started off with supporting a couple of small TFT displays

[image]

(image of an early trial on a Teensy 4 running RunCPM with an auxiliary ILI9341 320x240 LCD display)

before moving on to adding graphics support to my PiGFX ANSI terminal fork and writing a driver for it.

This isn't straight GSX per se; there's no need of GENGRAF or GSX.SYS. I don't use loadable drivers in Z80 code; everything happens inside RunCPM. All the CP/M program has to do is set up the GSX data structures as it normally would, call BDOS 115, and RunCPM takes care of the rest.

I have skeletal GSK libraries for both Turbo Pascal and Turbo Modula2.

It's still all very much a work in progress. While I have lots of things working: lines, colours, polygons, text (rotatable), circles, bars, arc, all the fills, locator input... I still have lots of things to get to: line styles, pie slices, proper colour palette support, reading the text cursor, bitmaps (what GSX calls cell arrays, or more properly, sprites).

You can take a look at what I've done in my [ZCPR3.3 fork of RunCPM](https://github.com/mecparts/RunCPM/tree/zcpr33). Just keep in mind that it's nowhere near finished, or commented, or any of that good stuff.

closed 2020-05-03 by MockbaTheBorg