MockbaTheBorg/RunCPM #109: RunCPM software repository.

MockbaTheBorg, 2020-05-03:
Hi Folks,

I have proposed this idea on the Discord channel as well:
I am thinking of creating a repository for software that runs on RunCPM. We could put there things that we use and are proven to be working fine on it, like Hitech C, Wordstar, etc...
The idea is to make a repository that would contain the folders in zipped format, preferably untainted original distributions, so anyone willing to run RunCPM would have a central location to go look for software for it.
If we find original distribution disks of things it would be even better.
We could maybe create a repository here on GitHub itself, not sure if they would give me grief for that, but this would be like a "curated" collection.
However this only works if everyone using RunCPM would collaborate with the idea, sending software which I believe could be done by sending push requests to the collection, or sending attachments by open "issues" to the collection repository.
I do not want to go with a completely different hosting solution, so as close as GitHub as possible would probably be the best.

Please let me know what you guys think about it.

drawkula, 2020-05-12:
I don't "discord", so maybe my questions already have been answered.

Can we look from the other end at this problem?

------

How would pre packaged software or curated disk sets be managed on the running system?

The fixed drive directories are not well suited for having a big archive (virtual floppy disk box) and swapping them in/out into the 15 (or 16? Allow swapping A: too? Would be ok with reboot?) drive letters.

Looking at the drive letter (or even the drive/user?) directories as mount points for disks held in a different space on the SD card would make juggling disks possible.

Then this virtual disk archive would be one end of having pre packaged software. Mounting them to the drive letters, maybe even with a copy on write layer? Having a pristine (compressed?) "disk image" and un-/mounting a writeable (unpacked) working copy of it would automagically lead to a kind of package system. The read only "seeds" for the work copies could be uploaded to the curated archive of user supplied images.

------

[YAZE](http://www.mathematik.uni-ulm.de/users/ag/yaze-ag) somehow can mount/swap disk images (but I've not used it much):
```
A>mount

A: r/w  BOOT_UTILS.ydsk
B: r/w  CPM3_SYS.ydsk
C: r/o  ./disksort/
D: r/w  DISKSORT.ydsk
F: r/w  ZINC.ydsk
G: r/w  TEST-UTILS-1.10.ydsk
H: r/w  HI-C-Z280-Compiler.ydsk
I: r/w  UNIXLIKE.ydsk
J: r/w  MMU-Utils.ydsk
K: r/w  Kermit_SZRZ.ydsk
L: r/w  spl.ydsk
M: r/w  Turbo-Modula-2.ydsk
N: r/w  testdsk.ydsk
P: r/w  BIOS3.ydsk

 DRIVES:  A B C/ D . F G H I J K L M N . P

A>sys

$>?
help        Display this text or give help about a command
?           Synonym for `help'
attach      Attach CP/M device to a unix file
detach      Detach CP/M device from file
setaccess   Turns on/off access time stamps for mounted directories
mount       Mount a unix file or directory as a CP/M disk
remount     Remount a CP/M disk
umount      Unmount a CP/M disk
create      Create a new disk
128         Set sektor size to 128 for all disks (only CP/M 3.1)
interrupt   Set user interrupt key
go          Start/Continue CP/M execution
!           Execute a unix command
quit        Terminate yaze-ag
time        Display elapsed time since last `time' command
$>█
```

------

Aaaaand [this TRS-80 emulation](https://hackaday.io/project/9077-trs-80-model-1-on-a-pic32) does similar stuff with disk and tape images.

------

Happy end.  
Really.  
Really?  
;-)

MockbaTheBorg, 2020-05-12:
Hey there. Hopefully safe and sound.

I think you are overthinking the idea.
What I have in mind is just a GitHub repository where we would organize by sub-folders like "Databases", "Text Editors", "Compilers", etc.

Then inside those sub-folders we would put the software on .zip files, ready to be extracted onto a letter/number folder of your choice and use right away, without having to extract disk images or anything.

In there we would have the software as close to original (freshly installed from distribution disks) as possible, ready to start using.

And by "curated" I mean: we would keep the file names under a certain naming convention to avoid things like "WS.ZIP" which doesn't say anything about it.

And also "curated" in terms of content, so the Wordstar "disk", for example, won't be a mess of unrelated CP/M executables which also happens to have WS.COM in there.

If we happen to have original distribution disks of some software, we would put that there too, for the sake of completeness, so we would have, for example, WS_3.3.0.ZIP (ready to go) and WS_3.3.0_DIST_D1.ZIP, WS_3.3.0_DIST_D2.ZIP and so on. All ready to unzip onto some RunCPM number folder.

The collaboration would be done by pull requests, and the "curation" by folks added as collaborators to the repository, something like that. Any discussions regarding the distributions would happen as issues on the repository.

No disk images on the repository, only extracted disk images. The README on each folder would describe what each .ZIP available is.

In case we have real original disk images we could probably put them there inside a .ZIP file too, just for historical reasons, completely unrelated to RunCPM usage.
But this is only for known-to-be-pristine original disk images. Not disk-I-was-using-when-in-high-school disk image.

This is all I have in mind so far. This way someone willing to use whatever software with RunCPM would find it there, clean, ready to be extracted onto a drive letter/number and ready to use.

Cheers!

prlaba, 2020-05-12:
I think the issue isn't so much what to _name_ these zip files, but how to organize and name their _content_.

In the simplest case - a software package containing a single distribution disk - the zip file could be organized as a flat collection of files.  A user would "mount" the disk by simply unzipping the zip file into a named RunCPM drive folder/user area ("A\0", "B\0", etc).

But this scheme gets clumsy if the software package has multiple distribution disks.  In that case the zip file would contain multiple folders (one for each distribution disk).  But what to name these folders?  If they are named the same as the disks in the original distribution, the user will need to: 1. Extract the entire zip file into a temporary folder, and 2. Copy each folder in the temporary folder to a RunCPM drive/user area folder.

In a real CP/M system, a user would typically mount each distribution disk on an available drive, and either run the files on the disk directly (no install), or copy them (PIP) to the same or another drive.  That's not easily accomplished in the current design.

It may be your thinking that these zip files represent the state of software packages _after_ their installation.  That's ok, but I think it would so much more easier, and much more valuable, to maintain and use the original distribution disks _as is_.

There's a significant difference between a _drive_ and a _disk_.  In the current RunCPM design there are only drives (folders "A", "B", "C", etc.).  No Mount and Dismount commands are needed, because drive = disk.  That's fine if you have fewer than 16 disks.  But if you want to manage a larger collection of disks, you must maintain them outside of RunCPM, and then copy them to a RunCPM drive folder whenever you want to "mount" them (and make sure you don't overwrite an existing drive folder).

This is further complicated if a software package provides its own INSTALL program, prompting you to mount each distribution disk, sequentially, in the same drive.  That simply wouldn't work in RunCPM, as best I can tell.

I'm not sure how difficult it would be to provide Mount and Dismount commands in RunCPM, where Mount would "link" (whatever that means) a named disk folder to a CP/M drive letter, and Dismount would unlink the two.  The mounts would obviously need to persist across reboots.  It sounds non-trivial, especially if you want to maintain backward compatibility.  The advantage of such a scheme is that I would now be able to manage my disk folders with more meaningful names (e.g., "WORDSTAR v1.7"), and be able to mount those folders on any CP/M drive I want (assuming no other disk is currently mounted in that drive).  Otherwise I'm forced to continually rename/copy my disk folders, which will only get worse once this repository idea takes off.

Without that capability, at a minimum the new repository should have clearly stated guidelines on how to name and organize the files and folders within a distribution zip file, and instructions on how to "mount" a distribution disk under RunCPM.

One last comment: Are we sure there aren't already CP/M repositories out there somewhere?  (I've never looked for any.)

MockbaTheBorg, 2020-05-12:
@prlaba Yes, there are still a few CP/M repositories around, but it is hard to find disk images containing a single software, and even harder to find original distributions of software. Not to mention .ZIP (or other format) files containing clean files from original distributions.

Regarding the "mounting" I do something like that on my Windows machine. I have some things on regularly names folders and I just symlink these folders onto B/0 or B/1 for example.
I do it manually, but it could become a "mount.bat" (which will survive reboots) easily.
But this is only for Windows. Linux would have to do somewhat differently and other platforms may even lack the symlink capability.

My thought regarding the repository is much more in order to provide RunCPM users with software they can readily unzip and use, than with a disk image in .DSK or .IMG that they will now have to figure out how to extract the files from.
And then when they do extract, they discover that the DbaseII they wanted has pieces missing and is also plastered with other CP/M software that is not part of the DbaseII distribution.

So my idea is much more targeted towards making software available to people in a more centralized, convenient (from RunCPM's perspective) way, than having these distributions integrated onto RunCPM in a form of "simulated disk ejecting/mounting".

I am starting to think like the "build it and they will come" approach. Maybe I will start the repository with what I have and then people will see it and start contributing.

guidol70, 2020-06-12:
Did you check the .dsk-images from https://schorn.ch/altair.html ?
There are
- Programming Languges
- Office Applications
- Games and Tools

PS: Now downloading RunCPM 4.3 ;)

MockbaTheBorg, 2020-06-15:
I will put this on hold for now. Suddenly things changed and I won't have time to prepare/curate the repository either. 
So it goes into the backburner for now, but will happen.

closed 2020-06-15 by MockbaTheBorg