MockbaTheBorg/RunCPM #112: Unimplemented BDOS call

riwide, 2020-06-11:
When running LU.COM v.1.1 (Jon Saxton's version, in C compiled with his version of hitech c v3.09)
the following error is given
Unimplemented BDOS call.
C = 0x3c

lu11 .zip contains lu11.arc.

[lu11.zip](https://github.com/MockbaTheBorg/RunCPM/files/4767351/lu11.zip)

Note:  If you need to recompile LU.COM, it requires Jon Saxton's library tweaks to build. - Tony Nicholson has released some new HiTech 3.09 versions incorporating known patches and fixes. (https://github.com/agn453/HI-TECH-Z80-C
This program needs 3.09-5  (I think, I tested with an interim modified 3.09-4).

agn453, 2020-06-11:
BDOS function 0x3c (60 decimal) is a CP/M 3 function to call an resident system extension (RSX) module.  Jon Saxton may not have included a test for the CP/M version to avoid this happening under CP/M 2.2.  I'll check the HI-TECH C libraries and build lu11 later today to determine if any changes are required.

agn453, 2020-06-11:
I've found the issue in CRTCPM.OBJ and added a check for the CP/M version.  Fetch and replace your CRTCPM.OBJ for HI-TECH Z80 C from https://github.com/agn453/HI-TECH-Z80-C/blob/master/dist/CRTCPM.OBJ to remedy this.

riwide, 2020-06-12:
Problem resolved by using the modified HiTech C file.
Thanks

closed 2020-06-12 by riwide